openwrt/packages #20895: syslog-ng: segfault on startup

pprindeville, 2023-04-27:
Maintainer: @BKPepe
Environment: x86_64, generic, HEAD (c3cb2d48da7)

Description:

`syslog-ng` dies on startup with the following message:

```
[ 6544.220357] syslog-ng[7120]: segfault at 180 ip 00007f22e95e0d43 sp 00007ffd27fd4ce8 error 6 in libssl.so.3[7f22e95d5000+45000]
[ 6544.223725] Code: e8 03 00 ff 15 8e dc 05 00 31 d2 31 c0 be 11 01 00 00 bf 14 00 00 00 ff 15 f2 d8 05 00 31 c0 5a c3 89 d1 48 8d 87 88 01 00 00 <48> 89 8f 80 01 00 00 48 39 f0 73 09 48 8d 14 08 48 39 d6 eb 0c 48
```

cc: @cotequeiroz

pprindeville, 2023-04-27:
This was happening with syslog-ng-3.38.1 and -4.4.1 which makes me wonder if the problem isn't really with Openssl introducing an regression or incompatibility.

BKPepe, 2023-04-28:
In my case, the latest version of syslog-ng is running, but its true that tested it on the stable release not on the latest edge.

pprindeville, 2023-04-28:
I can make a sandbox VM available to you if you want to look into it.  Just send me a public key.

pprindeville, 2023-09-06:
Turns out it was failing to initialize `openssl` because the engines/providers hadn't been configured when I upgraded from 1.1.x to 3.0.

closed 2023-09-06 by pprindeville