openwrt/packages #20897: netdata: Why remove ssl support from netdata

muink, 2023-04-28:
Maintainer: @BKPepe, @diizzyy
Environment: (x86_64, x86/64, 22.03.4)

Description:
Why remove ssl support from netdata?
Most modern browsers block [Mixed content][]
So, when I enable https access on luci, [ifame][] will be blocked
I had to open another http://host:port page to access it

[Mixed content]: https://developer.mozilla.org/zh-TW/docs/Web/Security/Mixed_content
[ifame]: https://github.com/muink/luci-app-netdata/blob/master/htdocs/luci-static/resources/view/netdata/main.js#L20

BKPepe, 2023-04-28:
Hi,

I don't see any reason why you would embed by using iframe Netdata to LuCI. In my opinion, it does not make any sense.
Please check my comment here: https://github.com/openwrt/luci/pull/5443#pullrequestreview-816663495 , where someone was doing exactly the same as you do and tried to upstream it.

We haven't touch Netdata for a while, so if you want to have it updated, please go ahead and create pull request.

While going through your details, I don't see that this is somehow related to this packages repository and that's why I am closing this.

closed 2023-04-28 by BKPepe

muink, 2023-04-28:
I can change the iframe to a hyperlink to open the Web UI. If there is no other solution.
What I really concerned about is the TLS capability of netdata, because I need to access netdata on my router from outside of home. TLS is needed to protect the plaintext connection.

BKPepe, 2023-04-28:
What is the reason or rather what are those reasons, that you try to inject a standalone web app to the LuCI? 
 - What is the use case?
 - Why would you have it there?
 
Netdata is using its own web server. Please follow https://learn.netdata.cloud/docs/deployment-in-production/secure-your-nodes/web-server and even Netdata itself recommends a proxy for the web server, please see https://learn.netdata.cloud/docs/deployment-in-production/secure-your-nodes/reverse-proxies/nginx

Otherwise, the Netdata would depend on OpenSSL, etc., which would take more internal storage on the router.

diizzyy, 2023-04-30:
I have same questions as BKPepe, just use the proxy module in nginx for example or modify the source yourself.