openwrt/packages #20899: nft-qos: Down limit using MAC broken

shahzaib-ai, 2023-04-28:
Maintainer: @wulfy23 
Environment: (ARMv7 Processor rev 5 (v7l), Linksys WHW01, Snapshot  r22642-90ad13c763)

Description:

Download Limit using MAC Address is not working but the upload limit works fine.

francisuk1989, 2023-05-01:
@rosywrt any help?

wulfy23, 2023-05-23:
bidirectional shaping is functional here, that is;

1) in bridged format
2) on r23001

relevant config/s;

```
#nft list ruleset | grep -A20 nft-qos
table bridge nft-qos-mac {
	chain upload {
		type filter hook prerouting priority 0; policy accept;
		ether saddr d0:50:99:zz:yy:xx limit rate over 70 kbytes/second drop
	}

	chain download {
		type filter hook postrouting priority 0; policy accept;
		ether daddr d0:50:99:zz:yy:xx limit rate over 300 kbytes/second drop
	}
}
```

```
#uci show nft-qos | grep -E '(limit|client)'
nft-qos.default.limit_enable='1'
nft-qos.default.limit_mac_enable='1'
nft-qos.@client[0]=client
nft-qos.@client[0].drunit='kbytes'
nft-qos.@client[0].urunit='kbytes'
nft-qos.@client[0].hostname='pc'
nft-qos.@client[0].macaddr='d0:50:99:zz:yy:xx'
nft-qos.@client[0].drate='300'
```

suggest providing more relevant info to assist in troubleshooting your issue

shahzaib-ai, 2023-05-24:
Here is what my config looks like after running above commands
The upload limit is working fine, only the download limit is not working.
Also, I am using luci to set limits, this is how it looks like there

https://i.imgur.com/nPFTg5A.png

```
# nft list ruleset | grep -A20 nft-qos
table inet nft-qos-mac {
        chain upload {
                type filter hook postrouting priority filter; policy accept;
                ether saddr e4:e1:71:1a:11:44 limit rate over 600 kbytes/second drop
        }

        chain download {
                type filter hook prerouting priority filter; policy accept;
                ether daddr e4:e1:71:1a:11:44 limit rate over 600 kbytes/second drop
        }
}
```

```
# uci show nft-qos | grep -E '(limit|client)'
nft-qos.default.limit_enable='0'
nft-qos.default.limit_mac_enable='1'
nft-qos.@client[0]=client
nft-qos.@client[0].drate='600'
nft-qos.@client[0].drunit='kbytes'
nft-qos.@client[0].urate='600'
nft-qos.@client[0].urunit='kbytes'
nft-qos.@client[0].hostname='phone'
nft-qos.@client[0].macaddr='e4:e1:71:1a:11:44'
```

shahzaib-ai, 2023-05-26:
Yes, offloading is disabled, and I never installed the sqm package
I ran the command, and here are the results:
```
# uci show network | grep -E '(device|ports|name)' | grep -v macadd
network.loopback.device='lo'
network.@device[0]=device
network.@device[0].name='br-lan'
network.@device[0].type='bridge'
network.@device[0].ports='eth2'
network.lan.device='br-lan'
network.wan.device='eth1'
```

wulfy23, 2023-05-26:
ok, so basically for POC, if you/ someone was to edit **/lib/nft-qos/core.sh**

3/4 down the file you will find the init function **qosdef_init_env()**

after this line
```
[ "$lt" = "bridge" ] && export NFT_QOS_HAS_BRIDGE="y"
```

add
```
	local ltD="$(uci_get "network.lan.device")"
	[ "$ltD" = "br-lan" ] && export NFT_QOS_HAS_BRIDGE="y"
```

then restart the service...

if it works would just be a matter of someone pushing cleaner code there abouts to bring package net topology enumeration up to current norms...

shahzaib-ai, 2023-05-26:
So I edited the file using nano and then restarted the nft-qos service but no success

wulfy23, 2023-05-28:
Well, that would lead us down the road of PEBCAK or PLATFORM quirks afaict... 
(I'm not really that familiar to assist beyond this point)

Searching around for your board indicates looks to me like it likely is running DSA and should report lan1 etc. in /etc/config/network ... but I could be wrong on that.

Any chance you sysupgraded from STABLE and kept configs? Not sure if your device would even have functional networking at that point... but the "lan" naming and "eth" naming seem possibly odd to me (not knowing much about the board)

In other words, we may need either someone on the same platform to reproduce/refute your issue OR others on dissimilar platforms to do the same to move forward with much confidence...

shahzaib-ai, 2023-05-28:
There is no stable release available for whw01, I installed snapshot when shifted from OEM to openwrt

openwrtdiy, 2023-06-24:
> ok, so basically for POC, if you/ someone was to edit **/lib/nft-qos/core.sh**
> 
> 3/4 down the file you will find the init function **qosdef_init_env()**
> 
> **after** this line
> 
> ```
> [ "$lt" = "bridge" ] && export NFT_QOS_HAS_BRIDGE="y"
> ```
> 
> **add**
> 
> ```
> 	local ltD="$(uci_get "network.lan.device")"
> 	[ "$ltD" = "br-lan" ] && export NFT_QOS_HAS_BRIDGE="y"
> ```
> 
> then restart the service...
> 
> if it works would just be a matter of someone pushing cleaner code there abouts to bring package net topology enumeration up to current norms...
> 
> (if you are unfamiliar with text file editing on linux there is a windows program WinSCP that may allow easy editing, although do your own due diligence before installing)

**nft-qos: Mac Address speed limit fixed**
https://github.com/openwrtdiy/packages/commit/f20f4b66d3af96ade6f650e6b000859369512c3d

Architecture	Intel(R) Celeron(R) CPU 3865U @ 1.80GHz
Target Platform	x86/64
Firmware Version	OpenWrt 21.02-SNAPSHOT r16901-6e14919fc0 / LuCI pppoe-21.02 branch git-23.106.40242-d322bff
Kernel Version	5.4.238

feckert, 2023-10-19:
I have had a look to this. My config looks like this.
```
config interface 'lan'
        option proto 'static'
        option ipaddr '192.168.0.53'
        option netmask '255.255.255.0'
        option force_link '1'
        option device 'br-lan'

config interface 'wan'
        option device 'br-wan'
        option proto 'dhcp'
        option metric '100'
        option force_link '0'

config device
        option type 'bridge'
        option name 'br-wan'
        list ports 'wan'

config device
        option type 'bridge'
        option name 'br-lan'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
```
So if we want to fix this sustainably, we must evaluate this programmatically from the `lan` interface section the device name `br-lan`. And then have to iterate over all `config device` section and compare the option `name` with the device name. And if this match then evaluating the device type. In my case this is a type `bridge`. And then we could set `NFT_QOS_HAS_BRIDGE="y"` for this interface.

wehagy, 2025-09-25:
Package removed in the commit https://github.com/openwrt/packages/commit/14bff6e90ef8ebe22d72b4ede398e306ea3b46da.

closed 2025-09-25 by wehagy