openwrt/packages #20904: nginx fail since last commit series.

anomeome, 2023-04-29:
@Ansuel 

_nginx_ failure to launch:
```
Sat Apr 29 15:48:23 2023 daemon.err nginx_init: 2023/04/29 15:48:23 [emerg] 2765#0: unknown directive "ubus_interpreter" in /etc/nginx/conf.d/luci.locations:21
Sat Apr 29 15:48:23 2023 daemon.err nginx_init: nginx: configuration file /etc/nginx/uci.conf test failed
Sat Apr 29 15:48:23 2023 daemon.err nginx_init: NOT using conf file!
Sat Apr 29 15:48:23 2023 daemon.notice procd: /etc/rc.d/S80nginx: nginx_init: NOT using conf file!
Sat Apr 29 15:48:23 2023 daemon.notice procd: /etc/rc.d/S80nginx: show config to be used by: nginx -T -c '/etc/nginx/uci.conf'
```
is there any manual changes needed. mvebu : rango

Ansuel, 2023-04-29:
Yes but in theory a migration script should have applied and added the new stuff

Arduous, 2023-04-30:
Encountering the same failure. What are the migration scripts? What are the triggers to these scripts? Thank you

bauw2008, 2023-05-01:
nginx_init: NOT using conf file!
show config to be used by: nginx -T -c '/etc/nginx/uci.conf'
root@OpenWrt:~# nginx -T
2023/05/01 23:01:07 [emerg] 7935#0: open() "/etc/nginx/nginx.conf" failed (2: No                   such file or directory)
nginx: configuration file /etc/nginx/nginx.conf test failed
root@OpenWrt:~# /etc/init.d/rpcd restart
root@OpenWrt:~# service nginx reload
Command failed: Not found

Encountering the same failure.

Ansuel, 2023-05-01:
Yep already know the problem plan to solve today

hgl, 2023-05-04:
This is addressed in #20902, but you will still encounter a 502 error when visiting LuCI. To fix that, the uwsgi package needs to be updated, as its `chdir` option currently points to `/usr/lib/lua/luci/`, which doesn't exist.

Ansuel, 2023-05-04:
@hgl can you point me how you fixed the issue?

hgl, 2023-05-04:
> @hgl can you point me how you fixed the issue?

To be honest, I have no idea. I guess it was fixed as a side effect of the refactoring. I have manually run the built result to confirm.

hgl, 2023-05-04:
But I only tested building into firmware with the image builder, from your fix it seems the issue manifests when install as standalone package?

mmartinv, 2023-05-05:
It seems the `module.d/luci.module` is not actually included within the generated `uci.conf`. That's why the `ubus_interpreter` directive is unknown. Adding `include module.d/*.module;` to the `uci.conf.template` seems to fix the issue. However it looks like there's something wrong with the included module as it fails now with:
```
2023/05/05 09:14:16 [emerg] 4626#0: module "/usr/lib/nginx/modules/ngx_http_ubus_module.so" is not binary compatible in /etc/nginx/module.d/luci.module:1
```

Ansuel, 2023-05-06:
@mmartinv what is the arch of this? can you post your config?

darekxan, 2023-05-07:
Got same problem

```
root@OpenWrt:~# nginx -T -c '/etc/nginx/uci.conf'
2023/05/07 17:37:02 [emerg] 7895#0: module "/usr/lib/nginx/modules/ngx_http_ubus_module.so" is not binary compatible in /etc/nginx/module.d/luci.module:1
nginx: configuration file /etc/nginx/uci.conf test failed
root@OpenWrt:~# file /usr/sbin/nginx 
/usr/sbin/nginx: ELF 64-bit LSB executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-musl-aarch64.so.1, no section header
root@OpenWrt:~# file /usr/lib/nginx/modules/ngx_http_ubus_module.so
/usr/lib/nginx/modules/ngx_http_ubus_module.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, no section header
```

mmartinv, 2023-05-10:
@Ansuel the device is a Linksys wrt32x (mvebu: cortexa9) and I got similar results to what @droid4joy already posted.
I cannot post my current configuration because I reinstalled the router with 21.02.7 :(

anomeome, 2023-05-10:
After the addition of the include as per the PR thread I am seeing [no issue](https://termbin.com/3x950) on a _rango_ (WRT3200ACM), which is pretty much the same as the above _venom_ WRT32X.

hgl, 2023-05-11:
@droid4joy [From nginx's mailing list](https://mailman.nginx.org/pipermail/nginx-devel/2018-May/011123.html), they did recommend enabling the flag (and even considered enabling it by default), and debian's nginx package also enabled this flag.

I've added it in the refactor PR (also rebased), could folks who encountered the issue try it out?

@Ansuel: still waiting on your review.

closed 2023-05-20 by Ansuel

Arduous, 2023-06-06:
My build and deployment process is pretty specific, so for some reason the line `include module.d/*.module;` was added at the bottom of the `uci.conf.template` file. This caused the same `unknown directive "ubus_interpreter"` to happen on current (June 2023) master. My workaround was to manually edit `uci.conf.template`, and cut and paste the include line close to the top of the file.

commit 6a3cc76e (2023-05-20): nginx: introduce migration for old uci conf template

Introduce support for migration of old uci conf template to new version.
Uci conf template are saved in config backup. This cause problem on config
restore as old config template might have compatibility problem with new
nginx implementation.

Add logic to migrate the template script at runtime to correctly align
to latest change from nginx and nginx-util.

Fixes: 65a676ed56fb ("nginx: introduce support for dynamic modules")
Fixes: #20904
PR #20928 (2023-05-20): nginx: fix missing module entry on package update
Uci defaults script are run only on first installation. On package upgrade, the ubus module for the nginx-mod-luci is not installed in the nginx conf. Fix this by adding a post install script.

Fixes: 65a676ed56fb ("nginx: introduce support for dynamic modules")
Fixes: #20904

---

@anomeome can you test? I assume your problem was with upgrading the package right? or it was from a clean image?