openwrt/packages #20906: nginx-full fail to compile after last commit

zcy85611, 2023-04-30:
with everthing unchanged, nginx-full fail to compile, it gives "install: cannot stat '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-all-module/nginx-1.24.0/ipkg-install/usr/lib/nginx/modules/ngx_stream_module.so': No such file or directory"

would you please take a look, thanks?

```
make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/nginx-util'
mkdir -p /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-util-1.6/.pkgdir/nginx-ssl-util/. /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.nginx-ssl-util_installed
echo "nginx-ssl-util" >> /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/nginx-util.default.install
mkdir -p /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-util-1.6/.pkgdir/nginx-util/. /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.nginx-util_installed
echo "nginx-util" >> /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/nginx-util.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#0.17#0.06#0.22
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/dbus'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/dbus'
time: package/feeds/packages/dbus/compile#0.16#0.04#0.18
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/attr'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/attr'
time: package/feeds/packages/attr/compile#0.14#0.04#0.17
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/libffi'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/compile#0.18#0.04#0.21
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/pcre2'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/pcre2'
time: package/feeds/packages/pcre2/compile#0.18#0.04#0.21
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/glib2'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.22#0.05#0.25
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/libical'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libical'
time: package/feeds/packages/libical/compile#0.13#0.06#0.18
make[3]: Entering directory '/workdir/openwrt/package/libs/readline'
mkdir -p /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
echo '8' | cmp -s - /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libreadline.version || { echo '8' > /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/libreadline.version;  } 
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/readline-8.2/.pkgdir/libreadline/. /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libreadline_installed
echo "libreadline" >> /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/readline.default.install
make[3]: Leaving directory '/workdir/openwrt/package/libs/readline'
time: package/libs/readline/compile#0.22#0.07#0.27
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/bluez'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/bluez'
time: package/feeds/packages/bluez/compile#0.15#0.06#0.19
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/gdbm'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/gdbm'
time: package/feeds/packages/gdbm/compile#0.15#0.03#0.17
make[3]: Entering directory '/workdir/openwrt/feeds/packages/lang/python/python-pip-conf'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/python/python-pip-conf'
time: package/feeds/packages/python-pip-conf/compile#0.28#0.06#0.34
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/libffi'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/libffi'
time: package/feeds/packages/libffi/host-compile#0.17#0.04#0.20
make[3]: Entering directory '/workdir/openwrt/package/utils/bzip2'
make[3]: Leaving directory '/workdir/openwrt/package/utils/bzip2'
time: package/utils/bzip2/host-compile#0.35#0.05#0.38
make[3]: Entering directory '/workdir/openwrt/feeds/packages/lang/python/python3'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/lang/python/python3'
time: package/feeds/packages/python3/host-compile#0.35#0.06#0.38
make[3]: Entering directory '/workdir/openwrt/feeds/packages/libs/sqlite3'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/libs/sqlite3'
time: package/feeds/packages/sqlite3/compile#0.16#0.03#0.18
make[3]: Entering directory '/workdir/openwrt/feeds/packages/utils/xz'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/xz'
time: package/feeds/packages/xz/compile#0.16#0.04#0.19
make[3]: Entering directory '/workdir/openwrt/package/utils/bzip2'
make[3]: Leaving directory '/workdir/openwrt/package/utils/bzip2'
time: package/utils/bzip2/compile#0.36#0.05#0.39
[... 13 lines cut ...]
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/pypi/uwsgi-2.0.20/.pkgdir/uwsgi-cgi-plugin/. /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.uwsgi-cgi-plugin_installed
echo "uwsgi-cgi-plugin" >> /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/uwsgi.default.install
mkdir -p /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp
SHELL= flock /workdir/openwrt/tmp/.root-copy.flock -c 'cp -fpR /workdir/openwrt/build_dir/target-mipsel_24kc_musl/pypi/uwsgi-2.0.20/.pkgdir/uwsgi-luci-support/. /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/'
touch /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.uwsgi-luci-support_installed
echo "uwsgi-luci-support" >> /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo/uwsgi.default.install
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/uwsgi'
time: package/feeds/packages/uwsgi/compile#0.26#0.08#0.33
make[3]: Entering directory '/workdir/openwrt/feeds/packages/net/nginx'
mkdir -p /workdir/openwrt/bin/targets/ramips/mt7621/packages /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-all-module/nginx-1.24.0/ipkg-mipsel_24kc/nginx-mod-stream/CONTROL /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/pkginfo
install -d -m0755 /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-all-module/nginx-1.24.0/ipkg-mipsel_24kc/nginx-mod-stream/usr/lib/nginx/modules
install -m0755 /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-all-module/nginx-1.24.0/ipkg-install/usr/lib/nginx/modules/ngx_stream_module.so /workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-all-module/nginx-1.24.0/ipkg-mipsel_24kc/nginx-mod-stream/usr/lib/nginx/modules
install: cannot stat '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/nginx-all-module/nginx-1.24.0/ipkg-install/usr/lib/nginx/modules/ngx_stream_module.so': No such file or directory
make[3]: *** [Makefile:643: /workdir/openwrt/bin/packages/mipsel_24kc/packages/nginx-mod-stream_1.24.0-1_mipsel_24kc.ipk] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/nginx'
time: package/feeds/packages/nginx/all-module/compile#0.23#0.06#0.26
    ERROR: package/feeds/packages/nginx failed to build (build variant: all-module).
make[2]: *** [package/Makefile:116: package/feeds/packages/nginx/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
```

Ansuel, 2023-05-04:
@zcy85611 still there?

zcy85611, 2023-05-04:
> @zcy85611 still there?

yes

zcy85611, 2023-05-05:
> @zcy85611 output of `/workdir/openwrt/build_dir/target-mipsel_24kc_musl/` ?

I'm using github actions, so I'm not sure what's the output of this folder

EdiTurn, 2023-05-05:
> @zcy85611 output of `/workdir/openwrt/build_dir/target-mipsel_24kc_musl/` ?

Same issue here. 
Here is the output, with a new clone of source code, only select the nginx-full package . 
[image]

commit 8c704f2c (2023-05-06): nginx: fix compilation error for nginx-full

Fix compilation error for stream module not converted to use the PACKAGE
config flag and a missing required dependency for the DAV ext module.

Drop additional config for STREAM module since they are now included and
built by default.

Fixes: 65a676ed56fb ("nginx: introduce support for dynamic modules")
Fixes: #20906
PR #20934 (2023-05-06): nginx: fix compilation error for nginx-full
Fix compilation error for stream module not converted to use the PACKAGE config flag and a missing required dependency for the DAV ext module.

Drop additional config for STREAM module since they are now included and built by default.

Fixes: 65a676ed56fb ("nginx: introduce support for dynamic modules")
Fixes: #20906

---

@EdiTurn @zcy85611 Can you test this?
closed 2023-05-06 by Ansuel