openwrt/packages #20911: golang: GO_PKG_GCFLAGS could be improved

hgl, 2023-04-30:
Maintainer: @jefferyto 
Environment: (put here arch, model, OpenWrt version)

Description:
Since Go 1.10, `-gcflags` accepts `[pattern=]arg list`. WIthout the pattern, the flags only applies to the current package. `GO_PKG_GCFLAGS` currently doesn't support specifying the pattern, making the commented example `GO_PKG_GCFLAGS:=-N -l` only disables optimization for the current package, not its dependencies.

A few thoughts regarding the fix:

1. One of the most common usage for gcflags is probably disabling optimization to test binary on OpenWrt, should something like `GO_PKG_DISABLE_OPTIMIZATION` be introduced?
2. To specify multiple patterns, multiple `-gcflags` has to be used. This makes it impossible to specify such values in a single make variable, e.g.,`GO_PKG_GCFLAGS := foo='-a -b' bar='-c -d'` won't work because of the spaces. The only solution I can think of is to introduce a make variable that can add flags directly in the command line, e.g., `GO_PKG_ARGS := -gcflags foo='-a -b' -gcflags bar='-c -d'`
3. The previous might be overkill, given that current no package uses `GO_PKG_GCFLAGS`, if we simply make it add its value after the `all=` pattern, it can probably address 80% of the use cases, and `GO_PKG_GCFLAGS:=-N -l` can work as expected.

Thoughts?

hgl, 2023-05-05:
@jefferyto ping

jefferyto, 2023-09-04:
Apologies for the late response - do you have a specific use-case/issue with `GO_PKG_GCFLAGS`? As you mentioned, there are no packages in this repo that set gcflags, and I would prefer not to expand this functionality without a concrete purpose.

closed 2023-09-06 by hgl