openwrt/packages #20927: [utils/shadow] bcrypt breaks things like web login and pam login

Arunmor1510, 2023-05-04:
Maintainer: unknown (none stated)

Environment: tree wide

Description: In #12490 there's a issue on whether `shadow` should use `bcrypt` to generate password hash, which is fixed by #12530, later #12577 makes `shadow` uses `libxcrypt` so the `bcrypt` is always available and preferred. However it seems that packages like `rpcd` and `libpam` don't use `libxcrypt` so they may fail when password is hashed by `bcrypt`. Although manually specifying hashing algorithm is a solution, it's better to change the default behavior.

brada4, 2023-05-04:
No multi-iteration format available in busybox, and default is MD5 aka $1$, the best can be done default-wise is to select SHA-512 $6$ default algo?

Arunmor1510, 2023-05-05:
These packages uses `crypt()` function from the libc library, so it depends on the libc's implementation. 
For example, [glibc](https://www.gnu.org/software/libc/manual/html_node/Passphrase-Storage.html) listed their supported hashes.

brada4, 2023-05-05:
It is not possible to plug extra authentication scheme for root and web and dropbear
Normally only root is with password due to very embedded nature of the system.
You can use other schemes for some .htaccess and similar, or even unlikely extra users but that is all there could be.
Say what about campaigning for SHA2 hash by default or even logging DES and MD5 usage?

Arunmor1510, 2023-05-05:
I've tested sha512 and it's working fine. I'm using `glibc+openssh+libpam`. In fact it's better to remove `libxcrypt` because only `shadow` is using it. Or from `libxcrypt` 's README, it can be built with compatibility for `glibc's libcrypt`, but I don't know if it works for `musl`.

brada4, 2023-05-05:
I based my opinion on musl.
`-a ALG  des,md5,sha256/512 (default md5)`

Do you know packages that by default try to set root pasword to incompatible format? It takes some digging across libxcrypt dependencies and verifying that they default to either SHA2?

Do you want to take up on improving world by changing busybox build default to either SHA2 (5 or 6) hash in ../openwrt (some packages, like kmods and those in system images are maintained over there).

Arunmor1510, 2023-05-05:
`passwd` from `shadow` uses `/etc/login.defs` to define default password algorithm. Not sure about `busybox` but I think it's only configurable while building? It sounds good to use stronger algorithm if available.

brada4, 2023-05-05:
its in file /etc/login.defs of package shadow-common 

```diff
diff --git a/utils/shadow/Makefile b/utils/shadow/Makefile
index ab2ba224b..243702e8f 100644
--- a/utils/shadow/Makefile
+++ b/utils/shadow/Makefile
@@ -132,7 +132,7 @@ define Package/shadow-common/install
        $(INSTALL_DIR) $(1)/etc
        $(INSTALL_CONF) $(PKG_INSTALL_DIR)/etc/login.defs $(1)/etc/
        $(SED) 's,SU_NAME,#SU_NAME,g' $(1)/etc/login.defs
-       $(SED) 's,#ENCRYPT_METHOD DES,ENCRYPT_METHOD BCRYPT,g' $(1)/etc/login.defs
+       $(SED) 's,#ENCRYPT_METHOD DES,ENCRYPT_METHOD SHA512,g' $(1)/etc/login.defs
 endef
 
 define Package/shadow-utils/install
```

brada4, 2023-05-05:
@neheb - please review if xcrypt dependency is still needed, but certainly default shall not break the system authentication for unaware.
@Arunmor1510 feel free to rephrase the original post according to `opkg search /etc/login.defs` etc. Applies to both releases and snapshots as of today.

Arunmor1510, 2023-05-05:
Thanks for the solution. 
`libxcrpt`  should be either removed, or made available to install and linked by other programs, but compatibility test is needed. It can also produce library with different SONAME while dropping compatibility api, but I'm not familiar with the building mechanism, so don't know how to make use of it.

brada4, 2023-05-05:
Lets leave that decision to maintainer. Minimally package shall not lock out root user after changing password.