openwrt/packages #20933: [utils/coreutils] please update to 9.3 to solve bug for openwrt run in lxc containers

Arunmor1510, 2023-05-05:
Environment: openwrt x86_64 built with lastest git version, run in unprivileged lxc container

Description: With `coreutils 9.2`, when using utilities like `cp`,`mv`, they will fail. [bug report on gnu.org](https://debbugs.gnu.org/cgi/bugreport.cgi?bug=62404)

stangri, 2023-05-05:
Tagging last committer and maintainer @httpstorm @jow.

httpstorm, 2023-05-08:
@Arunmor1510 @stangri 
coreutils: update to 9.3 https://github.com/openwrt/packages/pull/20965

1715173329, 2023-05-09:
i guess this issue can be closed ;)

httpstorm, 2023-05-09:
@1715173329 
Thanks! I'm curious why does it say `Unverified`, considering I have added `Signed-off-by:`?

1715173329, 2023-05-09:
ahhh i enabled [vigilant mode](https://docs.github.com/github/authenticating-to-github/displaying-verification-statuses-for-all-of-your-commits).
i cannot sign it when using rebase merging provided by github web. it's not a big issue, just ignore it

closed 2023-05-10 by Arunmor1510