openwrt/packages #20949: libwebsockets: fails to build with gcc-13

oskarirauta, 2023-05-07:
but adding `TARGET_CFLAGS += -Wno-error=enum-int-mismatch` to Makefile allows building..

oskarirauta, 2023-05-07:
@blogic

karlp, 2023-05-08:
Please track upstream's fixes for this as well, looks like https://github.com/warmcat/libwebsockets/pull/2824 is the path they're taking.

oskarirauta, 2023-05-08:
FYI

https://github.com/openwrt/packages/issues/20961#issuecomment-1538111054

similar issue fixed like this.
But surely libwebsockets comes up with a new release with proposed fix soon..

oskarirauta, 2023-05-28:
@karlp warm cat/libwebsockets#2824 is not sufficient. I tried it. It works on no-ssl variant, but fails on mbedtls variant. Probably with openssl as well.

oskarirauta, 2023-05-28:
PR #21206 is available

karlp, 2023-05-29:
closed as superseded

closed 2023-05-29 by karlp