openwrt/packages #20961: nginx-util: fail to compile on GCC 13

EdiTurn, 2023-05-08:
When choose to use GCC 13 in `make menuconfig`, nginx-util fail to compile. 

```
make[2]: Entering directory '/home/editurn/openwrt/openwrt/feeds/packages/net/nginx-util'
rm -f /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/.built
touch /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/.built_check
MAKEFLAGS="" /home/editurn/openwrt/openwrt/staging_dir/host/bin/ninja  -j1 -C /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6
ninja: Entering directory `/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6'
[1/4] Building CXX object CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o
FAILED: CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o
/home/editurn/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DVERSION=1.6  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6=nginx-util-1.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error=deprecated-declarations -DNDEBUG -std=gnu++17   -Os -Wall -Werror -Wextra -g3 -Wno-unused-parameter -Wmissing-declarations -Wshadow -MD -MT CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o -MF CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o.d -o CMakeFiles/nginx-ssl-util.dir/nginx-util.cpp.o -c /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/nginx-util.cpp
In file included from /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/nginx-ssl-util.hpp:12,
                 from /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/nginx-util.cpp:4:
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp: In function 'EVP_PKEY_ptr gen_eckey(int)':
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:94:29: warning: 'EC_KEY* EC_KEY_new()' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   94 |     auto* eckey = EC_KEY_new();
      |                   ~~~~~~~~~~^~
In file included from /home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/x509.h:33,
                 from /home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/pem.h:23,
                 from /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:8:
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/ec.h:968:31: note: declared here
  968 | OSSL_DEPRECATEDIN_3_0 EC_KEY *EC_KEY_new(void);
      |                               ^~~~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:97:30: warning: 'int EC_KEY_set_group(EC_KEY*, const EC_GROUP*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   97 |         if ((EC_KEY_set_group(eckey, group) == 0) || (EC_KEY_generate_key(eckey) == 0)) {
      |              ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/ec.h:1042:27: note: declared here
 1042 | OSSL_DEPRECATEDIN_3_0 int EC_KEY_set_group(EC_KEY *key, const EC_GROUP *group);
      |                           ^~~~~~~~~~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:97:74: warning: 'int EC_KEY_generate_key(EC_KEY*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   97 |         if ((EC_KEY_set_group(eckey, group) == 0) || (EC_KEY_generate_key(eckey) == 0)) {
      |                                                       ~~~~~~~~~~~~~~~~~~~^~~~~~~
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/ec.h:1101:27: note: declared here
 1101 | OSSL_DEPRECATEDIN_3_0 int EC_KEY_generate_key(EC_KEY *key);
      |                           ^~~~~~~~~~~~~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:98:24: warning: 'void EC_KEY_free(EC_KEY*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
   98 |             EC_KEY_free(eckey);
      |             ~~~~~~~~~~~^~~~~~~
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/ec.h:1003:28: note: declared here
 1003 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:117:20: warning: 'void EC_KEY_free(EC_KEY*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  117 |         EC_KEY_free(eckey);
      |         ~~~~~~~~~~~^~~~~~~
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/ec.h:1003:28: note: declared here
 1003 | OSSL_DEPRECATEDIN_3_0 void EC_KEY_free(EC_KEY *key);
      |                            ^~~~~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp: In function 'EVP_PKEY_ptr gen_rsakey(int, long unsigned int)':
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:142:24: warning: 'RSA* RSA_new()' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  142 |     auto* rsa = RSA_new();
      |                 ~~~~~~~^~
In file included from /home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/x509.h:36:
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/rsa.h:201:28: note: declared here
  201 | OSSL_DEPRECATEDIN_3_0 RSA *RSA_new(void);
      |                            ^~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:146:33: warning: 'int RSA_generate_key_ex(RSA*, int, BIGNUM*, BN_GENCB*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  146 |             (RSA_generate_key_ex(rsa, keysize, bignum, nullptr) == 0))
      |              ~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/rsa.h:260:27: note: declared here
  260 | OSSL_DEPRECATEDIN_3_0 int RSA_generate_key_ex(RSA *rsa, int bits, BIGNUM *e,
      |                           ^~~~~~~~~~~~~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:148:21: warning: 'void RSA_free(RSA*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  148 |             RSA_free(rsa);
      |             ~~~~~~~~^~~~~
/home/editurn/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/openssl/rsa.h:293:28: note: declared here
  293 | OSSL_DEPRECATEDIN_3_0 void RSA_free(RSA *r);
      |                            ^~~~~~~~
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/px5g-openssl.hpp:168:17: warning: 'void RSA_free(RSA*)' is deprecated: Since OpenSSL 3.0 [-Wdeprecated-declarations]
  168 |         RSA_free(rsa);
      |         ~~~~~~~~^~~~~
[... 58 lines cut ...]
                 from /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/nginx-ssl-util.hpp:11:
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/ubus-cxx.hpp: In instantiation of 'auto ubus::message::filter(Strings ...) [with Strings = {const char*}]':
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/nginx-ssl-util.hpp:108:63:   required from here
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/ubus-cxx.hpp:162:55: error: moving a temporary object prevents copy elision [-Werror=pessimizing-move]
  162 |         return std::move(message{msg, std::move(both)});
      |                                                       ^
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/ubus-cxx.hpp:162:55: note: remove 'std::move' call
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/ubus-cxx.hpp: In instantiation of 'auto ubus::message::filter(Strings ...) [with Strings = {const char*, const char*, const char*}]':
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/nginx-util.cpp:27:53:   required from here
/home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/ubus-cxx.hpp:162:55: error: moving a temporary object prevents copy elision [-Werror=pessimizing-move]
[... cut ...]
make[2]: *** [Makefile:134: /home/editurn/openwrt/openwrt/build_dir/target-x86_64_musl/nginx-util-1.6/.built] Error 1
make[2]: Leaving directory '/home/editurn/openwrt/openwrt/feeds/packages/net/nginx-util'
time: package/feeds/packages/nginx-util/compile#4.53#0.15#4.52
    ERROR: package/feeds/packages/nginx-util failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/nginx-util/compile] Error 1
make[1]: Leaving directory '/home/editurn/openwrt/openwrt'
make: *** [/home/editurn/openwrt/openwrt/include/toplevel.mk:231：package/nginx-util/compile] 错误 2
```

WYC-2020, 2023-05-08:
[image]

neheb, 2023-05-08:
ping @peter-stadler

peter-stadler, 2023-05-16:
Thank you. I will be able to test it in two weeks only, but the change looks fine. After two I will also look to fix the code. In the meantime I think it would be good to use the proposed change.

wbx-github, 2025-03-22:
Hi, the bug still exist in openwrt 24.10 branch. Is it possible to fix it?