openwrt/packages #20969: gnutls: build fails (gcc 13)

oskarirauta, 2023-05-08:
gnutls build fails with latest changes, **might** be related to gcc 13 - or not.

```
/bin/sh ../libtool  --tag=CC   --mode=link x86_64-openwrt-linux-musl-gcc -fno-builtin-strcmp  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/usr/src/openwrt/build_dir/target-x86_64_musl/gnutls-3.8.0=gnutls-3.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   -L/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/usr/lib -L/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib -znow -zrelro  -o psktool psk.o ../lib/libgnutls.la libcmd-psk.la ../gl/libgnu.la  gl/libgnu_gpl.la 
OpenWrt-libtool: link: x86_64-openwrt-linux-musl-gcc -fno-builtin-strcmp -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/usr/src/openwrt/build_dir/target-x86_64_musl/gnutls-3.8.0=gnutls-3.8.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -znow -zrelro -o .libs/psktool psk.o  -L/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/usr/lib -L/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib ../lib/.libs/libgnutls.so -L/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/lib -lnettle -lhogweed -lgmp ./.libs/libcmd-psk.a ../gl/.libs/libgnu.a gl/.libs/libgnu_gpl.a
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `u8_check'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Sc'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_general_category_and_not'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_So'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Cc'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Sm'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Sk'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Pe'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Pc'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Pi'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_is_property_not_a_character'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Ps'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `u16_to_u8'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_decomposition'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Lt'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Po'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Ll'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Mn'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `u8_to_u16'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_No'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_is_general_category'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `u32_to_u8'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_is_property_join_control'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_is_property_default_ignorable_code_point'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Pd'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Lm'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Pf'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `u8_to_u32'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `u16_normalize'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Lu'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uninorm_nfc'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Lo'
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/../../../../x86_64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CAT [... cut ...]
make[2]: *** [Makefile:269: /usr/src/openwrt/build_dir/target-x86_64_musl/gnutls-3.8.0/.built] Error 2
make[2]: Leaving directory '/usr/src/openwrt/feeds/packages/libs/gnutls'
time: package/feeds/packages/gnutls/compile#0.67#0.07#0.00
    ERROR: package/feeds/packages/gnutls failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
make[1]: Leaving directory '/usr/src/openwrt'
make: *** [/usr/src/openwrt/include/toplevel.mk:231: package/gnutls/compile] Error 2
dev@media:/usr/src/openwrt# 
```

oskarirauta, 2023-05-08:
from configure part...

```
Copying file po/Makevars.template
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: running: true --force
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/aclocal -I /usr/src/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /usr/src/openwrt/staging_dir/hostpkg/share/aclocal -I /usr/src/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . -I lib --force -I m4 -I src/gl/m4 -I lib/unistring/m4
autoreconf: configure.ac: tracing
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/libtoolize --install --force
OpenWrt-libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
OpenWrt-libtoolize: linking file 'build-aux/config.guess'
OpenWrt-libtoolize: linking file 'build-aux/config.sub'
OpenWrt-libtoolize: linking file 'build-aux/install-sh'
OpenWrt-libtoolize: linking file 'build-aux/ltmain.sh'
OpenWrt-libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
OpenWrt-libtoolize: linking file 'm4/libtool.m4'
OpenWrt-libtoolize: You should add the contents of 'm4/libtool.m4' to 'aclocal.m4'.
OpenWrt-libtoolize: linking file 'm4/ltoptions.m4'
OpenWrt-libtoolize: You should add the contents of 'm4/ltoptions.m4' to 'aclocal.m4'.
OpenWrt-libtoolize: linking file 'm4/ltsugar.m4'
OpenWrt-libtoolize: You should add the contents of 'm4/ltsugar.m4' to 'aclocal.m4'.
OpenWrt-libtoolize: linking file 'm4/ltversion.m4'
OpenWrt-libtoolize: linking file 'm4/lt~obsolete.m4'
OpenWrt-libtoolize: You should add the contents of 'm4/lt~obsolete.m4' to 'aclocal.m4'.
autoreconf: configure.ac: not using Intltool
autoreconf: running: true
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/aclocal -I /usr/src/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /usr/src/openwrt/staging_dir/hostpkg/share/aclocal -I /usr/src/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . -I lib --force -I m4 -I src/gl/m4 -I lib/unistring/m4
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/autoconf --include=/usr/src/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal --include=/usr/src/openwrt/staging_dir/hostpkg/share/aclocal --include=/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --include=lib --prepend-include=/usr/src/openwrt/staging_dir/host/share/aclocal --force
configure.ac:55: warning: The macro `AC_PROG_CC_C99' is obsolete.
configure.ac:55: You should run autoupdate.
./lib/autoconf/c.m4:1659: AC_PROG_CC_C99 is expanded from...
configure.ac:55: the top level
configure.ac:100: warning: The macro `AC_HEADER_STDC' is obsolete.
configure.ac:100: You should run autoupdate.
./lib/autoconf/headers.m4:704: AC_HEADER_STDC is expanded from...
configure.ac:100: the top level
configure.ac:337: warning: The macro `AC_TRY_COMPILE' is obsolete.
configure.ac:337: You should run autoupdate.
./lib/autoconf/general.m4:2847: AC_TRY_COMPILE is expanded from...
m4/hooks.m4:34: LIBGNUTLS_HOOKS is expanded from...
configure.ac:337: the top level
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/autoheader --include=/usr/src/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal --include=/usr/src/openwrt/staging_dir/hostpkg/share/aclocal --include=/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal --include=m4 --include=. --include=lib --prepend-include=/usr/src/openwrt/staging_dir/host/share/aclocal --force
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/automake --add-missing --force-missing
configure.ac:39: installing 'build-aux/compile'
configure.ac:29: installing 'build-aux/missing'
aminclude_static.am:52: warning: if $(CODE_COVERAGE_BRANCH_COVERAGE: non-POSIX variable name
aminclude_static.am:52: (probably a GNU make extension)
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:59: warning: if $(CODE_COVERAGE_BRANCH_COVERAGE: non-POSIX variable name
aminclude_static.am:59: (probably a GNU make extension)
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:63: warning: ':='-style assignments are not portable
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:64: warning: code_coverage_v_lcov_cap_$(V: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:65: warning: code_coverage_v_lcov_cap_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:67: warning: code_coverage_v_lcov_ign_$(V: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:68: warning: code_coverage_v_lcov_ign_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:70: warning: code_coverage_v_genhtml_$(V: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:71: warning: code_coverage_v_genhtml_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:73: warning: code_coverage_quiet_$(V: non-POSIX recursive variable expansion
Makefile.am:167:   'aminclude_static.am' included from here
aminclude_static.am:74: warning: code_coverage_quiet_$(AM_DEFAULT_VERBOSITY: non-POSIX recursive variable expansion
[... cut ...]
doc/Makefile.am: installing 'build-aux/depcomp'
autoreconf: Leaving directory '.'
autoreconf: export WARNINGS=
autoreconf: Entering directory '.'
autoreconf: running: true --force
autoreconf: running: /usr/src/openwrt/staging_dir/host/bin/aclocal -I /usr/src/openwrt/staging_dir/target-x86_64_musl/host/share/aclocal -I /usr/src/openwrt/staging_dir/hostpkg/share/aclocal -I /usr/src/openwrt/staging_dir/target-x86_64_musl/usr/share/aclocal -I m4 -I . -I lib --force -I m4 -I src/gl/m4 -I lib/unistring/m4

```

oskarirauta, 2023-05-08:
I got it to build when I moved staging_dir/usr/lib/libunistring** and staging_dir/usr/include/unistring momentarily to tmp during build process, srptool though is not built even when set in config to support srp..

oskarirauta, 2023-05-08:
I disabled srp support, with moving unistring lib and headers to safe keep during build, I got gnutls to build succesfully.... though with this trickery and without srp support.

Listed references belong to libunistring, gnutls should be built using included unistring, but for some reason it seems to link to _real_ libunistring instead - or something like that.

oskarirauta, 2023-05-08:
Unfortunately after this, I returned files to staging_dir and tried to attempt to continue full build - and it decided to try to re-build gnutls which of course caused failure, because files were returned to where they were... This needs a real solution.

wzming, 2023-05-10:
I also encounter this problem, when I select luci-app-samba4, the compilation reports this error

ecrasy, 2023-05-10:
I encounter the same issue when build for my raspberry pi 4b model ,
and it seems like not gcc13 related ?

```
2023-05-09T10:18:46.7437690Z /home/runner/work/workspace/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `UC_CATEGORY_Nl'
2023-05-09T10:18:46.7439013Z /home/runner/work/workspace/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uninorm_nfkc'
2023-05-09T10:18:46.7441408Z /home/runner/work/workspace/openwrt/staging_dir/toolchain-aarch64_cortex-a72_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/../../../../aarch64-openwrt-linux-musl/bin/ld: ../lib/.libs/libgnutls.so: undefined reference to `uc_general_category_or'
2023-05-09T10:18:46.7442128Z collect2: error: ld returned 1 exit status
2023-05-09T10:18:46.7442440Z make[6]: *** [Makefile:2637: psktool] Error 1
2023-05-09T10:18:46.7442752Z make[5]: *** [Makefile:2464: all] Error 2
2023-05-09T10:18:46.7443361Z make[4]: *** [Makefile:2278: all-recursive] Error 1
2023-05-09T10:18:46.7443664Z make[3]: *** [Makefile:2203: all] Error 2
2023-05-09T10:18:46.7444363Z make[2]: *** [Makefile:269: /home/runner/work/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/gnutls-3.8.0/.built] Error 2
2023-05-09T10:18:46.7456736Z make[6]: Leaving directory '/home/runner/work/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/gnutls-3.8.0/src'
2023-05-09T10:18:46.7459453Z make[5]: Leaving directory '/home/runner/work/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/gnutls-3.8.0/src'
2023-05-09T10:18:46.7460313Z make[4]: Leaving directory '/home/runner/work/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/gnutls-3.8.0'
2023-05-09T10:18:46.7461184Z make[3]: Leaving directory '/home/runner/work/workspace/openwrt/build_dir/target-aarch64_cortex-a72_musl/gnutls-3.8.0'
2023-05-09T10:18:46.7461975Z make[2]: Leaving directory '/home/runner/work/workspace/openwrt/feeds/packages/libs/gnutls'
2023-05-09T10:18:46.7462472Z time: package/feeds/packages/gnutls/compile#303.96#29.18#327.93
2023-05-09T10:18:46.7467824Z     ERROR: package/feeds/packages/gnutls failed to build.
2023-05-09T10:18:46.7468271Z make[1]: *** [package/Makefile:116: package/feeds/packages/gnutls/compile] Error 1
```

micmac1, 2023-05-13:
Hi all,

Unrelated to gcc-13. Kindly check/review pull request #21007.

Kind regards,
Seb

neheb, 2024-10-27:
No longer an issue.

closed 2024-10-27 by neheb