openwrt/packages #20971: syslog-ng: late start on boot prevents dnsmasq from starting

ardrabczyk, 2023-05-09:
Maintainer: @BKPepe
Environment: armv7l, InHand Networks VG710, OpenWrt 22.03.0-rc4

Description:

/etc/rc.d/S19dnsmasq tries to mount /dev/log in jail but fails because /dev/log doesn't exist yet because /etc/rc.d/S50syslog-ng hasn't been run yet. The end result is that dnsmasq is not running:

```
~# ps aux | grep '[d]nsmasq'
~#
```

The issue can be easily reproduced even without syslog-ng installed:

```
~# mv /dev/log /dev/log.bak
~# /sbin/ujail -t 5 -n test-ujail -l -- /bin/echo hi; echo $?
jail: stat(/dev/log) failed: No such file or directory
jail: mount_all() failed
jail: failed to build jail fs
jail: jail (9738) exited with exit: 1
1
```

I wonder if other users of OpenWRT also have this problem? I haven't seen any bug reports but it looks quite serious so I'm wondering. Would it be a bad idea to start syslog-ng earlier, S12 as with logd should be ok, in my test even S18 (so one less that S19 for dnsmasq) was enough?

brada4, 2023-05-09:
In principle yes, it conflicts logd on `logread` file, thus can take same place easily.

ardrabczyk, 2023-05-10:
I'll send a PR and see what [BKPepe](https://github.com/BKPepe) thinks

brada4, 2023-05-10:
I think there is no further benefit of this issue, you got additional backing point for your concept and can add to your PR.