openwrt/packages #20976: openconnect - build failing on aarch64_cortex-a53

greem, 2023-05-10:
Maintainer: @rsalvaterra (last specific committer!)
Environment: aarch64_cortex-a53, Dynalink DL-WRX36, SNAPSHOT

Description:

openconnect package is not in the available packages list. Checking [build logs](https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/openconnect/compile.txt) reveals:

```

Applying ./patches/001-Use-OpenSSL_version-not-deprecated-SSLeay_version.patch using plaintext: 
patching file openssl.c
configure: loading site script /builder/shared-workdir/build/sdk/include/site/aarch64
checking pkg-config is at least version 0.9.0... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... aarch64-openwrt-linux-gnu
checking whether to enable maintainer-specific portions of Makefiles... yes
checking for a BSD-compatible install... /builder/shared-workdir/build/sdk/staging_dir/host/bin/install -c
checking whether build environment is sane... yes
checking for aarch64-openwrt-linux-strip... aarch64-openwrt-linux-musl-strip
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking whether make supports nested variables... (cached) yes
checking for aarch64-openwrt-linux-gcc... aarch64-openwrt-linux-musl-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... yes
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether aarch64-openwrt-linux-musl-gcc accepts -g... yes
checking for aarch64-openwrt-linux-musl-gcc option to accept ISO C89... none needed
checking whether aarch64-openwrt-linux-musl-gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of aarch64-openwrt-linux-musl-gcc... none
checking for aarch64-openwrt-linux-musl-gcc option to accept ISO C99... none needed
configure: Applying feature macros for GNU build
checking for xdg-open... checking for xdg-open... no
checking for fdevname_r... no
checking for statfs... yes
checking for getline... yes
checking for strcasestr... yes
checking for strndup... yes
checking for strchrnul... yes
checking for asprintf... yes
checking for vasprintf... yes
checking for __builtin_clz... yes
checking for supported compiler flags...  -Wall -Wextra -Wno-missing-field-initializers -Wno-sign-compare -Wno-unused-parameter -Werror=pointer-to-int-cast -Werror-implicit-function-declaration -Wformat-nonliteral -Wformat-security -Winit-self -Wmissing-declarations -Wmissing-include-dirs -Wnested-externs -Wpointer-arith -Wwrite-strings
checking For memset_s... no
checking for explicit_memset... no
checking for explicit_bzero... yes
checking For localtime_r... yes
checking for socket... yes
checking for inet_aton... yes
checking for IPV6_PATHMTU socket option... yes
checking for __android_log_vprint... no
checking for __android_log_vprint in -llog... no
checking for nl_langinfo... yes
checking for posix_spawn... yes
checking for ld... aarch64-openwrt-linux-musl-ld
checking if the linker (aarch64-openwrt-linux-musl-ld) is GNU ld... yes
checking for shared library run path origin... done
checking 32-bit host C ABI... no
checking for ELF binary format... no
checking for the common suffixes of directories in the library search path... lib,lib,lib
checking for iconv... yes
[... 228 lines cut ...]
  CCLD     lzstest
  CC       seqtest.o
  CCLD     seqtest
  CC       buftest.o
  CCLD     buftest
  CC       bad_dtls_test-bad_dtls_test.o
  CCLD     bad_dtls_test
  CC       serverhash.o
  CCLD     serverhash
make[6]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/openconnect-9.01/tests'
make[6]: Entering directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/openconnect-9.01'
  CC       openconnect-xml.o
  CC       openconnect-main.o
  CCLD     openconnect
make[6]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/openconnect-9.01'
make[5]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/openconnect-9.01'
make[4]: Leaving directory '/builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/openconnect-9.01'
Package openconnect is missing dependencies for the following libraries:
libcrypto.so.3
libp11-kit.so.0
libp11.so.3
libssl.so.3
make[3]: *** [Makefile:104: /builder/shared-workdir/build/sdk/bin/packages/aarch64_cortex-a53/packages/openconnect_9.01-1_aarch64_cortex-a53.ipk] Error 1
time: package/feeds/packages/openconnect/compile#56.72#11.78#105.43
```

oskarirauta, 2023-05-20:
Seems that package should depend on libcrypto and openssl as they are linked with it..

greem, 2023-05-22:
> Seems that package should depend on libcrypto and openssl as they are linked with it..

Helpful, thanks. Not that I can fix it right now but that's good info.

nmav, 2024-05-09:
This looks like a local issue and there is no indication of which version of openwrt is that.