openwrt/packages #20977: borgbackup: package is currently missing in the default openwrt repos

asdil12, 2023-05-10:
https://github.com/openwrt/packages/blob/master/lang/python/borgbackup/Makefile
Maintainer: Julien Malik <julien.malik@paraiso.me> \ @julienmalik
Environment: bananapi,bpi-r3

Description:

The package is missing from the default repos currently:

```
oot@saphira2:~# opkg update
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/kmods/5.15.110-1-70043e90e0be563296616d2f36af8a67/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/kmods/5.15.110-1-70043e90e0be563296616d2f36af8a67/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /var/opkg-lists/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
root@saphira2:~# opkg list|grep borg
root@saphira2:~#

jefferyto, 2023-05-10:
The borgbackup package depends on python3-packaging. python3-packaging was marked BROKEN in #20657, and so any package that depends on it (like borgbackup) are not built.

python3-packaging was "fixed" in #20801, so both it and borgbackup should be built and available to be installed soon.

jefferyto, 2023-05-28:
Package is building successfully for aarch64_cortex-a53.

closed 2023-05-28 by jefferyto