openwrt/packages #20993: xtables-addons build failed

ptpt52, 2023-05-12:
since https://github.com/openwrt/openwrt/pull/12545 merged
```
  CC [M]  /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/xtables-addons-3.21/extensions/LUA/lua/lcode.o
  CC [M]  /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/xtables-addons-3.21/extensions/LUA/lua/ldebug.o
In file included from ./include/linux/string.h:9,
                 from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/xtables-addons-3.21/extensions/LUA/lua/include/string.h:1,
                 from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/xtables-addons-3.21/extensions/LUA/lua/ldebug.c:13:
./include/linux/stdarg.h:6: error: "va_start" redefined [-Werror]
    6 | #define va_start(v, l)  __builtin_va_start(v, l)
      | 
In file included from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/xtables-addons-3.21/extensions/LUA/lua/ldebug.c:8:
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/include/stdarg.h:47: note: this is the location of the previous definition
   47 | #define va_start(v,l)   __builtin_va_start(v,l)
      | 
./include/linux/stdarg.h:8: error: "va_arg" redefined [-Werror]
    8 | #define va_arg(v, T)    __builtin_va_arg(v, T)
      | 
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/include/stdarg.h:49: note: this is the location of the previous definition
   49 | #define va_arg(v,l)     __builtin_va_arg(v,l)
      | 
./include/linux/stdarg.h:9: error: "va_copy" redefined [-Werror]
    9 | #define va_copy(d, s)   __builtin_va_copy(d, s)
      | 
/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.2.0/include/stdarg.h:52: note: this is the location of the previous definition
   52 | #define va_copy(d,s)    __builtin_va_copy(d,s)
      | 
cc1: all warnings being treated as errors
scripts/Makefile.build:289: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/xtables-addons-3.21/extensions/LUA/lua/ldebug.o' failed
```

ptpt52, 2023-05-12:
ping @AuthorReflex @neheb 
any help to make this build failure?

Ansuel, 2023-05-12:
@ptpt52 i found i very similar problem... 

it's either we need to fix your include in toolchain to match (so the redefine gets automagically muted since nothing is redefined)

or add and ifdef...

in practice toolchain include for stdarg is mixing with the kernel one resulting in this redefine error.

@mpratt14 might help in sorting this mess..

Current fix i used was https://github.com/openwrt/openwrt/blob/3ccdc15c6ab14df898d9f36f0d7fa752ac1c15d6/package/kernel/broadcom-wl/patches/919-fix-compilation-warning-for-stack-limit.patch#L294

Ansuel, 2023-05-13:
Any news for a modified patch?

ptpt52, 2023-05-13:
@Ansuel could you help to create a new one since I have no time for now.

pprindeville, 2023-05-13:
There's no handling of `kmalloc()` returning `NULL`.

Ansuel, 2023-05-13:
@pprindeville already pointed out, i'm preparing a patch to address my comments

pprindeville, 2023-05-13:
BTW, could we (re)generate `Mbuild` dynamically based on what modules are selected?

closed 2023-05-14 by Ansuel