openwrt/packages #20994: List of software failing with GCC13 or new 5.15 kernel's werror

oskarirauta, 2023-05-12:
Since list of currently failing builds seems to be quite long, I think it's better to create unified thread with list of software that fails build:

- [ ] xtables-addons: #20993
- [ ] ttyd: #20987
- [ ] zmq: #20972
- [ ] gnutls: #20969 
- [ ] libwebsockets: #20949
- [ ] nginx-util: #20961
- [ ] pdns

ptpt52, 2023-05-12:
mark.

peterwillcn, 2023-05-12:
Seemingly the same problem
https://github.com/openwrt/openwrt/issues/12577

oskarirauta, 2023-05-18:
@neheb @byoda @blogic @miska @jmarcet @vk496 @svlobanov @thess @reinerh @dangowrt 
ping

oskarirauta, 2023-05-18:
@pprindeville @RetEECS 
ping (perl)

oskarirauta, 2023-05-18:
@PolynomialDivision 
ping (inotify-tools #21044)

oskarirauta, 2023-05-18:
please, kindly notify here if you make a PR against issue listed here, you find a new issue related to mentioned changes, or issue was fixed.

oskarirauta, 2023-05-18:
@1715173329
ping (mariadb)

jefferyto, 2023-05-19:
> numpy: missing dependencies wheel, setuptools

numpy was fixed in #20978.

> python3-packages: missing whq file (probably exists, but something has changed and does not appear to be found anymore)

The fix for python3-packages is #20981.

> in general, everything using python poetry, seem to be failing; (at least partial) list here.
> 
> dnspython: no module named 'poetry'
> astral: no module named 'poetry'
> python-dbus-fast: no module named 'poetry'

I don't know how you got this list because:

1. They currently build, i.e. there are no faillogs for these
2. They use poetry-core, not poetry
3. I got them building in #20801

Also, none of these (all of the above) are related to GCC13, werror or musl.

oskarirauta, 2023-05-19:
@jefferyto

possible not related, but they came up around same time; as did quite long list of failing packages, I made this issue to track a longer list of issues, and as I am not a huge fan of python- I made assumption that it was related to any of mentioned issues without any facts (sorry about that) basing it purely to timing. Sorry about that.

List with dnspython, astral, etc... Came from build process and errors reported by it. Previous built worked just fine, and when I pulled openwrt + package repositories, these failed. Several commits were made to openwrt between pulls, but biggest change at that time was gcc 13 introduction.

thess, 2023-05-19:
> zile: configure.ac:89: error: expected source file, required through AC_LIBSOURCES, not found, m4 failed with exit status: 1

Zile build failure due to gnulib being built in tools and include files/directories are located in a different tree. See: https://github.com/openwrt/openwrt/commit/fe33d5ba13370027165b9acf2bd895f5b4d9a4ea

There may be others.

oskarirauta, 2023-05-19:
@jefferyto

issue still remains.

jefferyto, 2023-05-19:
> issue still remains.

Do you get the same errors when building from a clean checkout of the main repo or sdk? I assume your packages repo is up-to-date?

dangowrt, 2023-05-19:
Is [coreutils failing to build](https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/coreutils/compile.txt) maybe also related to one of the three changes?

oskarirauta, 2023-05-19:
@jefferyto with clean checkout. Never used sdk. Nearly full build with almost fullest options, always openssl if choosable, and always almost all the bells and whistles (for e.g. opted out AD from samba, but chose everything else..)
My repo has been pulled less than 6 hours ago last time..

oskarirauta, 2023-05-19:
@dangowrt I doubt. Error about acl's seems to point to somewhere else, but I wouldn't know from just the log. Needs some investigating.

greem, 2023-05-20:
I suspect that https://github.com/openwrt/packages/issues/20976 is related to this, noting also that ocserv is currently also failing to build on same arch.

1715173329, 2023-05-20:
acl was fixed already. for ocserv: #21124

oskarirauta, 2023-05-20:
@greem check my answer at #20976 - seems unlink to be related to this in any possible way. openconnect seems to link against libcrypto and openssl, but package's makefile doesn't list them as depends. Disable linking to them some way, or add them as depends and your package completes..

1715173329, 2023-05-22:
>   File "/usr/src/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.10/importlib/__init__.py", line 126, in import_module
>     return _bootstrap._gcd_import(name[level:], package, level)
>   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
>   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
>   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 992, in _find_and_load_unlocked
>   File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
>   File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
>   File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
>   File "<frozen importlib._bootstrap>", line 1004, in _find_and_load_unlocked
> ModuleNotFoundError: No module named 'poetry'
> 
> ERROR Backend 'poetry.core.masonry.api' is not available.

now i got the same error on my buildbot (ubuntu 18.04/20.04 amd64)
for me it's python-pkgconfig

compiling with openwrt sdk worked fine, but when i built everything from source it just failed

oskarirauta, 2023-05-22:
@jefferyto 

numpy is **not** fixed. Issue still remains.

```
INFO: x86_64-openwrt-linux-musl-gcc: numpy/core/src/common/ufunc_override.c
cc1: note: someone does not honour COPTS correctly, passed 4 times
In file included from numpy/core/src/umath/loops.c.src:53:
numpy/core/src/umath/simd.inc.src: In function 'AVX512F_conjugate_CFLOAT':
numpy/core/src/umath/simd.inc.src:1119:13: warning: AVX512F vector return without AVX512F enabled changes the ABI [-Wpsabi]
 1119 |     @vindextype@ vindex = @vindexload@((@vindextype@*)index_ip1);
      |             ^~~~~~
numpy/core/src/umath/simd.inc.src: In function 'AVX512F_conjugate_CDOUBLE':
numpy/core/src/umath/simd.inc.src:1119:13: warning: AVX vector return without AVX enabled changes the ABI [-Wpsabi]
 1119 |     @vindextype@ vindex = @vindexload@((@vindextype@*)index_ip1);
      |             ^~~~~~
INFO: x86_64-openwrt-linux-musl-gcc: numpy/core/src/common/numpyos.c
INFO: x86_64-openwrt-linux-musl-gcc: numpy/core/src/common/npy_cpu_features.c
cc1: note: someone does not honour COPTS correctly, passed 4 times
cc1: note: someone does not honour COPTS correctly, passed 4 times
In file included from /usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/include/immintrin.h:53,
                 from numpy/core/src/umath/simd.inc.src:25:
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/include/avx512fintrin.h: In function 'AVX512F_square_CFLOAT':
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/include/avx512fintrin.h:314:1: error: inlining failed in call to 'always_inline' '_mm512_setzero_ps': target specific option mismatch
  314 | _mm512_setzero_ps (void)
      | ^~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1121:20: note: called from here
 1121 |     @vtype@ zeros = _mm512_setzero_@vsuffix@();
      |                    ^~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:740:1: error: inlining failed in call to 'always_inline' 'avx512_get_full_load_mask_ps': target specific option mismatch
  740 | avx512_get_full_load_mask_ps(void)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1120:27: note: called from here
 1120 |     @mask@ load_mask = avx512_get_full_load_mask_@vsuffix@();
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/include/avx512fintrin.h:6499:1: error: inlining failed in call to 'always_inline' '_mm512_loadu_si512': target specific option mismatch
 6499 | _mm512_loadu_si512 (void const *__P)
      | ^~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1119:22: note: called from here
 1119 |     @vindextype@ vindex = @vindexload@((@vindextype@*)index_ip1);
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:752:1: error: inlining failed in call to 'always_inline' 'avx512_get_partial_load_mask_ps': target specific option mismatch
  752 | avx512_get_partial_load_mask_ps(const npy_int num_elem, const npy_int total_elem)
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1125:25: note: called from here
 1125 |             load_mask = avx512_get_partial_load_mask_@vsuffix@(
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
 1126 |                                     num_remaining_elements, @num_lanes@);
      |                                     ~~~~~~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:782:1: error: inlining failed in call to 'always_inline' 'avx512_masked_load_ps': target specific option mismatch
  782 | avx512_masked_load_ps(__mmask16 mask, npy_float* addr)
      | ^~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1130:18: note: called from here
 1130 |             x1 = avx512_masked_load_@vsuffix@(load_mask, ip);
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:764:1: error: inlining failed in call to 'always_inline' 'avx512_masked_gather_ps': target specific option mismatch
  764 | avx512_masked_gather_ps(__m512 src,
      | ^~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1133:19: note: called from here
 1133 |             x1  = avx512_masked_gather_@vsuffix@(zeros, ip, vindex, load_mask);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/src/openwrt/staging_dir/toolchain-x86_64_gcc-13.1.0_musl/lib/gcc/x86_64-openwrt-linux-musl/13.1.0/include/avx512fintrin.h:6373:1: error: inlining failed in call to 'always_inline' '_mm512_mask_storeu_ps': target specific option mismatch
 6373 | _mm512_mask_storeu_ps (void *__P, __mmask16 __U, __m512 __A)
      | ^~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:1138:9: note: called from here
 1138 |         _mm512_mask_storeu_@vsuffix@(op, load_mask, out);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
numpy/core/src/umath/simd.inc.src:952:1: error: inlining failed in call to 'always_inline' 'avx512_csquare_ps': target specific option mismatch
  952 | avx512_csquare_@vsub@(@vtype@ x)
      | ^~~~~~~~~~~~~~~~~
[... 75 lines cut ...]
  Architecture: x64
  Compiler    : gcc

CPU baseline  : 
  Requested   : 'min'
  Enabled     : SSE SSE2 SSE3
  Flags       : -msse -msse2 -msse3
  Extra checks: none

CPU dispatch  : 
  Requested   : 'max -xop -fma4'
[... cut ...]
make[2]: *** [Makefile:88: /usr/src/openwrt/build_dir/target-x86_64_musl/pypi/numpy-1.23.3/.built] Error 1
make[2]: Leaving directory '/usr/src/openwrt/feeds/packages/lang/python/numpy'
time: package/feeds/packages/numpy/compile#192.81#19.22#0.00
    ERROR: package/feeds/packages/numpy failed to build.
make[1]: *** [package/Makefile:120: package/feeds/packages/numpy/compile] Error 1
make[1]: Leaving directory '/usr/src/openwrt'
make: *** [/usr/src/openwrt/include/toplevel.mk:231: package/numpy/compile] Error 2

```

Divadeer, 2023-05-22:
@jefferyto Yesterday I was having the poetry issue in building dns-python (using gcc 12 though). Just did a git pull and feeds update and this package is compiling without error now.

jefferyto, 2023-05-22:
> numpy is **not** fixed. Issue still remains.

Perhaps it would help to ping the maintainer of numpy: @commodo

commodo, 2023-05-23:
> > numpy is **not** fixed. Issue still remains.
> 
> Perhaps it would help to ping the maintainer of numpy: @commodo

will check it out;
the issue with numpy and x86, is that it often tries/assumes some CPU extensions we don't have on the OpenWrt x86[_64] compilers;

oskarirauta, 2023-05-23:
@commodo
Issue wasn't there before.....

jefferyto, 2023-05-23:
The fix for the poetry-related issues is https://github.com/openwrt/openwrt/pull/12710.

The reason why the error message is `ModuleNotFoundError: No module named 'poetry'` is because poetry-core uses itself to build, and buildroot's .gitignore causes it to exclude all of its own code, so it doesn't get installed properly.

commodo, 2023-05-24:
> @commodo Issue wasn't there before.....

depends on how you define `wasn't there before`;
i would get it (locally) with a few versions of numpy, which is why i don't update it too often;

feckert, 2023-05-24:
@oskarirauta the packages netavark and aardvark-dns do not build anymore. I think this is related to musl 1.2.4 update.
I already tried to fix this with `TARGET_CFLAGS += -D_LARGEFILE64_SOURCE` because of undefined refrence to `<name>64` functions but it does not fix the build.
I got the following error log (shortened):
```

= note: /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/hostpkg/cargo/lib/rustlib/x86_64-unknown-linux-musl/lib/libstd-e3430e5601b9f1d2.rlib(std-e3430e5601b9f1d2.std.e4d40498-cgu.12.rcgu.o): in function `std::sys::common::small_c_string::run_with_cstr_allocating':
          std.e4d40498-cgu.12:(.text.unlikely._ZN3std3sys6common14small_c_string24run_with_cstr_allocating17ha36da161e1c6bae9E+0x98): undefined reference to `stat64'
          /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/hostpkg/cargo/lib/rustlib/x86_64-unknown-linux-musl/lib/libstd-e3430e5601b9f1d2.rlib(std-e3430e5601b9f1d2.std.e4d40498-cgu.6.rcgu.o): in function `std::backtrace_rs::symbolize::gimli::mmap':
          std.e4d40498-cgu.6:(.text._ZN3std12backtrace_rs9symbolize5gimli4mmap17h7c64f37af86109f2E+0x171): undefined reference to `fstat64'
          /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/hostpkg/cargo/lib/rustlib/x86_64-unknown-linux-musl/lib/libstd-e3430e5601b9f1d2.rlib(std-e3430e5601b9f1d2.std.e4d40498-cgu.8.rcgu.o): in function `<std::fs::File as std::io::Read>::read_to_string':
          std.e4d40498-cgu.8:(.text._ZN47_$LT$std..fs..File$u20$as$u20$std..io..Read$GT$14read_to_string17h8b4d344335b86156E+0x53): undefined reference to `fstat64'
          /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: std.e4d40498-cgu.8:(.text._ZN47_$LT$std..fs..File$u20$as$u20$std..io..Read$GT$14read_to_string17h8b4d344335b86156E+0x76): undefined reference to `lseek64'
          /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/hostpkg/cargo/lib/rustlib/x86_64-unknown-linux-musl/lib/libstd-e3430e5601b9f1d2.rlib(std-e3430e5601b9f1d2.std.e4d40498-cgu.8.rcgu.o): in function `std::fs::metadata':
          std.e4d40498-cgu.8:(.text._ZN3std2fs8metadata17h0ac6f29dff1ff35eE+0xba): undefined reference to `stat64'
          /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/hostpkg/cargo/lib/rustlib/x86_64-unknown-linux-musl/lib/libstd-e3430e5601b9f1d2.rlib(std-e3430e5601b9f1d2.std.e4d40498-cgu.13.rcgu.o): in function `std::sys::unix::fs::File::open_c':
          std.e4d40498-cgu.13:(.text._ZN3std3sys4unix2fs4File6open_c17h8d9f66cb467887dcE+0xf5): undefined reference to `open64'
          /home/feckert/workspace/openwrt/LDM-owrt-23.05/build/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: std.e4d40498-cgu.13:(.text._ZN3std3sys4unix2fs4File6open_c17h8d9f66cb467887dcE+0x10c): undefined reference to `open64'
          collect2: error: ld returned 1 exit status

  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `netavark` due to previous error
```
If you need the whole log I could paste this but this is quit long!

1715173329, 2023-05-24:
@feckert unfortunately it needs to be fixed on rust side, see https://github.com/rust-lang/rust/issues/111285

oskarirauta, 2023-05-24:
There is a work-around- on the OpenWRT side there is a draft PR from me that provides a compatibility on musl 1.2.4 which overcomes all musl 1.2.4 related issues for those who need for e.g. Rust - and need it **now**..

commodo, 2023-05-26:
this may fix numpy:  https://github.com/openwrt/packages/pull/21184

works for me;
i tried x86 & x86_64

oskarirauta, 2023-05-26:
> this may fix numpy: #21184
> 
> works for me; i tried x86 & x86_64

#21184 fixes numpy

jefferyto, 2023-05-26:
What is the purpose of this ticket? It doesn't actually track issues related to GCC13 or 5.15 kernel werror or updated musl (btw that is three groups of problems unrelated to each other), this is a list of every build error encountered by one person.

Also "fixing all issues related to GCC13 or werror or musl" isn't a meaningful milestone that needs to be tracked and progress kept. These are just issues that should be resolved, just like every other issue in the issue tracker.

It would be more productive to open individual issues to track each build error.

oskarirauta, 2023-05-26:
@Habbie 

ping (pdns)

Habbie, 2023-05-26:
Thanks!

I see your patched tree still has 4.7.3, while openwrt/packages master has 4.7.4, which contains the same patch. I think you can check pdns off your list!

oskarirauta, 2023-05-26:
@Habbie  Great. Good work :)

theAeon, 2023-05-30:
seems libjson-c has some missing includes and some warnings when compiled on fedora 38

ArisAachen, 2023-08-22:
I got an error in arch linux 
``` bash
gcc --version
gcc (GCC) 13.2.1 20230801
```

``` log
checking for qsort_r... yes
checking for GNU-style qsort_r... no
checking for BSD-style qsort_r... no
configure: error: in `/home/aris/Desktop/OpenSource/lede-mt7981/build_dir/target-aarch64_cortex-a53_musl/strongswan-5.9.5':
configure: error: qsort_r has unknown semantics
```

pprindeville, 2023-11-04:
> @pprindeville @RetEECS
> ping (perl)

Do you have a link to a failing build I can look at?

AuthorReflex, 2024-04-03:
apcupsd fix: https://github.com/openwrt/packages/pull/23804

Neustradamus, 2024-10-27:
To follow this important ticket.

neheb, 2024-10-27:
Closing this. The logs on downloads.openwrt.org have GCC13 failures.

closed 2024-10-27 by neheb