openwrt/packages #20999: nginx-full not working properly

zcy85611, 2023-05-12:
after sovle the compile issue, now nginx-full can be compiled

but it fails to start because missing "/etc/nginx/uwsgi_params", add the file from old compile, nginx starts

Thu May 11 01:13:20 2023 daemon.err nginx_init: 2023/05/11 01:13:20 [emerg] 4746#0: open() "/etc/nginx/uwsgi_params" failed (2: No such file or directory) in /etc/nginx/conf.d/luci.locations:3
Thu May 11 01:13:20 2023 daemon.err nginx_init: nginx: configuration file /etc/nginx/uci.conf test failed
Thu May 11 01:13:20 2023 daemon.err nginx_init: NOT using conf file!

and after add webdav configuration, even with "nginx-mod-dav-ext" installed, it still tells "unknown directive "dav_ext_lock_zone"

it seems that nginx-mod-dav-ext is not working at all

Thu May 11 01:12:30 2023 daemon.err nginx_init: 2023/05/11 01:12:29 [emerg] 4699#0: unknown directive "dav_ext_lock_zone" in /etc/nginx/conf.d/webdav.conf:1
Thu May 11 01:12:30 2023 daemon.err nginx_init: nginx: configuration file /etc/nginx/uci.conf test failed
Thu May 11 01:12:30 2023 daemon.err nginx_init: NOT using conf file!

webdav configuration file works on old compile

could you help to look into this please? thanks

Ansuel, 2023-05-12:
you need to load the module for dav ext to correctly work.

for uwsgi i'm checking

zcy85611, 2023-05-12:
> you need to load the module for dav ext to correctly work.
> 
> for uwsgi i'm checking

may I ask how to load the module? and is it possible to do it automaticly like the old ones?

Ansuel, 2023-05-12:
nginx documentation, there is nothing special here actually we are aligning on how things works in the rest of the world (debian, ubuntu)

zcy85611, 2023-05-12:
> nginx documentation, there is nothing special here actually we are aligning on how things works in the rest of the world (debian, ubuntu)

ok, I know how to load module now, thanks for the help

Ansuel, 2023-05-12:
@zcy85611 Nice! And sorry for the breakage... it's something we had to do for a long time so these regression are saddly needed.

zcy85611, 2023-05-12:
> @zcy85611 Nice! And sorry for the breakage... it's something we had to do for a long time so these regression are saddly needed.

it's okay, thanks for keeping improving the codes

sjkhsl, 2023-06-13:
> @zcy85611 Nice! And sorry for the breakage... it's something we had to do for a long time so these regression are saddly needed.

23.05 After the nginx is built, there are still problems，23.05 is still missing two patches commit 7b91373 commit f5d7a26，It is normal in master。

Ansuel, 2023-06-13:
@sjkhsl please wait the new version of nginx to be built... also those 2 commit are not needed

alex3d, 2023-08-31:
`uwsgi_params` and some other config files are missing in`nginx-full` package due to typo in Makefile.
This should be fixed when https://github.com/openwrt/packages/pull/21985 gets merged.

closed 2023-09-01 by zcy85611