openwrt/packages #21006: xtables-addons: xt_LUA uses variable-length arrays

pprindeville, 2023-05-13:
Maintainer: @jow- 
Environment: x86_64, generic, HEAD

Description:

The kernel flags include:

```
KBUILD_CFLAGS += -Wvla
```

which causes this line:

```
char res[255 + (array->length * 3)]; /* make sure the buffer is big enough*/
```

to fail compilation in `xtables-addons/extensions/LUA/byte_array.c`:

```
/home/pprindeville/work/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/byte_array.c: In function 'byte_array_to_string':
/home/pprindeville/work/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/byte_array.c:110:9: error: ISO C90 forbids variable length array 'buf' [-Werror=vla]
  110 |         uint8_t buf[(array->length * 3) + 255];
      |         ^~~~~~~
/home/pprindeville/work/openwrt/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.21/extensions/LUA/byte_array.c:112:9: error: ISO C90 forbids variable length array 'res' [-Werror=vla]
  112 |         char res[255 + (array->length * 3)]; /* make sure the buffer is big enough*/
      |         ^~~~
cc1: all warnings being treated as errors
```

This should be re-written to use `kmalloc()` instead as kernel modules do.

Ansuel, 2023-05-13:
We already have a fix in theory there is also the va args problem that results in a not so clean fix

pprindeville, 2023-05-13:
Also, deselecting `CONFIG_PACKAGE_kmod-ipt-lua` doesn't cause the module to not be built, as it's hardcoded into `xtables-addons/Mbuild`.

pprindeville, 2023-05-13:
> We already have a fix in theory there is also the va args problem that results in a not so clean fix

Good to know.  When did this creep in?  I did a build a couple of weeks ago after rebasing to `HEAD` and didn't see this.

Ansuel, 2023-05-13:
> > We already have a fix in theory there is also the va args problem that results in a not so clean fix
> 
> Good to know. When did this creep in? I did a build a couple of weeks ago after rebasing to `HEAD` and didn't see this.

warning was there from ages... just we are now using WERROR kernel config flag so any warning from anything related to kernel result in compilation error...

pprindeville, 2023-05-13:
I'm a big fan of `-Wall -Werror` so I'm not going to complain about progress...

Ansuel, 2023-05-13:
@pprindeville i tried to fix everything by using CI but some slipped out due to very specific config, and feeds not tested

pprindeville, 2023-05-13:
Yeah, I've long since thought we should do an `x86_64` build of the kitchen sink (everything selected)... better chance to catch these sorts of things.

Ansuel, 2023-05-13:
@pprindeville btw i have some concern on how the LUA module is designed do you have something that would actually use xt_LUA?

Ansuel, 2023-05-13:
Ok no... thank god my concern is not true... ok good things can be fixed!

pprindeville, 2023-05-13:
No, I don't even use LUA.

closed 2023-05-14 by Ansuel