openwrt/packages #21013: perl compile with gcc 12.3 failing compile opmini

rmandrad, 2023-05-14:
Description: Perl compile of opmini.c fails with "'off64_t'; did you mean 'off_t'"

since the upgrade of gcc to 12.3 i am getting the below error when compiling perl 

environment - aarch64 ipq807x
```
@aarch64-openwrt-linux-musl-gcc -c -DPERL_CORE -D_REENTRANT -D_GNU_SOURCE -fwrapv -fno-strict-aliasing -pipe -fstack-protector-strong -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -I/home/rmandrad/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/usr/include -I/home/rmandrad/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include/fortify -I/home/rmandrad/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.2.0_musl/include -O2 -Wall -fPIC -DPERL_IS_MINIPERL -DPERL_EXTERNAL_GLOB opmini.c
In file included from op.c:163:
perl.h:2494:22: error: unknown type name 'off64_t'; did you mean 'off_t'?
 2494 | #       define Off_t off64_t
      |                      ^~~~~~~
make[3]: *** [Makefile:304: opmini.o] Error 1
```

rmandrad, 2023-05-14:
i reverted 8d2ab0fac6 (the bump to gcc 12.3) and it now compiles (using git rebase --rebase-merges --onto 8d2ab0fac6^ 8d2ab0fac6^)

rmandrad, 2023-05-14:
note i noticed xtables-addon also failing due to the upgrade

robimarko, 2023-05-14:
Did you maybe include the musl 1.2.4 update as well?

Cause, for me its failing with musl updated with the exact fault due to:

> On the API level, the legacy "LFS64" ("large file support") interfaces, which were provided by macros remapping them to their standard names (#define stat64 stat and similar) have been deprecated and are no longer provided under the _GNU_SOURCE feature profile, only under explicit _LARGEFILE64_SOURCE. The latter will also be removed in a future version. Builds broken by this change can be fixed short-term by adding -D_LARGEFILE64_SOURCE to CFLAGS, but should be fixed to use the standard interfaces.

Adding `-D_LARGEFILE64_SOURCE` to `CFLAGS` fixes it

closed 2023-05-14 by rmandrad

rmandrad, 2023-05-15:
closing this issue.  added -D_LARGEFILE64_SOURCE to the perl Makefile.