openwrt/packages #21015: frr: does not compile with gcc 13

PolynomialDivision, 2023-05-14:
Maintainer: @lucize
Environment: mt7622

Error:
```
make[2]: Entering directory '/home/nick/openwrt/feeds/packages/net/frr'
rm -f /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80/.built
touch /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80/.built_check
CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80=frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80=frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -flto=auto -fno-fat-lto-objects  -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " LDFLAGS="-L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -flto=auto -fuse-linker-plugin -znow -zrelro -Wl,--gc-sections,--as-needed -latomic " make  -C /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80=frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[3]: Entering directory '/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80'
Makefile:4393: WARNING: you have overwritten the "CC" variable on the make command line.
Makefile:4405: WARNING: you have overwritten the "LD" variable on the make command line.
Makefile:4412: ------
Makefile:4413: While overwriting these variables works most of the time, it is not recommended and can cause confusing build errors.
Makefile:4414: This is especially problematic when cross-compiling, since tools that run on the build system during the build process will not be compiled correctly.
Makefile:4415: All of these variables should be supplied to 'configure', and they will be remembered and correctly applied during 'make'.
Makefile:4416: ------
true
make  all-am
make[4]: Entering directory '/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80'
Makefile:4393: WARNING: you have overwritten the "CC" variable on the make command line.
Makefile:4405: WARNING: you have overwritten the "LD" variable on the make command line.
Makefile:4412: ------
Makefile:4413: While overwriting these variables works most of the time, it is not recommended and can cause confusing build errors.
Makefile:4414: This is especially problematic when cross-compiling, since tools that run on the build system during the build process will not be compiled correctly.
Makefile:4415: All of these variables should be supplied to 'configure', and they will be remembered and correctly applied during 'make'.
Makefile:4416: ------
/bin/sh ./libtool  --tag=CC   --mode=link aarch64-openwrt-linux-musl-gcc -fms-extensions -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers  -I/home/nick/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include      -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80=frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -export-dynamic     -L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -flto=auto -fuse-linker-plugin -znow -zrelro -Wl,--gc-sections,--as-needed -latomic  -g -o lib/grammar_sandbox lib/grammar_sandbox_main.o lib/libfrr.la -lresolv   -ljson-c -lrt
OpenWrt-libtool: link: aarch64-openwrt-linux-musl-gcc -fms-extensions -fno-omit-frame-pointer -funwind-tables -Wall -Wextra -Wstrict-prototypes -Wmissing-prototypes -Wmissing-declarations -Wpointer-arith -Wbad-function-cast -Wwrite-strings -Wundef -Wno-unused-result -Wno-unused-parameter -Wno-missing-field-initializers -I/home/nick/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/include -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80=frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80 -flto=auto -fno-fat-lto-objects -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z -Wl,now -Wl,-z -Wl,relro -flto=auto -fuse-linker-plugin -znow -zrelro -Wl,--gc-sections -Wl,--as-needed -g -o lib/.libs/grammar_sandbox lib/grammar_sandbox_main.o -Wl,--export-dynamic  -L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib lib/.libs/libfrr.so -L/home/nick/openwrt/staging_dir/target-aarch64_cortex-a53_musl/usr/lib /home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib/libatomic.so -lcap -lyang -lcrypt -lresolv -ljson-c -lrt -pthread -Wl,-rpath -Wl,/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib
/home/nick/openwrt/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/12.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: lib/.libs/libfrr.so: undefined reference to `vty_pass_fd'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:8010: lib/grammar_sandbox] Error 1
make[4]: Leaving directory '/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80'
make[3]: *** [Makefile:6371: all] Error 2
make[3]: Leaving directory '/home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80'
make[2]: *** [Makefile:246: /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/frr-410d78b7a4f2c8b4d4c3736c6d6cdb710287ca80/.built] Error 2
make[2]: Leaving directory '/home/nick/openwrt/feeds/packages/net/frr'
time: package/feeds/packages/frr/compile#0.35#0.06#0.39
    ERROR: package/feeds/packages/frr failed to build.
make[1]: *** [package/Makefile:120: package/feeds/packages/frr/compile] Error 1
make[1]: Leaving directory '/home/nick/openwrt'
make: *** [/home/nick/openwrt/include/toplevel.mk:231: package/frr/compile] Error 2

```

lucize, 2023-05-15:
this will work in the upcoming 8.5.1 update that I'll propose a bit later

closed 2023-05-16 by PolynomialDivision

h9j6k, 2024-11-10:
Hello, 

I am still having this issue `vty_pass_fd` undefined reference when cross compiling for ipq806x on debian 12. I even tried frr 8.5.6, same error. Any ideas? Thanks.

```
/home/op_builder/dev/compileOpenwrt/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-12.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: lib/.libs/libfrr.so: undefined reference to `vty_pass_fd'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:7894: lib/grammar_sandbox] Error 1
make[4]: *** Waiting for unfinished jobs....
/home/op_builder/dev/compileOpenwrt/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-12.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: lib/.libs/libfrr.so: undefined reference to `vty_pass_fd'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:8936: tools/gen_yang_deviations] Error 1
/home/op_builder/dev/compileOpenwrt/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-12.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: lib/.libs/libfrr.so: undefined reference to `vty_pass_fd'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:8942: tools/permutations] Error 1
/home/op_builder/dev/compileOpenwrt/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-12.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: lib/.libs/libfrr.so: undefined reference to `vty_pass_fd'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:8930: tools/gen_northbound_callbacks] Error 1
/home/op_builder/dev/compileOpenwrt/openwrt/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-12.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/12.3.0/../../../../arm-openwrt-linux-muslgnueabi/bin/ld: lib/.libs/libfrr.so: undefined reference to `vty_pass_fd'
collect2: error: ld returned 1 exit status
make[4]: *** [Makefile:7723: bgpd/bgp_btoa] Error 1
make[4]: Leaving directory '/home/op_builder/dev/compileOpenwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/frr-64d2196e59a7a304266e8a1b80c822d0d62fdc20'
make[3]: *** [Makefile:6245: all] Error 2
make[3]: Leaving directory '/home/op_builder/dev/compileOpenwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/frr-64d2196e59a7a304266e8a1b80c822d0d62fdc20'
make[2]: *** [Makefile:248: /home/op_builder/dev/compileOpenwrt/openwrt/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/frr-64d2196e59a7a304266e8a1b80c822d0d62fdc20/.built] Error 2
make[2]: Leaving directory '/home/op_builder/dev/compileOpenwrt/openwrt/feeds/packages/net/frr'
time: package/feeds/packages/frr/compile#180.25#14.43#59.37
    ERROR: package/feeds/packages/frr failed to build.
make[1]: *** [package/Makefile:129: package/feeds/packages/frr/compile] Error 1
make[1]: Leaving directory '/home/op_builder/dev/compileOpenwrt/openwrt'
make: *** [/home/op_builder/dev/compileOpenwrt/openwrt/include/toplevel.mk:233: package/frr/compile] Error 2

```

PR #21036 (2023-05-16): frr: update to 8.5.1 branch
also fixes gcc 13 compile

Maintainer: me
Tested x86
Fixes: #21015