openwrt/packages #21025: cloudflared: does not compile with gcc-13?

PolynomialDivision, 2023-05-15:
Maintainer: @1715173329 
Environment: mt7622

Does not seem like it is a gcc13 problem. However, when trying to compile it fails with:
```
Finding targets

Building targets
github.com/lucas-clemente/quic-go/internal/qtls
# github.com/lucas-clemente/quic-go/internal/qtls
../../vendor/github.com/lucas-clemente/quic-go/internal/qtls/go120.go:6:13: cannot use "The version of quic-go you're using can't be built on Go 1.20 yet. For more details, please see https://github.com/lucas-clemente/quic-go/wiki/quic-go-and-Go-versions." (untyped string constant "The version of quic-go you're using can't be built on Go 1.20 yet. F...) as int value in variable declaration

make[2]: *** [Makefile:68: /home/nick/openwrt/build_dir/target-aarch64_cortex-a53_musl/cloudflared-2023.5.0/.built] Error 1
make[2]: Leaving directory '/home/nick/openwrt/feeds/packages/net/cloudflared'
time: package/feeds/packages/cloudflared/compile#1.11#0.42#1.06
    ERROR: package/feeds/packages/cloudflared failed to build.
make[1]: *** [package/Makefile:120: package/feeds/packages/cloudflared/compile] Error 1
make[1]: Leaving directory '/home/nick/openwrt'
make: *** [/home/nick/openwrt/include/toplevel.mk:231: package/cloudflared/compile] Error 2

```

1715173329, 2023-05-15:
cloudflared still doesn't support go 1.20 yet
hope it will get fixed recently: https://github.com/cloudflare/cloudflared/pull/945

PolynomialDivision, 2023-05-15:
Sorry, should have investigated it more. I'm just compiling all packages with gcc-13 to check which packages fail. I thought cloudflared would be fine. However, same issue on buildbot:
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/cloudflared/compile.txt

I will close. ;)