openwrt/packages #21034: nfs-kernel-server build failed.

ptpt52, 2023-05-16:
```
make[7]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-mipsel_24kc_musl/nfs-utils-2.6.2/support/export'
mipsel-openwrt-linux-musl-gcc -DHAVE_CONFIG_H -I. -I../../support/include  -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-mipsel_24kc_musl/usr/include/tirpc -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/usr/include -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include/fortify -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/toolchain-mipsel_24kc_gcc-12.3.0_musl/include  -pipe  -Wall  -Wextra  -Wstrict-prototypes  -Werror=missing-prototypes  -Werror=missing-declarations  -Werror=format=2  -Werror=undef  -Werror=missing-include-dirs  -Werror=strict-aliasing=2  -Werror=init-self  -Werror=implicit-function-declaration  -Werror=return-type  -Werror=switch  -Werror=overflow  -Werror=parentheses  -Werror=aggregate-return  -Werror=unused-result  -fno-strict-aliasing  -Werror=format-overflow=2 -Werror=int-conversion -Werror=incompatible-pointer-types -Werror=misleading-indentation -Wno-cast-function-type -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/nfs-utils-2.6.2=nfs-utils-2.6.2 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -Wno-error=implicit-function-declaration -Wno-error=strict-prototypes -Wno-error=incompatible-pointer-types -Wno-error=format-security -Wno-error=undef -Wno-error=missing-include-dirs  -c -o cache.o cache.c
cache.c: In function 'uuid_by_path':
cache.c:363:25: error: storage size of 'st' isn't known
  363 |         struct statfs64 st;
      |                         ^~
cache.c:363:25: warning: unused variable 'st' [-Wunused-variable]
Makefile:487: recipe for target 'cache.o' failed
make[7]: *** [cache.o] Error 1
make[7]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-mipsel_24kc_musl/nfs-utils-2.6.2/support/export'
Makefile:416: recipe for target 'all' failed
make[6]: *** [all] Error 2
make[6]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-mipsel_24kc_musl/nfs-utils-2.6.2/support/export'
Makefile:442: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-mipsel_24kc_musl/nfs-utils-2.6.2/support'
Makefile:482: recipe for target 'all-recursive' failed
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-mipsel_24kc_musl/nfs-utils-2.6.2'
Makefile:211: recipe for target '/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/nfs-utils-2.6.2/.built' failed
make[3]: *** [/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-mipsel_24kc_musl/nfs-utils-2.6.2/.built] Error 2
make[3]: Leaving directory '/mnt/Data/Sources/openwrt/openwrt-feeds/packages/net/nfs-kernel-server'
time: package/feeds/packages/nfs-kernel-server/compile#0.16#0.13#0.26

```

1715173329, 2023-05-16:
it's got fixed in nfs-utils v2.6.3-rc6, consider creating a pr for bumping it to 2.6.3?

1715173329, 2023-05-17:
okay it still fails in 2.6.3

closed 2023-05-17 by 1715173329