openwrt/packages #21063: lcd4linux/custom - failed to build

1715173329, 2023-05-17:
Maintainer: @feckert 
Environment: snapshot/master

Description:
configure log:
https://downloads.openwrt.org/snapshots/faillogs/aarch64_cortex-a53/packages/lcd4linux/custom/compile.txt
```
checking that generated files are newer than configure... done
configure: error: conditional "ANSICXX" was never defined.
Usually this means the macro was only invoked conditionally.
make[3]: *** [Makefile:297: /builder/shared-workdir/build/sdk/build_dir/target-aarch64_cortex-a53_musl/lcd4linux-custom/lcd4linux-2019-07-19-0b83bb3a/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 1
time: package/feeds/packages/lcd4linux/custom/compile#53.72#1.99#71.04
```

feckert, 2023-05-17:
Thank you for pointing this out. Does it only break on this target? I only have a mips (lantiq_xrx200) or an x86_64 to reproduce thist. Is this related to gcc 13 update or das this haben in the current master branch build with the current default config gcc 12?

1715173329, 2023-05-17:
> Does it only break on this target? 

it fails for all architectures currently, with the same error
https://downloads.openwrt.org/snapshots/faillogs/x86_64/packages/lcd4linux/custom/compile.txt

> das this happen in the current master branch build with the current default config gcc 12?

yes, with standard configuration, gcc 12.

oskarirauta, 2023-05-18:
#20994

Check if solution for gcc 13 there will help - same error there, and a solution, even though my version is newer from stock version. If someone by the way could check custom build while at it, all drivers (dpf) didn't want to build- I made a issue about it quite a long time ago already.

feckert, 2023-05-19:
From my point of view, there is nothing to be said against it to use a maintained version of lcd4linux. I only have the following patches on top of the last master version.
I use them in my setup so I need them.
https://github.com/jmccrohan/lcd4linux/compare/trunk...feckert:lcd4linux:pr/feature
But I first have to see whether they are easy to port.
Maybe i could upstream them to the new maintained version repository.

Your changes https://github.com/oskarirauta/packages2/commit/0aaf49bc0184ac0a8a731534c9ca6bb0ae5f70c2 fixes the current build issue.

That would be a workaround for now. Of course, it is better to use a maintained version

oskarirauta, 2023-05-19:
@feckert same workaround will work on your version as well.

I didn't inspect very closely, but with a quick peak, it seemed that maintained version already had quite a lot in it, check it out - maybe they already have imported some/all your work.... But that package Makefile issue should be probably checked by someone..

Mind showing me what your lcd looks like? :)

feckert, 2023-05-23:
lcd4linux-custom build fixed with commit https://github.com/openwrt/packages/commit/917ea9d214fe4a6ef8e000c50761ac6d302f8162
I will have to look at switching to the new repository at a later date after branching openwrt-23.xx. Why drv_dpf is not built is something I am looking into it now.

closed 2023-05-23 by feckert