openwrt/packages #21085: coreutils: Missing coreutils and coreutils-related packages on snapshot aarch64_cortex-a53

elisiariocouto, 2023-05-18:
Maintainer: @httpstorm
Environment: aarch64_cortex-a53 /ARMv8 Processor rev 4, Dynalink DL-WRX36, OpenWrt SNAPSHOT r22847-5ca3386c8a

Description:

Hello. I tried to install `adblock` package but `coreutils` and `coreutils-*` packages are missing for this architecture. Maybe related to @ttocsr [commit](https://github.com/openwrt/packages/commit/1ff189e75adaab8f7d0dede816d4677994d06e9c)?

httpstorm, 2023-05-18:
@elisiariocouto 
Looking at the `opkg` URLs: `coreutils` are available under generic, but not under a53.
https://downloads.openwrt.org/snapshots/packages/aarch64_generic/packages/
https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/

I checked the commit you are using: https://github.com/openwrt/openwrt/commit/5ca3386c8a156f0feb659bacffd88306a82f5545, as well as the current master https://github.com/openwrt/openwrt/commit/c9cb6411c1a70516d5922e88c45b338de041baba: `coreutils` are available in `menuconfig`:
```
Target System (Qualcomm Atheros IPQ807x), Subtarget (Generic), Target Profile (Dynalink DL-WRX36)
LuCI, Applications, luci-app-adblock: selects PACKAGE_adblock, …
Network, adblock: selects PACKAGE_coreutils, PACKAGE_coreutils-sort, …
Utilities, coreutils, coreutils-sort: selected by PACKAGE_adblock
```

_Build on master_
I was able to compile `coreutils` and `adblock`.
Note: I have never built this target on macOS, so the failure below might also be related to that:
```
ERROR: package/kernel/qca-ssdk failed to build.
make[4]: *** No rule to make target '/Volumes/test/2023-05-18/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/build/linux/KSLIB/ref_fdb.d', needed by 'ref_fdb.o'.  Stop.
make[4]: *** No rule to make target '/Volumes/test/2023-05-18/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/build/linux/KSLIB/adpt.d', needed by 'adpt.o'.  Stop.
make[4]: *** No rule to make target '/Volumes/test/2023-05-18/openwrt/build_dir/target-aarch64_cortex-a53_musl/linux-ipq807x_generic/qca-ssdk-2022-09-12-628b22bc/build/linux/KSLIB/api_access.d', needed by 'api_access.o'.  Stop.
```

Build logs, diffconfig, and compiled `coreutils-sort` and `adblock` packages:
https://httpstorm.com/openwrt/test/2023-05-18_qca-ssdk/

@jow- @Ansuel

robimarko, 2023-05-18:
Yeah, SSDK is currently broken when `make 4.4` is used.
I managed to bisect make to commit:
https://git.savannah.gnu.org/cgit/make.git/commit/?id=9e2fa24649b870d79e2582f46e851fb34daa4762

However, as suspected it was a long-time bug that made SSDK compile but I have no clue how to fix the mess that SSDK is.

1715173329, 2023-05-19:
coreutils is failing due to broken libacl, which was fixed in 6c6d8fe5420a1fc88797f406e4483d743b87ac60

1715173329, 2023-05-20:
Available on snapshots now.

closed 2023-05-20 by 1715173329