openwrt/packages #21087: lighttpd package missing in snapshot

AddRemover, 2023-05-18:
Maintainer: @gstrauss 
Environment: arm_cortex-a9_vfpv3-d16,  Linksys WRT3200ACM, OpenWrt snapshot

Description:

[lighttpd ](https://github.com/openwrt/packages/tree/master/net/lighttpd) does not compile for the arch.

Googling[ over snapshot folder](https://www.google.com/search?q=lighttpd+site:https://downloads.openwrt.org/snapshots/packages/), just a few arch's have lighttpd compiled.

```
opkg update
opkg install lighttpd
Unknown package 'lighttpd'.
```

gstrauss, 2023-05-18:
https://openwrt.org/packages/pkgdata/lighttpd
All arches built for lighttpd 1.4.70 released to openwrt 5 days ago in #20980

Try a mirror? https://openwrt.org/downloads

AddRemover, 2023-05-18:
I took few mirrors and none of them have lighttpd compiled :(
[1](http://mirrors.linux.ro/lede/downloads/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/)
[2](http://mirror.ps.kz/openwrt/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/)
[3](http://openwrt.tetaneutral.net/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/)
[4](http://mirror.0x.sg/openwrt/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/)

xabolcs, 2023-05-18:
The [faillogs](https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/lighttpd/compile.txt) for `arm_cortex-a9_vfpv3-d16` / `lighttpd` shows:

```

Applying ./patches/020-meson-mod_webdav_min.patch using plaintext: 
patching file src/meson.build

Applying ./patches/030-mod_h2-HTTP-2-separate-mod.patch using plaintext: 
patching file src/CMakeLists.txt
patching file src/Makefile.am
patching file src/SConscript
patching file src/meson.build
The Meson build system
Version: 0.61.5
Source dir: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.70
Build dir: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.70/openwrt-build
Build type: cross build
Project name: lighttpd
Project version: 1.4.70
C compiler for the host machine: arm-openwrt-linux-muslgnueabi-gcc (gcc 12.3.0 "arm-openwrt-linux-muslgnueabi-gcc (OpenWrt GCC 12.3.0 r22900-abec62a542) 12.3.0")
C linker for the host machine: arm-openwrt-linux-muslgnueabi-gcc ld.bfd 2.40.0
C compiler for the build machine: gcc (gcc 10.2.1 "gcc (Debian 10.2.1-6) 10.2.1 20210110")
C linker for the build machine: gcc ld.bfd 2.35.2
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: arm
Host machine cpu: generic
Target machine cpu family: arm
Target machine cpu: generic
Has header "sys/inotify.h" : YES 
Has header "sys/loadavg.h" : NO 
Has header "sys/poll.h" : YES 
Has header "sys/prctl.h" : YES 
Has header "sys/procctl.h" : NO 
Has header "sys/sendfile.h" : YES 
Has header "sys/un.h" : YES 
Has header "sys/wait.h" : YES 
Has header "sys/time.h" : YES 
Has header "unistd.h" : YES 
Has header "dlfcn.h" : YES 
Has header "getopt.h" : YES 
Has header "inttypes.h" : YES 
Has header "poll.h" : YES 
Has header "pwd.h" : YES 
Has header "stdint.h" : YES 
Has header "stdlib.h" : YES 
Has header "strings.h" : YES 
Has header "syslog.h" : YES 
Has header "crypt.h" : YES 
Has header "malloc.h" : YES 
Has header "signal.h" : YES 
Has header "sys/epoll.h" : YES 
Has header "sys/event.h" : NO 
Has header "sys/mman.h" : YES 
Has header "sys/random.h" : YES 
Has header "linux/random.h" : YES 
Has header "sys/resource.h" : YES 
Has header "sys/uio.h" : YES 
Checking for function "crypt_r" : YES 
Checking for type "socklen_t" : YES 
Header <stdlib.h> has symbol "arc4random_buf" : NO 
Header <unistd.h> has symbol "chroot" : YES 
Header <unistd.h> has symbol "copy_file_range" : YES 
Header <sys/epoll.h> has symbol "epoll_ctl" : YES 
Header <strings.h> has symbol "explicit_bzero" : NO 
Header <string.h> has symbol "explicit_memset" : NO 
Header <unistd.h> has symbol "fork" : YES 
Header <sys/random.h> has symbol "getentropy" : NO 
Header <stdlib.h> has symbol "getloadavg" : YES 
Header <linux/random.h> has symbol "getrandom" : NO 
[... 52 lines cut ...]
Library ldap found: YES
Library lber found: YES
Library pam found: YES
Dependency libunwind skipped: feature with_libunwind disabled
Run-time dependency lua found: YES 5.1.5
Run-time dependency libmariadb found: YES 3.1.18
Run-time dependency libssl found: YES 3.0.8
Run-time dependency libcrypto found: YES 3.0.8
Dependency libcrypto found: YES 3.0.8 (cached)
Run-time dependency wolfssl found: YES 5.5.4
Dependency wolfssl found: YES 5.5.4 (cached)
Library mbedtls found: YES
Library mbedx509 found: YES
Library mbedcrypto found: YES
Fetching value of define "FORCE_WOLFSSL_CRYPTO" :  
Library mbedcrypto found: YES
Found CMake: /builder/shared-workdir/build/sdk/staging_dir/host/bin/cmake (3.26.3)
Run-time dependency gnutls found: NO (tried pkgconfig and cmake)

../../../../build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.70/src/meson.build:382:1: ERROR: Dependency "gnutls" not found, tried pkgconfig and cmake

A full log can be found at /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.70/openwrt-build/meson-logs/meson-log.txt
make[3]: *** [Makefile:244: /builder/shared-workdir/build/sdk/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/lighttpd-1.4.70/.configured_168fd760e1f8d9c98b752825e5327138] Error 1
time: package/feeds/packages/lighttpd/compile#7.71#3.55#14.35
```

gstrauss, 2023-05-18:
Thanks @xabolcs 

lighttpd mod_gnutls is not new in lighttpd, so the failure to build lighttpd is not due to any recent changes in lighttpd.

The failure is in gnutls:
https://downloads.openwrt.org/snapshots/faillogs/arm_cortex-a9_vfpv3-d16/packages/gnutls/compile.txt

gnutls was updated to gnutls v3.8.0 in openwrt on May 6 in #20940 
However, there were subsequent issues opened for gnutls: #20969 maybe fixed in #21007 merged 16 hours ago?  I am not sure.

This does not appear to be an issue in lighttpd, but rather an issue in an underlying dependency: gnutls
This is affecting other packages as well, e.g. samba4

AddRemover, 2023-05-19:
So, we hope when gnutls issues resolved, I'll see compiled lighttpd on snapshot builds sometime in (near) future?
Work for me. Thanks!

1715173329, 2023-05-22:
Now you should be able to install via opkg.