openwrt/packages #21095: Rust failed to build

oskarirauta, 2023-05-19:
Rust has issues with musl- missing 64bit definitions of some types and commands. My host is running with musl as well, adding largefile64 define to host_cflags and host_cxxflags did not help.

oskarirauta, 2023-05-19:
@1715173329 @lu-zero 
Ping

lu-zero, 2023-05-19:
It is related to https://blog.rust-lang.org/2023/05/09/Updating-musl-targets.html

See also https://bugs.gentoo.org/903607

lu-zero, 2023-05-19:
Either we stay with musl-1.2.3 until the update is sorted out or we'll need to carry owned bootstrap binaries for 1.2.4

lu-zero, 2023-05-19:
See:
https://github.com/rust-lang/rust/pull/107129
https://github.com/rust-lang/libc/issues/3248

oskarirauta, 2023-05-19:
> Either we stay with musl-1.2.3 until the update is sorted out or we'll need to carry owned bootstrap binaries for 1.2.4

We have already gone to 1.2.4..

oskarirauta, 2023-05-19:
On alpine they chose to patch musl instead of all packages with issues..

https://git.alpinelinux.org/aports/tree/main/musl/lfs64-2.patch

oskarirauta, 2023-05-19:
@lu-zero 

_dirty patch_ at gentoo discussion did not work. Same result.

lu-zero, 2023-05-19:
Once the musl support is rolled in upstream one low impact solution is to build a beta or a nightly snapshot that has the support so everything works until there is a stable release.

oskarirauta, 2023-05-19:
Is that available from rust-lang's site? Maybe from git...

For now, I patched my musl with alpine's patches - but that's not good solution either as it renders my system a lot different..

oskarirauta, 2023-05-19:
rust builds without issues with alpine's patches. Using them fixes all recent musl related issues, but makes your build non-standard and prevents from working with those musl related issues.

But if you are working with something else, and just have to get stuff built, rust for example - this will give you a huge head start to that.
openwrt/openwrt#12667

1715173329, 2023-06-02:
Rust 1.70.0 was released and the issue seemed fixed.

lu-zero, 2023-06-02:
If you didn't beat me already I'll update it tomorrow (as I'm also preparing related releases)

1715173329, 2023-06-02:
Aha nice waiting for your PR!

lu-zero, 2023-06-05:
https://github.com/openwrt/packages/pull/21270 should had addressed it, please confirm :)

1715173329, 2023-06-05:
hmmmm [libc 0.2.145](https://github.com/rust-lang/libc/releases/tag/0.2.145) update was not included in rust 1.70.0 ;(

1715173329, 2023-06-10:
fixed by #21296

closed 2023-06-10 by 1715173329