openwrt/packages #21137: nebula: 1.7.1 fails to build

stangri, 2023-05-22:
Seeking help from Go experts (@jefferyto).

I've updated nebula Makefile to 1.7.1 and I get the following error:
```txt
Building targets
../../cert/cert.go:5:2: package crypto/ecdh is not in GOROOT (/home/stangri/development/Builder/SDK_x86-64_22.03.5/staging_dir/hostpkg/lib/go-cross/src/crypto/ecdh)
```

Any ideas on how to overcome this?

Complete build log below:
```txt
make package/nebula/{clean,compile} -j5 V=sc
make[2]: Entering directory '/home/stangri/development/Builder/SDK_x86-64_22.03.5/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_22.03.5/scripts/config'
tmp/.config-package.in:7:warning: ignoring type redefinition of 'PACKAGE_ca-bundle' from 'bool' to 'tristate'
tmp/.config-package.in:29:warning: ignoring type redefinition of 'PACKAGE_dnsmasq' from 'bool' to 'tristate'
tmp/.config-package.in:176:warning: ignoring type redefinition of 'PACKAGE_libc' from 'bool' to 'tristate'
tmp/.config-package.in:204:warning: ignoring type redefinition of 'PACKAGE_libgcc' from 'bool' to 'tristate'
tmp/.config-package.in:325:warning: ignoring type redefinition of 'PACKAGE_libpthread' from 'bool' to 'tristate'
tmp/.config-package.in:353:warning: ignoring type redefinition of 'PACKAGE_librt' from 'bool' to 'tristate'
tmp/.config-package.in:507:warning: ignoring type redefinition of 'PACKAGE_rpcd' from 'bool' to 'tristate'
tmp/.config-package.in:524:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-file' from 'bool' to 'tristate'
tmp/.config-package.in:538:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-iwinfo' from 'bool' to 'tristate'
tmp/.config-package.in:582:warning: ignoring type redefinition of 'PACKAGE_ubus' from 'bool' to 'tristate'
tmp/.config-package.in:596:warning: ignoring type redefinition of 'PACKAGE_ubusd' from 'bool' to 'tristate'
tmp/.config-package.in:609:warning: ignoring type redefinition of 'PACKAGE_uci' from 'bool' to 'tristate'
tmp/.config-package.in:686:warning: ignoring type redefinition of 'PACKAGE_libiwinfo-data' from 'bool' to 'tristate'
tmp/.config-package.in:851:warning: ignoring type redefinition of 'PACKAGE_libiwinfo-lua' from 'bool' to 'tristate'
tmp/.config-package.in:865:warning: ignoring type redefinition of 'PACKAGE_lua' from 'bool' to 'tristate'
tmp/.config-package.in:2187:warning: ignoring type redefinition of 'PACKAGE_libwolfssl' from 'bool' to 'tristate'
tmp/.config-package.in:2296:warning: ignoring type redefinition of 'PACKAGE_libblkid' from 'bool' to 'tristate'
tmp/.config-package.in:2312:warning: ignoring type redefinition of 'PACKAGE_libblobmsg-json' from 'bool' to 'tristate'
tmp/.config-package.in:2713:warning: ignoring type redefinition of 'PACKAGE_libiwinfo' from 'bool' to 'tristate'
tmp/.config-package.in:2729:warning: ignoring type redefinition of 'PACKAGE_libjson-c' from 'bool' to 'tristate'
tmp/.config-package.in:2753:warning: ignoring type redefinition of 'PACKAGE_liblua' from 'bool' to 'tristate'
tmp/.config-package.in:2768:warning: ignoring type redefinition of 'PACKAGE_liblucihttp' from 'bool' to 'tristate'
tmp/.config-package.in:2779:warning: ignoring type redefinition of 'PACKAGE_liblucihttp-lua' from 'bool' to 'tristate'
tmp/.config-package.in:2805:warning: ignoring type redefinition of 'PACKAGE_libmnl' from 'bool' to 'tristate'
tmp/.config-package.in:2891:warning: ignoring type redefinition of 'PACKAGE_libnl-tiny' from 'bool' to 'tristate'
tmp/.config-package.in:3015:warning: ignoring type redefinition of 'PACKAGE_libsmartcols' from 'bool' to 'tristate'
tmp/.config-package.in:3097:warning: ignoring type redefinition of 'PACKAGE_libubox' from 'bool' to 'tristate'
tmp/.config-package.in:3121:warning: ignoring type redefinition of 'PACKAGE_libubus' from 'bool' to 'tristate'
tmp/.config-package.in:3133:warning: ignoring type redefinition of 'PACKAGE_libubus-lua' from 'bool' to 'tristate'
tmp/.config-package.in:3146:warning: ignoring type redefinition of 'PACKAGE_libuci' from 'bool' to 'tristate'
tmp/.config-package.in:3183:warning: ignoring type redefinition of 'PACKAGE_libuuid' from 'bool' to 'tristate'
tmp/.config-package.in:3274:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-luci' from 'bool' to 'tristate'
tmp/.config-package.in:3303:warning: ignoring type redefinition of 'PACKAGE_rpcd-mod-rrdns' from 'bool' to 'tristate'
tmp/.config-package.in:3350:warning: ignoring type redefinition of 'PACKAGE_luci' from 'bool' to 'tristate'
tmp/.config-package.in:3424:warning: ignoring type redefinition of 'PACKAGE_luci-ssl' from 'bool' to 'tristate'
tmp/.config-package.in:3479:warning: ignoring type redefinition of 'PACKAGE_luci-base' from 'bool' to 'tristate'
tmp/.config-package.in:3517:warning: ignoring type redefinition of 'LUCI_LANG_ar' from 'bool' to 'tristate'
tmp/.config-package.in:3520:warning: ignoring type redefinition of 'LUCI_LANG_bg' from 'bool' to 'tristate'
tmp/.config-package.in:3523:warning: ignoring type redefinition of 'LUCI_LANG_bn_BD' from 'bool' to 'tristate'
tmp/.config-package.in:3526:warning: ignoring type redefinition of 'LUCI_LANG_ca' from 'bool' to 'tristate'
tmp/.config-package.in:3529:warning: ignoring type redefinition of 'LUCI_LANG_cs' from 'bool' to 'tristate'
tmp/.config-package.in:3532:warning: ignoring type redefinition of 'LUCI_LANG_da' from 'bool' to 'tristate'
tmp/.config-package.in:3535:warning: ignoring type redefinition of 'LUCI_LANG_de' from 'bool' to 'tristate'
tmp/.config-package.in:3538:warning: ignoring type redefinition of 'LUCI_LANG_el' from 'bool' to 'tristate'
tmp/.config-package.in:3541:warning: ignoring type redefinition of 'LUCI_LANG_en' from 'bool' to 'tristate'
tmp/.config-package.in:3544:warning: ignoring type redefinition of 'LUCI_LANG_es' from 'bool' to 'tristate'
tmp/.config-package.in:3547:warning: ignoring type redefinition of 'LUCI_LANG_fi' from 'bool' to 'tristate'
tmp/.config-package.in:3550:warning: ignoring type redefinition of 'LUCI_LANG_fr' from 'bool' to 'tristate'
tmp/.config-package.in:3553:warning: ignoring type redefinition of 'LUCI_LANG_he' from 'bool' to 'tristate'
tmp/.config-package.in:3556:warning: ignoring type redefinition of 'LUCI_LANG_hi' from 'bool' to 'tristate'
tmp/.config-package.in:3559:warning: ignoring type redefinition of 'LUCI_LANG_hu' from 'bool' to 'tristate'
tmp/.config-package.in:3562:warning: ignoring type redefinition of 'LUCI_LANG_it' from 'bool' to 'tristate'
tmp/.config-package.in:3565:warning: ignoring type redefinition of 'LUCI_LANG_ja' from 'bool' to 'tristate'
tmp/.config-package.in:3568:warning: ignoring type redefinition of 'LUCI_LANG_ko' from 'bool' to 'tristate'
[... 581 lines cut ...]
go: downloading github.com/prometheus/client_golang v1.15.1
go: downloading golang.org/x/net v0.9.0
go: downloading google.golang.org/protobuf v1.30.0
go: downloading github.com/imdario/mergo v0.3.15
go: downloading golang.org/x/sys v0.8.0
go: downloading github.com/cespare/xxhash/v2 v2.2.0
go: downloading github.com/prometheus/client_model v0.4.0
go: downloading github.com/prometheus/common v0.42.0
go: downloading github.com/prometheus/procfs v0.9.0
go: downloading github.com/golang/protobuf v1.5.3
go: downloading github.com/matttproud/golang_protobuf_extensions v1.0.4
go: downloading github.com/vishvananda/netns v0.0.4

Building targets
../../cert/cert.go:5:2: package crypto/ecdh is not in GOROOT (/home/stangri/development/Builder/SDK_x86-64_22.03.5/staging_dir/hostpkg/lib/go-cross/src/crypto/ecdh)

make[2]: *** [Makefile:134: /home/stangri/development/Builder/SDK_x86-64_22.03.5/build_dir/target-x86_64_musl/nebula-1.7.1/.built] Error 1
make[2]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_22.03.5/feeds/packages/stangri/nebula'
time: package/nebula/compile#1.97#0.65#2.22
    ERROR: package/nebula failed to build.
make[1]: *** [package/Makefile:116: package/nebula/compile] Error 1
make[1]: Leaving directory '/home/stangri/development/Builder/SDK_x86-64_22.03.5'
make: *** [/home/stangri/development/Builder/SDK_x86-64_22.03.5/include/toplevel.mk:223: package/nebula/compile] Error 2

```

jefferyto, 2023-05-22:
nebula 1.7.0 [requires Go 1.20](https://github.com/slackhq/nebula/blob/165b671e70aabfd6f06d90f14e8ca4a3476e8ad6/CHANGELOG.md#fixed-1), 22.03 has Go 1.19.

stangri, 2023-05-23:
Thank you for your prompt reply @jefferyto! With snapshots SDK it compiled successfully. Do you have any plans to backport Go 1.20 to 22.03 branch? I'm asking because my test environment is stable, so I'd prefer to keep the same version/code in snapshots and stable.

jefferyto, 2023-05-23:
> Do you have any plans to backport Go 1.20 to 22.03 branch?

No - [release branches receive security and bug fixes only](https://github.com/openwrt/packages/blob/72c8977b3b4b3b8c905bd53db7557cf6bc8963db/CONTRIBUTING.md#release-branches).

closed 2023-05-26 by stangri