openwrt/packages #21144: vala: does not compile

flyn-org, 2023-05-23:
Maintainer: No maintainer listed in Makefile
Environment: x86_64 master
Description:

Building the Vala package fails for me with the following errors. I am building on Fedora 38, and this is a host package, so I wonder if the problem stems from the version of dependencies I have installed on my computer.
```
/usr/bin/env bash ../libtool  --tag=CC   --mode=link gcc  -O2 -I/home/mike/Download/openwrt-beholder/staging_dir/host/include -I/home/mike/Download/openwrt-beholder/staging_dir/hostpkg/include -I/home/mike/Download/openwrt-beholder/staging_dir/target-x86_64_musl/host/include  -L/home/mike/Download/openwrt-beholder/staging_dir/host/lib -L/home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib -L/home/mike/Download/openwrt-beholder/staging_dir/target-x86_64_musl/host/lib -o valac valacompiler.o    -L/home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib -lgobject-2.0 -lpcre2-8 -lffi -lglib-2.0 -pthread -lm ../vala/libvala-0.56.la ../ccode/libvalaccode.la ../codegen/libvalaccodegen.la  
libtool: link: gcc -O2 -I/home/mike/Download/openwrt-beholder/staging_dir/host/include -I/home/mike/Download/openwrt-beholder/staging_dir/hostpkg/include -I/home/mike/Download/openwrt-beholder/staging_dir/target-x86_64_musl/host/include -o valac valacompiler.o  -L/home/mike/Download/openwrt-beholder/staging_dir/host/lib -L/home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib -L/home/mike/Download/openwrt-beholder/staging_dir/target-x86_64_musl/host/lib ../vala/.libs/libvala-0.56.a ../ccode/.libs/libvalaccode.a ../codegen/.libs/libvalaccodegen.a /home/mike/Download/openwrt-beholder/build_dir/hostpkg/vala-0.56.3/vala/.libs/libvala-0.56.a -lgmodule-2.0 -lgobject-2.0 -lpcre2-8 /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libffi.a -lglib-2.0 -lm -pthread
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `enable_jit_with_match_options':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:906: undefined reference to `pcre2_jit_compile_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `get_matched_substring_number':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1463: undefined reference to `pcre2_substring_nametable_scan_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `recalc_match_offsets':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:852: undefined reference to `pcre2_get_ovector_count_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:862: undefined reference to `pcre2_get_ovector_pointer_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:852: undefined reference to `pcre2_get_ovector_count_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:852: undefined reference to `pcre2_get_ovector_count_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_match_info_next':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1088: undefined reference to `pcre2_match_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1078: undefined reference to `pcre2_jit_match_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1112: undefined reference to `pcre2_match_data_free_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1113: undefined reference to `pcre2_match_data_create_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `regex_compile':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1760: undefined reference to `pcre2_compile_context_create_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1763: undefined reference to `pcre2_set_newline_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1773: undefined reference to `pcre2_set_bsr_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1789: undefined reference to `pcre2_compile_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1795: undefined reference to `pcre2_compile_context_free_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1778: undefined reference to `pcre2_compile_context_free_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `match_info_new':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:808: undefined reference to `pcre2_pattern_info_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:811: undefined reference to `pcre2_match_context_create_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:827: undefined reference to `pcre2_match_data_create_from_pattern_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_unref':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1642: undefined reference to `pcre2_code_free_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_match_info_unref':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1006: undefined reference to `pcre2_match_context_free_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1008: undefined reference to `pcre2_match_data_free_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_new':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1694: undefined reference to `pcre2_config_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `get_pcre2_inline_compile_options':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1836: undefined reference to `pcre2_pattern_info_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1843: undefined reference to `pcre2_pattern_info_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_get_max_backref':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1887: undefined reference to `pcre2_pattern_info_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_get_capture_count':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1907: undefined reference to `pcre2_pattern_info_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_get_has_cr_or_lf':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1927: undefined reference to `pcre2_pattern_info_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o):/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1949: more undefined references to `pcre2_pattern_info_8' follow
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_match_all_full':
[... cut ...]
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `g_regex_get_string_number':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:2454: undefined reference to `pcre2_substring_number_from_name_8'
/usr/bin/ld: /home/mike/Download/openwrt-beholder/staging_dir/hostpkg/lib/libglib-2.0.a(gregex.c.o): in function `get_matched_substring_number':
/home/mike/Download/openwrt-beholder/build_dir/hostpkg/glib-2.74.0/openwrt-build/../glib/gregex.c:1459: undefined reference to `pcre2_substring_number_from_name_8'
```

neheb, 2023-06-11:
This seems to require pcre2/host

flyn-org, 2023-06-12:
I tried `HOST_BUILD_DEPENDS:=glib2/host pcre2-8/host` and `HOST_BUILD_DEPENDS:=glib2/host pcre2/host`, but neither seemed to help. I also noticed Vala requires glib2, and glib2 already requires pcre2/host.

neheb, 2023-06-12:
I'll debug this when I get back from my trip.

Quite weird that it's not working though. My guess is OS paths are leaking into the build.

flyn-org, 2023-06-12:
I thought the OS paths might be the issue too, but I uninstalled my OS-level pcre2-devel and pcre-devel packages. I must be overlooking something. I would appreciate very much if you took a look when you are able.

neheb, 2023-06-12:
So those linking errors happen when linking is attempted with a shared library. But pcre2/host is static only. No idea why shared linking is attempted.

flyn-org, 2023-06-12:
I am starting to think this is an issue of argument ordering. With static libraries, -lx must come after the object that requires -lx. This is not the case for dynamic libraries. My guess is that libpcre2 is required by one of the .a files, which appear last in the command-line arguments.

flyn-org, 2023-06-12:
I think I figured this out. See https://github.com/openwrt/packages/issues/21359.

flyn-org, 2023-06-26:
Pull request: https://github.com/openwrt/packages/pull/21434.

neheb, 2024-02-11:
@flyn-org this is happening again

esaaprillia, 2024-05-16:
@neheb 

when compiling glib2 it looks like pcre2 is created again.

This should not happen if pcre2 is already built.

adding pcre2 dependency in glib2 is pointless in my opinion

```
2024-05-16T06:23:00.1782209Z Checking if "atomic ops define" compiles: YES 
2024-05-16T06:23:00.1782573Z Checking if "sync swap" : links: NO 
2024-05-16T06:23:00.1783943Z ../../../build_dir/hostpkg/glib-2.80.2/meson.build:2053: WARNING: DEPRECATION: Option 'force_posix_threads' is deprecated and will be removed after GLib 2.72; please file an issue with your use case if you still require it
2024-05-16T06:23:00.1785024Z Run-time dependency threads found: YES
2024-05-16T06:23:00.1785460Z Header "pthread.h" has symbol "pthread_attr_setstacksize" : YES 
2024-05-16T06:23:00.1786004Z Header "pthread.h" has symbol "pthread_attr_setinheritsched" : YES 
2024-05-16T06:23:00.1786544Z Header "pthread.h" has symbol "pthread_condattr_setclock" : YES 
2024-05-16T06:23:00.1787101Z Header "pthread.h" has symbol "pthread_cond_timedwait_relative_np" : NO 
2024-05-16T06:23:00.1787637Z Header "pthread.h" has symbol "pthread_getname_np" : YES 
2024-05-16T06:23:00.1788122Z Header "pthread.h" has symbol "pthread_getaffinity_np" : YES 
2024-05-16T06:23:00.1788836Z Checking if "pthread_setname_np(const char*)" with dependency threads: links: NO 
2024-05-16T06:23:00.1789546Z Checking if "pthread_setname_np(pthread_t, const char*)" with dependency threads: links: YES 
2024-05-16T06:23:00.1790144Z Checking if "stack grows check" runs: NO (1)
2024-05-16T06:23:00.1790548Z Run-time dependency iconv found: YES
2024-05-16T06:23:00.1791063Z Looking for a fallback subproject for the dependency libpcre2-8 because:
2024-05-16T06:23:00.1791546Z Use of fallback dependencies is forced.
2024-05-16T06:23:00.1792316Z Downloading pcre2 source from https://github.com/PhilipHazel/pcre2/releases/download/pcre2-10.42/pcre2-10.42.tar.bz2
2024-05-16T06:23:01.8398365Z Downloading pcre2 patch from https://wrapdb.mesonbuild.com/v2/pcre2_10.42-2/get_patch
2024-05-16T06:23:04.0722957Z 
2024-05-16T06:23:04.0723610Z Executing subproject pcre2 
2024-05-16T06:23:04.0724113Z 
2024-05-16T06:23:04.0724332Z pcre2| Project name: pcre2
2024-05-16T06:23:04.0724861Z pcre2| Project version: 10.42
2024-05-16T06:23:04.0726807Z pcre2| C compiler for the host machine: /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/gcc (gcc 11.4.0 "gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
2024-05-16T06:23:04.0728910Z pcre2| C linker for the host machine: /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/gcc ld.bfd 2.38
2024-05-16T06:23:04.0730143Z pcre2| Has header "inttypes.h" : YES (cached)
2024-05-16T06:23:04.0730784Z pcre2| Has header "bzlib.h" : YES
2024-05-16T06:23:04.0731382Z pcre2| Has header "sys/stat.h" : YES (cached)
2024-05-16T06:23:04.0731997Z pcre2| Has header "sys/types.h" : YES (cached)
2024-05-16T06:23:04.0732672Z pcre2| Has header "sys/wait.h" : YES (cached)
2024-05-16T06:23:04.0733340Z pcre2| Has header "inttypes.h" : YES (cached)
2024-05-16T06:23:04.0733991Z pcre2| Has header "dirent.h" : YES (cached)
2024-05-16T06:23:04.0734594Z pcre2| Has header "dlfcn.h" : YES
2024-05-16T06:23:04.0735155Z pcre2| Has header "limits.h" : YES (cached)
2024-05-16T06:23:04.0735767Z pcre2| Has header "stdint.h" : YES (cached)
2024-05-16T06:23:04.0736368Z pcre2| Has header "stdlib.h" : YES (cached)
2024-05-16T06:23:04.0736975Z pcre2| Has header "string.h" : YES (cached)
2024-05-16T06:23:04.0737559Z pcre2| Has header "windows.h" : NO
2024-05-16T06:23:04.0738125Z pcre2| Has header "unistd.h" : YES (cached)
2024-05-16T06:23:04.0738743Z pcre2| Checking for function "memmove" : YES
2024-05-16T06:23:04.0739412Z pcre2| Checking for function "strerror" : YES
2024-05-16T06:23:04.0740691Z pcre2| Found pkg-config: YES (/home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/pkg-config) 2.1.1
2024-05-16T06:23:04.0742733Z pcre2| Program RunTest found: YES (/home/runner/work/mosdns/mosdns/sdk/build_dir/hostpkg/glib-2.80.2/subprojects/pcre2-10.42/RunTest)
2024-05-16T06:23:04.0745976Z pcre2| Program RunGrepTest found: YES (/home/runner/work/mosdns/mosdns/sdk/build_dir/hostpkg/glib-2.80.2/subprojects/pcre2-10.42/RunGrepTest)
2024-05-16T06:23:04.0747316Z pcre2| Build targets in project: 6
2024-05-16T06:23:04.0747994Z pcre2| NOTICE: Future-deprecated features used:
2024-05-16T06:23:04.0749033Z pcre2| * 0.64.0: {'copy arg in configure_file'}
2024-05-16T06:23:04.0749636Z pcre2| Subproject pcre2 finished.
2024-05-16T06:23:04.0749998Z 
2024-05-16T06:23:04.0750605Z Dependency libpcre2-8 from subproject subprojects/pcre2-10.42 found: YES 10.42
2024-05-16T06:23:04.0751290Z 
2024-05-16T06:23:04.0751463Z Executing subproject gvdb 
2024-05-16T06:23:04.0751760Z 
2024-05-16T06:23:04.0751919Z gvdb| Project name: gvdb
2024-05-16T06:23:04.0752363Z gvdb| Project version: 0.0
2024-05-16T06:23:04.0753898Z gvdb| C compiler for the host machine: /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/gcc (gcc 11.4.0 "gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0")
2024-05-16T06:23:04.0755888Z gvdb| C linker for the host machine: /home/runner/work/mosdns/mosdns/sdk/staging_dir/host/bin/gcc ld.bfd 2.38
2024-05-16T06:23:04.0757024Z gvdb| Build targets in project: 6
2024-05-16T06:23:04.0757539Z gvdb| Subproject gvdb finished.
2024-05-16T06:23:04.0757875Z 
2024-05-16T06:23:04.0758084Z Dependency gvdb found: YES 0.0 (overridden)
```

neheb, 2024-11-12:
Is this still a problem?