openwrt/packages #21155: python-msgpack fails to build

datahost, 2023-05-24:
Maintainer: Jan Pavlinec <jan.pavlinec1@gmail.com>
Environment: ipq807x, openwrt master
Description: python-msgpack fails to build

```
make[2]: Entering directory '/home/user/openwrt/2023-05-24/feeds/packages/lang/python/python-msgpack'
touch /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.prepared_c8404c0a783a4dd14d4257b33a633184_6664517399ebbbc92a37c5bb081b5c53_check
. /home/user/openwrt/2023-05-24/include/shell.sh; /home/user/openwrt/2023-05-24/staging_dir/host/bin/libdeflate-gzip -dc /home/user/openwrt/openwrt-dl/msgpack-1.0.2.tar.gz | tar -C /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2
touch /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.prepared_c8404c0a783a4dd14d4257b33a633184_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.configured_*
rm -f /home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl/stamp/.python-msgpack_installed
(cd /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/./; if [ -x ./configure ]; then find /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.guess | xargs -r chmod u+w; find /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.guess | xargs -r -n1 cp --remove-destination /home/user/openwrt/2023-05-24/scripts/config.guess; find /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.sub | xargs -r chmod u+w; find /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.sub | xargs -r -n1 cp --remove-destination /home/user/openwrt/2023-05-24/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache aarch64-openwrt-linux-musl-gcc" GCC="ccache aarch64-openwrt-linux-musl-gcc" CXX="ccache aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " LDFLAGS="-L/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  _python_sysroot="/home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr" ; fi; )
if [ -d "../host-pip-requirements" ] && [ -n "$(find ../host-pip-requirements -maxdepth 1 -mindepth 1 -name '*.txt' -print -quit 2>/dev/null)" ]; then if grep -q "msgpack==" ../host-pip-requirements/*.txt ; then if ! grep -q "msgpack==1.0.2" ../host-pip-requirements/*.txt ; then printf "\nPlease update version of msgpack to 1.0.2 in 'host-pip-requirements'/\n\n" ; exit 1 ; fi fi fi
touch /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.built
touch /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.built_check
cd "/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2" && CC="ccache aarch64-openwrt-linux-musl-gcc" CCSHARED="ccache aarch64-openwrt-linux-musl-gcc -DPIC -fPIC" CXX="ccache aarch64-openwrt-linux-musl-g++" LD="ccache aarch64-openwrt-linux-musl-gcc" LDSHARED="ccache aarch64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include -I/home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl/usr/include/python3.11" LDFLAGS="-L/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/user/openwrt/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -znow -zrelro -lpython3.11" _PYTHON_HOST_PLATFORM="linux-aarch64" __PYVENV_LAUNCHER__="/usr/bin/python3.11" PYTHONPATH="/home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.11:/home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl//usr/lib/python3.11/site-packages:/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ipkg-install//usr/lib/python3.11/site-packages" PYTHONDONTWRITEBYTECODE=1 _python_sysroot="/home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr"    /home/user/openwrt/2023-05-24/staging_dir/hostpkg/bin/python3.11  -m build --no-isolation --outdir "/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/"/openwrt-build --wheel   "/home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/" 
Traceback (most recent call last):
  File "<frozen runpy>", line 189, in _run_module_as_main
  File "<frozen runpy>", line 148, in _get_module_details
  File "<frozen runpy>", line 112, in _get_module_details
  File "/usr/lib/python3.11/site-packages/build/__init__.py", line 17, in <module>
    import subprocess
  File "/home/user/openwrt/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl/usr/lib/python3.11/subprocess.py", line 104, in <module>
    from _posixsubprocess import fork_exec as _fork_exec
ModuleNotFoundError: No module named '_posixsubprocess'
make[2]: *** [Makefile:44: /home/user/openwrt/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.built] Error 1
make[2]: Leaving directory '/home/user/openwrt/2023-05-24/feeds/packages/lang/python/python-msgpack'
time: package/feeds/packages/python-msgpack/compile#0.37#0.19#0.53

```

datahost, 2023-05-24:
I started a new attempt under Debian. Previously I tried Manjaro Linux. An error still occurs.
```

make[3]: Entering directory '/home/user/2023-05-24/feeds/packages/lang/python/python-msgpack'
touch /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.prepared_e50ff9285419679303737aa2a3c00f06_6664517399ebbbc92a37c5bb081b5c53_check
. /home/user/2023-05-24/include/shell.sh; /home/user/2023-05-24/staging_dir/host/bin/libdeflate-gzip -dc /home/user/openwrt-dl/msgpack-1.0.2.tar.gz | tar -C /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2
touch /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.prepared_e50ff9285419679303737aa2a3c00f06_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.configured_*
rm -f /home/user/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl/stamp/.python-msgpack_installed
(cd /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/./; if [ -x ./configure ]; then find /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.guess | xargs -r chmod u+w; find /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.guess | xargs -r -n1 cp --remove-destination /home/user/2023-05-24/scripts/config.guess; find /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.sub | xargs -r chmod u+w; find /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/ -name config.sub | xargs -r -n1 cp --remove-destination /home/user/2023-05-24/scripts/config.sub; AR="aarch64-openwrt-linux-musl-gcc-ar" AS="ccache aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="ccache aarch64-openwrt-linux-musl-gcc" GCC="ccache aarch64-openwrt-linux-musl-gcc" CXX="ccache aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/user/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/include -I/home/user/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include/fortify -I/home/user/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/include " LDFLAGS="-L/home/user/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/usr/lib -L/home/user/2023-05-24/staging_dir/toolchain-aarch64_cortex-a53_gcc-12.3.0_musl/lib -znow -zrelro "   ./configure --target=aarch64-openwrt-linux --host=aarch64-openwrt-linux --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  _python_sysroot="/home/user/2023-05-24/staging_dir/target-aarch64_cortex-a53_musl" _python_prefix="/usr" _python_exec_prefix="/usr" ; fi; )
if [ -d "../host-pip-requirements" ] && [ -n "$(find ../host-pip-requirements -maxdepth 1 -mindepth 1 -name '*.txt' -print -quit 2>/dev/null)" ]; then if grep -q "msgpack==" ../host-pip-requirements/*.txt ; then if ! grep -q "msgpack==1.0.2" ../host-pip-requirements/*.txt ; then printf "\nPlease update version of msgpack to 1.0.2 in 'host-pip-requirements'/\n\n" ; exit 1 ; fi fi fi
touch /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.built
touch /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.built_check
cd "/home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2" && CC="ccache aarch64-openwrt-linux-musl-gcc" CCSHARED="ccache aarch64-openwrt-linux-musl-gcc -DPIC -fPIC" CXX="ccache aarch64-openwrt-linux-musl-g++" LD="ccache aarch64-openwrt-linux-musl-gcc" LDSHARED="ccache aarch64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -mcpu=cortex-a53 -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2=msgpack-1.0.2 -Wformat -Werror=format-security -fstack-pro [... cut ...] d_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/"/openwrt-build --wheel   "/home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/" 
/usr/bin/python3.11: No module named build
make[3]: *** [Makefile:44: /home/user/2023-05-24/build_dir/target-aarch64_cortex-a53_musl/pypi/msgpack-1.0.2/.built] Error 1
make[3]: Leaving directory '/home/user/2023-05-24/feeds/packages/lang/python/python-msgpack'
time: package/feeds/packages/python-msgpack/compile#0.34#0.14#0.43

```

jefferyto, 2023-05-26:
#21167 has been merged - please check if you still a build error.

datahost, 2023-05-26:
Looks good. Problem fixed for now.

closed 2023-05-26 by datahost