openwrt/packages #21157: Question : libwebsockets -enbling support for libevent

snaami, 2023-05-24:
Hello All,

I see that **libwebsockets-full** doesn't enable libevent2 support by default see (https://github.com/openwrt/packages/blob/master/libs/libwebsockets/Makefile#L87). by adding the ( **-DLWS_WITH_LIBEVENT=ON**)

Do you think it would be beneficial to enable it by default in the config, there are many application relays on it out there and many of them use libevent as event loop?

Thanks,

brada4, 2023-05-24:
Maybe enable event plugins an build available ones that in turn pull larger dependencies?
https://libwebsockets.org/git/libwebsockets/tree/lib/event-libs

brada4, 2023-05-24:
libwebsockets can plug event modules dynamically now. What I am saying is to not introduce any new dependencies in current package, just let it load those plugins from separate packages, and -full package is just dependency-set to emulate the might it once has?

mhei, 2023-05-24:
The different event backends are different shared libraries if I understand correctly. Maybe it is possible to pack each backend into its own "backend" package. If packed all into the -full package, the build system would insist to add  dependencies to the -full package. Just 2 minutes thought... I'm not sure yet if that makes sense or is doable.

karlp, 2023-05-24:
I'm no longer using this for $dayjob anymore, so if anyone of you want to work on this, feel _extremely_ welcome to take over maintainership on this package too, it's been a _lot_ of pain over the years

snaami, 2023-05-25:
> The different event backends are different shared libraries if I understand correctly. Maybe it is possible to pack each backend into its own "backend" package. If packed all into the -full package, [...]

I think it is better to have each event-lib packed into its own config maybe something like **libwebsockets-event**, **libwebsockets-uv** ...

snaami, 2023-05-25:
On Debian they have all of them enabled by default :

    **libwebsockets-evlib-ev
    libwebsockets-evlib-uv
    libwebsockets-evlib-glib**

All of them then depend on core/main libwebsockets package

They use the following configuration for that :

**dh_auto_configure -- -DCMAKE_LIBRARY_ARCHITECTURE=${arch} \
            -DLIB_SUFFIX=/${arch} -DLWS_WITHOUT_DAEMONIZE=OFF \
            -DLWS_WITH_LIBEV=ON -DLWS_WITH_LIBUV=ON \
            -DLWS_UNIX_SOCK=ON -DLWS_IPV6=ON \
            -DLWS_WITH_ZLIB=ON -DLWS_WITHOUT_EXTENSIONS=OFF \
            -DLWS_WITH_GLIB=ON**

As we can see, all event-libs are enabled at compile time in the same config.

brada4, 2023-05-25:
Check if libs are separate files just like that, its ok to install them one by one in separate packages.
Default should work without plugins for all uses that fit in embedded device.