openwrt/packages #21158: rust: offline build for rust packages

feckert, 2023-05-24:
Maintainer: @lu-zero @1715173329 
Environment: all

Description:
This is not a problem, it is more a question of how I can build `rust` programs offline.
I noticed or was not aware that for `rust` program, the external dependencies are also downloaded each time before building (judging by the log). This means that if I don't have an internet connection, I can't build the `rust` binaries I built before because I don't have an internet connection to resolve the dependencies. Wouldn't it make sense to use a cache here? 

I hope I have not misunderstood and that this is something completely different. Can one of those who know about this briefly explain to me how this is now handled in the OpenWrt build system for `rust`?

lu-zero, 2023-05-24:
`cargo` normally work assuming a connection, it can work offline by passing --offline

see also [cargo vendor](https://doc.rust-lang.org/cargo/commands/cargo-vendor.html) and [cargo fetch](https://doc.rust-lang.org/cargo/commands/cargo-fetch.html)

diizzyy, 2023-05-27:
Looking at the code I don't think that was ever considered, --offline "might" work but you likely need more logic than that. It's actually quite hard to contain languagues with their own "package system" for consistency/reproducibility, rust and go are duable but npm is very troublesome which is why barely any npm-based software packaged at all in repos.

FreeBSD for example requires offline build capability for all ports and there's a lot of glue to hold it together.
Example 57 gives you an idea: https://docs.freebsd.org/en/books/porters-handbook/book/#using-cargo

lu-zero, 2023-05-27:
Which are the requirements for OpenWrt? Having a deeper integration with cargo in the fetch phase is surely feasible, I can also mention the Gentoo eclass as way to support offline building.