openwrt/packages #21174: Use python3-packages install python package. It install to wrong directory.

p870613, 2023-05-26:
Hi, 
when I ues the python3-package to install python package.
It will install to wrong directory.

reproduce 
1. make menuconfig
2. open `Advanced configuration options`
3. `Languages` -> `python` -> `python3-packages` -> `List of python3 packages to install on host `: fill with the name of python package, example `cryptography`
4. make package/python3-packages/compille V=s
5. It will install in `/mfs/xxxxx/xxxx/mt7986/openwrt/lede/staging_dir/hostpkg/mfs/xxxxx/xxxx/mt7986/openwrt/lede/staging_dir/hostpkg/lib/python3.9/site-packages/cryptography` 

Expected behaviour:
It will install in `/mfs/xxxxx/xxxx/mt7986/openwrt/lede/staging_dir/hostpkg/lib/python3.9/site-packages/cryptography` 

I review the Makefile code in python3-packages (https://github.com/openwrt/packages/blob/master/lang/python/python3-packages/Makefile) 

line 91-98
```c==n
HOST_PYTHON3_PIP_INSTALL=$(HOST_PYTHON3_PIP) install \
	--root=$(1) \
	--prefix=$(2) \
	--ignore-installed \
	--no-build-isolation \
	--no-compile \
	$(if $(CONFIG_PACKAGE_python3-packages-index-url), --index-url "$(CONFIG_PACKAGE_python3-packages-index-url)") \
	$(if $(CONFIG_PACKAGE_python3-packages-pip-opts), $(CONFIG_PACKAGE_python3-packages-pip-opts)) \
```
when I change the pip parameter to `--prefix=$(1)`, the package will install  `/mfs/xxxxx/xxxx/mt7986/openwrt/lede/staging_dir/hostpkg/lib/python3.9/site-packages/cryptography` . It is Expected behaviour.

jefferyto, 2023-05-26:
Ping maintainer @yousong 

Also please try to reproduce your issue with current master.

schm0, 2023-05-29:
python3-host-build.mk
Seems to have problems too.
`PYTHON3_HOST_BUILD_DIR?=$(HOST_BUILD_DIR)/$(PYTHON3_HOST_BUILD_PATH)`
PYTHON3_HOST_BUILD_PATH seems to be empty which results in double // in the Path.
PYTHON3_HOST_BUILD_DIR gets quoted mid path which in having " in the path.
`$(PYTHON3_HOST_WHEEL_NAME)-$(PYTHON3_HOST_WHEEL_VERSION)-*.whl`
The * doesn't seem to work here and should be "py3-none-any" I think.

jefferyto, 2023-05-29:
@schm0 python3-host-build.mk is unrelated to python3-packages. python3-packages is a way to install host packages using pip; python3-host-build.mk is to install host packages using a host build of a OpenWrt source package. If you are using problems with python3-host-build.mk please open a separate issue.

schm0, 2023-05-29:
Yes, I know but I actually didn't want to open a separate issue for this.

jefferyto, 2023-05-29:
> Yes, I know but I actually didn't want to open a separate issue for this.

Then don't expect anyone to help you.

schm0, 2023-05-29:
Sorry.
I did not ask for help, but noticed that this file also has problems and fixed it myself.

p870613, 2023-07-12:
Hi, 
I think it also reproduce this issue in master branch becasue it is use this `pip` command to install.
And, it have had no change from openwrt21.02 to master.
It maybe can be soluted by change `pip` parameter like changing `--root=$(1) --prefix=$(2)` to `--prefix=$(1)`.
If you need, I can launch a PR for this issue.

Thanks!!

p870613, 2023-07-24:
Hi
I test this package in `master` and `openwrt22.02`.
It can't reproduce this bug.
But, it can reproduce in openwrt.
The main difference between `openwrt22.02` and `openwrt21.02` is python version.
The python version in `Openwrt 21.02` is 3.9.5 .
I use same command to install some python package.
It installs in wrong path.
[image]
[image]

However, The python version in `Openwrt 22.02` is 3.10.5 and use same command to install python package.
It installs in the correct path.
[image]
[image]

Thanks !!

closed 2024-05-04 by p870613