openwrt/packages #21176: nmap: (full variant) build error due to python error

AcidSlide, 2023-05-26:
Maintainer: @nunojpg @jefferyto 
Environment: x86/64

Description:
**nmap** (full variant)l failing to build due to phython error `ModuleNotFoundError: No module named 'math'`

Below is the snippet of the build log error

```
make[4]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93'
install -d -m0755 /home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ipkg-install//usr/lib/python3.11/site-packages
cd "/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ndiff" && CC="x86_64-openwrt-linux-musl-gcc" CCSHARED="x86_64-openwrt-linux-musl-gcc -DPIC -fpic" CXX="x86_64-openwrt-linux-musl-g++" LD="x86_64-openwrt-linux-musl-gcc" LDSHARED="x86_64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93=nmap-7.93 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/python3.11" LDFLAGS="-L/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -znow -zrelro -lpython3.11" _PYTHON_HOST_PLATFORM="linux-x86_64" __PYVENV_LAUNCHER__="/usr/bin/python3.11" PYTHONPATH="/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11:/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl//usr/lib/python3.11/site-packages:/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ipkg-install//usr/lib/python3.11/site-packages" PYTHONDONTWRITEBYTECODE=1 _python_sysroot="/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl" _python_prefix="/usr" _python_exec_prefix="/usr"    /home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/bin/python3.11  setup.py  install --prefix="/usr" --root="/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ipkg-install"  
Traceback (most recent call last):
  File "/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ndiff/setup.py", line 11, in <module>
    import distutils.command
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages/_distutils_hack/__init__.py", line 97, in find_spec
    return method()
           ^^^^^^^^
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages/_distutils_hack/__init__.py", line 104, in spec_for_distutils
    import importlib.abc
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/importlib/abc.py", line 19, in <module>
    from .resources.abc import ResourceReader, Traversable, TraversableResources
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/importlib/resources/__init__.py", line 3, in <module>
    from ._common import (
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/importlib/resources/_common.py", line 3, in <module>
    import tempfile
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/tempfile.py", line 45, in <module>
    from random import Random as _Random
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/random.py", line 49, in <module>
    from math import log as _log, exp as _exp, pi as _pi, e as _e, ceil as _ceil
ModuleNotFoundError: No module named 'math'
make[3]: *** [Makefile:205: /home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/.built] Error 1
make[3]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/feeds/packages/net/nmap'
time: package/feeds/packages/nmap/full/compile#0.64#0.40#1.02
    ERROR: package/feeds/packages/nmap failed to build (build variant: full).
make[2]: *** [package/Makefile:120: package/feeds/packages/nmap/compile] Error 1
make[2]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make[1]: *** [package/Makefile:114: /home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make: *** [/home/xxxxxxxx/openwrt/openwrt/include/toplevel.mk:231: world] Error 2
make[4]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93'
install -d -m0755 /home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ipkg-install//usr/lib/python3.11/site-packages
cd "/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ndiff" && CC="x86_64-openwrt-linux-musl-gcc" CCSHARED="x86_64-openwrt-linux-musl-gcc -DPIC -fpic" CXX="x86_64-openwrt-linux-musl-g++" LD="x86_64-openwrt-linux-musl-gcc" LDSHARED="x86_64-openwrt-linux-musl-gcc -shared" CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93=nmap-7.93 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CPPFLAGS="-I/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include/fortify -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/include -I/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/include/python3.11" LDFLAGS="-L/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/usr/lib -L/home/xxxxxxxx/openwrt/openwrt/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib -znow -zrelro -lpython3.11" _PYTHON_HOST_PLATFORM="linux-x86_64" __PYVENV_LAUNCHER__="/usr/bin/python3.11" PYTHONPATH="/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11:/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl//usr/lib/python3.11/site-packages:/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ipkg-install//usr/lib/python3.11/site-packages" PYTHONDONTWRITEBYTECODE=1 _python_sysroot="/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl" _python_prefix="/usr" _python_exec_prefix="/usr"    /home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/bin/python3.11  setup.py  install --prefix="/usr" --root="/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ipkg-install"  
Traceback (most recent call last):
  File "/home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/ndiff/setup.py", line 11, in <module>
    import distutils.command
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages/_distutils_hack/__init__.py", line 97, in find_spec
    return method()
           ^^^^^^^^
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/hostpkg/lib/python3.11/site-packages/_distutils_hack/__init__.py", line 104, in spec_for_distutils
    import importlib.abc
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/importlib/abc.py", line 19, in <module>
    from .resources.abc import ResourceReader, Traversable, TraversableResources
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/importlib/resources/__init__.py", line 3, in <module>
    from ._common import (
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/importlib/resources/_common.py", line 3, in <module>
    import tempfile
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/tempfile.py", line 45, in <module>
    from random import Random as _Random
  File "/home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/usr/lib/python3.11/random.py", line 49, in <module>
    from math import log as _log, exp as _exp, pi as _pi, e as _e, ceil as _ceil
ModuleNotFoundError: No module named 'math'
make[3]: *** [Makefile:205: /home/xxxxxxxx/openwrt/openwrt/build_dir/target-x86_64_musl/nmap-full/nmap-7.93/.built] Error 1
make[3]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt/feeds/packages/net/nmap'
[... cut ...] #1.02
    ERROR: package/feeds/packages/nmap failed to build (build variant: full).
make[2]: *** [package/Makefile:120: package/feeds/packages/nmap/compile] Error 1
make[2]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make[1]: *** [package/Makefile:114: /home/xxxxxxxx/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xxxxxxxx/openwrt/openwrt'
make: *** [/home/xxxxxxxx/openwrt/openwrt/include/toplevel.mk:231: world] Error 2
```

jefferyto, 2023-05-26:
#21167 has been merged - please check if you still get a build error.

AcidSlide, 2023-05-28:
> #21167 has been merged - please check if you still get a build error.

Hi @jefferyto I'm able to build with no issues with the latest master.  But to be honest I'm not sure if this was the actual fix since I was seeing a lot of commit changes related to python libs.  I'll just take your word for it hahaha thanks :)

closed 2023-05-28 by jefferyto