openwrt/packages #21187: ddns-scripts-services: Call script when IP changes

doodieb15, 2023-05-26:
Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: WRT 1900-ACS, OpenWrt 21.02.3 

Description:

I would like a user exit where I can attach a script when the WAN IP has changed. This would simplify my setup significantly.

Thank you.

brada4, 2023-05-26:
You can reload ddns via hotplug? https://openwrt.org/docs/guide-user/base-system/hotplug#iface

doodieb15, 2023-05-26:
Hmm, not sure. Between the router and the internet there is another Modem (not in bridge Mode), so the IF does not trigger a hotplug event when the WAN IP changes (I think)

brada4, 2023-05-26:
Actually there is hotplug script that handles only ifup/ifdown, not address change. Will not help in your case as there is no signalling from your router about public IP changes.

feckert, 2023-10-04:
If the WAN IP has changed then there is an event at netifd. I assume that the WAN is connected via DHCP.
The netifd generates a renew event see for example this hook in the [dhcp.script](https://github.com/openwrt/openwrt/blob/master/package/network/config/netifd/files/lib/netifd/dhcp.script#L114) . 

https://github.com/openwrt/openwrt/blob/master/package/network/config/netifd/files/lib/netifd/dhcp.script#L114

closed 2023-10-04 by feckert