openwrt/packages #21200: openvpn: new hotplug events breaks server mode configs

mtpi, 2023-05-27:
Please make sure that the issue subject starts with `<package-name>: `

Also make sure that the package is maintained in this repository and not in base which should be submitted at https://bugs.openwrt.org or in the LuCI repository which should be submitted at https://github.com/openwrt/luci/issues.

Issues related to releases below 18.06 and forks are not supported or maintained and will be closed.

# Issue template (remove lines from top till here)

Maintainer: @miska (not sure about the maintainer)
Environment: OpenWrt SNAPSHOT (r23104-ef98dc3b3e) on Linksys WRT3200ACM

Description:
On the latest snapshot build i cannot start anymore my openvpn server.
The system log gives this error:
`Sat May 27 19:21:19 2023 daemon.err openvpn(server_tun)[10827]: Options error: --ipchange cannot be used with --mode server (use --client-connect instead)
Sat May 27 19:21:19 2023 daemon.warn openvpn(server_tun)[10827]: Use --help for more information.`
The config for this istance is a simple server config.
I think the error has been introduces by [this commit](https://github.com/openwrt/packages/commit/f8a8b71e26b9bdbf86fbb7d4d1482637af7f3ba4). This introduces the hotplug event for ipchange that is not supported for server istances, but only for client ones.

BKPepe, 2023-05-28:
cc: @miska ⬆️

ptpt52, 2023-05-29:
same issue for my openvpn server setup.

BKPepe, 2023-05-31:
@neheb I think we should revert this until someone has time to look at it. We can not and we really don't want to have a broken OpenVPN server.

closed 2023-06-09 by neheb