openwrt/packages #21228: python-pkgconfig fails to build

datahost, 2023-05-29:
Maintainer: Jeffery To <jeffery.to@gmail.com>
Environment: ipq807x, Dynalink DL-WRX36, current master

Description: python-pkgconfig fails to build

```
make[3]: Entering directory '/home/user/2023-05-29/feeds/packages/lang/python/python-pkgconfig'
. /home/user/2023-05-29/include/shell.sh; /home/user/2023-05-29/staging_dir/host/bin/libdeflate-gzip -dc /home/user/openwrt-dl/pkgconfig-1.5.5.tar.gz | tar -C /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5
touch /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/.prepared80174871a645d4f7c5fe845d76d9ba87_6664517399ebbbc92a37c5bb081b5c53
(cd /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/; if [ -x configure ]; then cp -fpR /home/user/2023-05-29/scripts/config.{guess,sub} /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5// && CC="ccache gcc" CFLAGS="-O2 -I/home/user/2023-05-29/staging_dir/host/include -I/home/user/2023-05-29/staging_dir/hostpkg/include -I/home/user/2023-05-29/staging_dir/target-aarch64_cortex-a53_musl/host/include" CXX="ccache g++" CPPFLAGS="-I/home/user/2023-05-29/staging_dir/host/include -I/home/user/2023-05-29/staging_dir/hostpkg/include -I/home/user/2023-05-29/staging_dir/target-aarch64_cortex-a53_musl/host/include" CXXFLAGS="" LDFLAGS="-L/home/user/2023-05-29/staging_dir/host/lib -L/home/user/2023-05-29/staging_dir/hostpkg/lib -L/home/user/2023-05-29/staging_dir/target-aarch64_cortex-a53_musl/host/lib" CONFIG_SHELL="/usr/bin/env bash"  bash ./configure --target=x86_64-pc-linux-gnu --host=x86_64-pc-linux-gnu --build=x86_64-pc-linux-gnu --disable-dependency-tracking --program-prefix="" --program-suffix="" --prefix=/home/user/2023-05-29/staging_dir/hostpkg --exec-prefix=/home/user/2023-05-29/staging_dir/hostpkg --sysconfdir=/home/user/2023-05-29/staging_dir/hostpkg/etc --localstatedir=/home/user/2023-05-29/staging_dir/hostpkg/var --sbindir=/home/user/2023-05-29/staging_dir/hostpkg/bin ; fi )
touch /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/.configured
cd "/home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5" && ARCH="x86_64" CC="ccache gcc" CCSHARED="ccache gcc -DPIC -fPIC" CXX="ccache g++" LD="ccache gcc" LDSHARED="ccache gcc -shared" CFLAGS="-O2 -I/home/user/2023-05-29/staging_dir/host/include -I/home/user/2023-05-29/staging_dir/hostpkg/include -I/home/user/2023-05-29/staging_dir/target-aarch64_cortex-a53_musl/host/include" CPPFLAGS="-I/home/user/2023-05-29/staging_dir/host/include -I/home/user/2023-05-29/staging_dir/hostpkg/include -I/home/user/2023-05-29/staging_dir/target-aarch64_cortex-a53_musl/host/include -I/home/user/2023-05-29/staging_dir/hostpkg/include/python3.11" LDFLAGS="-L/home/user/2023-05-29/staging_dir/host/lib -L/home/user/2023-05-29/staging_dir/hostpkg/lib -L/home/user/2023-05-29/staging_dir/target-aarch64_cortex-a53_musl/host/lib -lpython3.11 -Wl,-rpath,/home/user/2023-05-29/staging_dir/hostpkg/lib"    /home/user/2023-05-29/staging_dir/hostpkg/bin/python3.11  -m build --no-isolation --outdir "/home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/"/openwrt-build --wheel   "/home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/"
* Getting build dependencies for wheel...

Traceback (most recent call last):
  File "/home/user/2023-05-29/staging_dir/hostpkg/lib/python3.11/site-packages/pyproject_hooks/_impl.py", line 321, in _call_hook
    raise BackendUnavailable(data.get('traceback', ''))
pyproject_hooks._impl.BackendUnavailable: Traceback (most recent call last):
  File "/home/user/2023-05-29/staging_dir/hostpkg/lib/python3.11/site-packages/pyproject_hooks/_in_process/_in_process.py", line 77, in _build_backend
    obj = import_module(mod_path)
          ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/2023-05-29/staging_dir/hostpkg/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1128, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1142, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'poetry'

ERROR Backend 'poetry.core.masonry.api' is not available.
make[3]: *** [Makefile:49: /home/user/2023-05-29/build_dir/hostpkg/pypi/pkgconfig-1.5.5/.built] Error 1
make[3]: Leaving directory '/home/user/2023-05-29/feeds/packages/lang/python/python-pkgconfig'
time: package/feeds/packages/python-pkgconfig/host-compile#0.33#0.14#0.49
```

jefferyto, 2023-05-29:
Do you get the same error when building from a clean checkout of buildroot / sdk?

datahost, 2023-05-29:
I have already tried with a recent and clean build root.

jefferyto, 2023-05-29:
Oh this is https://github.com/openwrt/packages/issues/20994#issuecomment-1555071637

The fix is here: https://github.com/openwrt/openwrt/pull/12710

datahost, 2023-05-29:
it works, thank you very much

closed 2023-05-29 by datahost