openwrt/packages #21234: Problem building nginx web server

sjkhsl, 2023-05-31:
After using the nginx web server to build, nginx cannot run, there is no nginx.conf file in the /etc/nginx/ directory, and after service nginx restarts, it prompts to execute the `nginx -T -c '/etc/nginx/uci.conf'` command, but after execution Another hint: 
```
[emerg] 9971#0: open() "/etc/nginx/uwsgi_params" failed (2: No such file or directory) in /etc/nginx/conf.d/luci.locations:3
nginx: configuration file /etc/nginx/uci.conf test failed
```
Check the cause is the lack of uwsgi_params file, there is no uwsgi_params file in the /etc/nginx/ directory.

brada4, 2023-05-31:
It is regarding luci-nginx config or nginx wothout LuCI?

sjkhsl, 2023-06-01:
> 

Should it be a problem with luci-nginx?

sjkhsl, 2023-06-01:
> It is regarding luci-nginx config or nginx wothout LuCI?

Should it be a problem with luci-nginx?

brada4, 2023-06-01:
uci.conf is from that package

sjkhsl, 2023-06-01:
> uci.conf 来自那个包

I didn't pay attention to which package uci. conf comes from, probably the nginx package? This file was previously available and should also generate a default configuration file for nginx. conf, but it will not be automatically generated now. And uwsgi_ Params file, extracted from the old version, uploaded to the new version, and you can access the web page!

brada4, 2023-06-01:
nginx start script checks for presence of nginx-luci conf files and starts luci in nginx finding one. Without it it uses normal layout in /etc/nginx. Does not look like you can have both at the same time.
Try sh -x /etc/init.d/nginx start fro some hints why it detectc luci nginx in place of normal one.

sjkhsl, 2023-06-02:
> nginx start script checks for presence of nginx-luci conf files and starts luci in nginx finding one. Without it it uses normal layout in /etc/nginx. Does not look like you can have both at the same [...]

The problem now is that the nginx.conf and uwsgi_params files are missing after the build is complete. It should be a problem during the build process, right?

brada4, 2023-06-02:
The problem is that nginx-luci template gets detected probably misding all other files to run luci in nginx.
Please check script trace. Blank nginx has all the files needed to start.

sjkhsl, 2023-06-06:
> The problem is that nginx-luci template gets detected probably misding all other files to run luci in nginx. Please check script trace. Blank nginx has all the files needed to start.

A new build of 23.05, with a different problem.

```
root@OpenWrt:~# nginx -T -c '/etc/nginx/uci.conf'
2023/06/06 11:19:03 [emerg] 32725#0: unknown directive "ubus_interpreter" in /etc/nginx/conf.d/luci.locations:20
nginx: configuration file /etc/nginx/uci.conf test failed
root@OpenWrt:~# nginx -T -c '/etc/nginx/nginx.conf'
2023/06/06 11:19:15 [emerg] 813#0: unknown directive "ubus_interpreter" in /etc/nginx/conf.d/luci.locations:20
nginx: configuration file /etc/nginx/nginx.conf test failed
```

brada4, 2023-06-06:
It seems that your installation keeps separate files originally from nginx-luci package.
They are kept over sysupgrade - is it the case?
Please uninstall nginx and nginx-luci and remove(archiving) /etc/nginx and then install nginx, and nginx-luci ONLY if you want nginx to serve ONLY luci web admin.

sjkhsl, 2023-06-06:
> It seems that your installation keeps separate files originally from nginx-luci package. They are kept over sysupgrade - is it the case? Please uninstall nginx and nginx-luci and remove(archiving) /etc/nginx and then install nginx, and nginx-luci ONLY if you want nginx to serve ONLY luci web admin.

I'm looking to see if it's caused by keeping the file

sjkhsl, 2023-06-06:
> It seems that your installation keeps separate files originally from nginx-luci package. They are kept over sysupgrade - is it the case? Please uninstall nginx and nginx-luci and remove(archiving) /etc/nginx and then install nginx, and nginx-luci ONLY if you want nginx to serve ONLY luci web admin.

Delete the previously built folder and build it after re-git clone 23.05, the problem remains

```
BusyBox v1.36.1 (2023-06-03 09:42:00 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05-SNAPSHOT, r23069-e2701e0f33
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# /etc/init.d/nginx restart
Command failed: Not found
nginx_init: NOT using conf file!
show config to be used by: nginx -T -c '/etc/nginx/uci.conf'
root@OpenWrt:~# nginx -T -c '/etc/nginx/uci.conf'
2023/06/06 11:51:36 [emerg] 10411#0: open() "/etc/nginx/uwsgi_params" failed (2: No such file or directory) in /etc/nginx/conf.d/luci.locations:3
nginx: configuration file /etc/nginx/uci.conf test failed
```

brada4, 2023-06-06:
run
`/bin/sh /etc/init.d/nginx restart`
Just to understand how normal nginx config poisons itself to use luci injects for no reason.

sjkhsl, 2023-06-06:
> run `/bin/sh -x /etc/init.d/nginx restart` Just to understand how normal nginx config poisons itself to use luci injects for no reason.

Testing the main version of nginx is normal, with two more updates on nginx compared to 23.05 for main

regulskimichal, 2023-06-12:
For some reason nginx stopped loading modules such as nginx-ubus-module.
To fix it on the top level of nginx.conf I included load_module e.g.
```
load_module /usr/lib/nginx/modules/ngx_http_ubus_module.so;
```

sjkhsl, 2023-06-13:
> For some reason nginx stopped loading modules such as nginx-ubus-module. To fix it on the top level of nginx.conf I included load_module e.g.
> 
> ```
> load_module /usr/lib/nginx/modules/ngx_http_ubus_module.so;
> ```

23.05 After the nginx is built, there are still problems，23.05 is still missing two patches commit 7b91373 commit f5d7a26，It is normal in master。

Ansuel, 2023-07-07:
@sjkhsl those 2 commit should not be related to this bug they are just cleanup. Please provide more info like the version of nginx and the template installed

alex3d, 2023-08-31:
`uwsgi_params` and some other config files are missing in`nginx-full` package due to typo in Makefile.
This should be fixed when https://github.com/openwrt/packages/pull/21985 gets merged.

ronhombre, 2025-10-20:
> For some reason nginx stopped loading modules such as nginx-ubus-module. To fix it on the top level of nginx.conf I included load_module e.g.
> 
> ```
> load_module /usr/lib/nginx/modules/ngx_http_ubus_module.so;
> ```

This should be documented somewhere. Thank you!

Also, the default nginx config sucks for OpenWRT. The UCI stuff seems to be more hassle than anything