openwrt/packages #21254: package/feeds/packages/python-pyproject-hooks [host] failed to build.

snakwu, 2023-06-02:
make[3]: Entering directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-flit-core'
make[3]: Leaving directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-flit-core'
time: package/feeds/packages/python-flit-core/host-compile#0.12#0.06#0.23
make[3]: Entering directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-installer'
make[3]: Leaving directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-installer'
time: package/feeds/packages/python-installer/host-compile#0.10#0.05#0.20
make[3]: Entering directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-packaging'
make[3]: Leaving directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-packaging'
time: package/feeds/packages/python-packaging/host-compile#0.18#0.05#0.22
make[3]: Entering directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-pyproject-hooks'
rm -rf "/home/snakwu/openwrt/openwrt/tmp/host-stage-python-pyproject-hooks"
mkdir -p "/home/snakwu/openwrt/openwrt/tmp/host-stage-python-pyproject-hooks" "/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/stamp"
cd "/home/snakwu/openwrt/openwrt/build_dir/hostpkg/pypi/pyproject_hooks-1.0.0" && ARCH="x86_64" CC="gcc" CCSHARED="gcc -DPIC -fPIC" CXX="g++" LD="gcc" LDSHARED="gcc -shared" CFLAGS="-O2 -I/home/snakwu/openwrt/openwrt/staging_dir/host/include -I/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/include -I/home/snakwu/openwrt/openwrt/staging_dir/target-x86_64_musl/host/include" CPPFLAGS="-I/home/snakwu/openwrt/openwrt/staging_dir/host/include -I/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/include -I/home/snakwu/openwrt/openwrt/staging_dir/target-x86_64_musl/host/include -I/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/include/python3.11" LDFLAGS="-L/home/snakwu/openwrt/openwrt/staging_dir/host/lib -L/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/lib -L/home/snakwu/openwrt/openwrt/staging_dir/target-x86_64_musl/host/lib -lpython3.11 -Wl,-rpath,/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/lib"    /home/snakwu/openwrt/openwrt/staging_dir/hostpkg/bin/python3.11  -m installer --destdir "/home/snakwu/openwrt/openwrt/tmp/host-stage-python-pyproject-hooks" --prefix "" "/home/snakwu/openwrt/openwrt/build_dir/hostpkg/pypi/pyproject_hooks-1.0.0/"/openwrt-build/pyproject_hooks-1.0.0-*.whl
/home/snakwu/openwrt/openwrt/staging_dir/hostpkg/bin/python3.11: No module named installer
make[3]: *** [Makefile:54: /home/snakwu/openwrt/openwrt/staging_dir/hostpkg/stamp/.python-pyproject-hooks_installed] Error 1
make[3]: Leaving directory '/home/snakwu/openwrt/openwrt/feeds/packages/lang/python/python-pyproject-hooks'
time: package/feeds/packages/python-pyproject-hooks/host-compile#0.15#0.06#0.44
    ERROR: package/feeds/packages/python-pyproject-hooks [host] failed to build.
make[2]: *** [package/Makefile:118: package/feeds/packages/python-pyproject-hooks/host/compile] Error 1
make[2]: Leaving directory '/home/snakwu/openwrt/openwrt'
make[1]: *** [package/Makefile:114: /home/snakwu/openwrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt/openwrt'
make: *** [/home/snakwu/openwrt/openwrt/include/toplevel.mk:231: world] Error 2

jefferyto, 2023-06-02:
Try `make dirclean`.

snakwu, 2023-06-02:
ok thanks

closed 2023-06-02 by snakwu

schm0, 2023-06-06:
"/home/snakwu/openwrt/openwrt/build_dir/hostpkg/pypi/pyproject_hooks-1.0.0/"/openwrt-build/pyproject_hooks-1.0.0-*.whl

```
Subject: [PATCH] python3: fix host build

---
 lang/python/python3-host-build.mk | 8 ++++----
 1 file changed, 4 insertions(+), 4 deletions(-)

diff --git a/lang/python/python3-host-build.mk b/lang/python/python3-host-build.mk
index 738d15728..64048249f 100644
--- a/lang/python/python3-host-build.mk
+++ b/lang/python/python3-host-build.mk
@@ -20,7 +20,7 @@ PYTHON3_HOST_INSTALL_VARS?=
 PYTHON3_HOST_WHEEL_NAME?=$(subst -,_,$(if $(PYPI_SOURCE_NAME),$(PYPI_SOURCE_NAME),$(PKG_NAME)))
 PYTHON3_HOST_WHEEL_VERSION?=$(PKG_VERSION)

-PYTHON3_HOST_BUILD_DIR?=$(HOST_BUILD_DIR)/$(PYTHON3_HOST_BUILD_PATH)
+PYTHON3_HOST_BUILD_DIR?=$(HOST_BUILD_DIR)

 PYTHON3_HOST_DIR_NAME:=$(lastword $(subst /,$(space),$(CURDIR)))
@@ -32,7 +32,7 @@ define Py3Host/Compile/Bootstrap
        $(call HostPython3/Run, \
                $(HOST_BUILD_DIR), \
                -m flit_core.wheel \
-                       --outdir "$(PYTHON3_HOST_BUILD_DIR)"/openwrt-build \
+                       --outdir "$(PYTHON3_HOST_BUILD_DIR)/openwrt-build" \
                        "$(PYTHON3_HOST_BUILD_DIR)" \
        )
 endef
@@ -42,7 +42,7 @@ define Py3Host/Compile
                $(HOST_BUILD_DIR), \
                -m build \
                        --no-isolation \
-                       --outdir "$(PYTHON3_HOST_BUILD_DIR)"/openwrt-build \
+                       --outdir "$(PYTHON3_HOST_BUILD_DIR)/openwrt-build" \
                        --wheel \
                        $(foreach setting,$(PYTHON3_HOST_BUILD_CONFIG_SETTINGS),--config-setting=$(setting)) \
                        $(PYTHON3_HOST_BUILD_ARGS) \
@@ -58,7 +58,7 @@ define Py3Host/Install/Installer
                -m installer \
                        --destdir "$(1)" \
                        --prefix "" \
-                       "$(PYTHON3_HOST_BUILD_DIR)"/openwrt-build/$(PYTHON3_HOST_WHEEL_NAME)-$(PYTHON3_HOST_WHEEL_VERSION)-*.whl \
+                       "$(PYTHON3_HOST_BUILD_DIR)/openwrt-build/$(PYTHON3_HOST_WHEEL_NAME)-$(PYTHON3_HOST_WHEEL_VERSION)-py3-none-any.whl" \
                        , \
                $(PYTHON3_HOST_INSTALL_VARS) \
        )
--
2.34.1
```