openwrt/packages #21269: avahi: Add patch no extend line length > 256chars

dMopp, 2023-06-04:
Maintainer: @\nxhack (find it by checking history of the package Makefile)
Environment: (x86/64, —, snapshot)

Description:
There is a small quick fix to extend the line length for avahi filters to 1024 chars instead of 256 (default), see: https://github.com/astlinux-project/astlinux/commit/4e5a506e32735915da0caf27a587e823683b5ae4

this would help me a lot to filter mdns reflector traffic a LOT :)

nxhack, 2023-06-05:
Hi @dMopp 

I have checked the discussion on Upstream.
It looks like it would be better to fix it upstream than to patch it here.

dMopp, 2023-06-06:
Yeah I agree, but it seems to be there is absolute no effort there :(

nxhack, 2023-06-07:
I understand that upstream is moving slowly :)

I also confirmed that debian, buildroot, and alpinelinux packages have fixed themselves, and since there seems to be CVE fixes in the works, I'll see what I can do about it in this package.

nxhack, 2023-06-08:
I sent a pull request.

https://github.com/openwrt/packages/pull/21303

diizzyy, 2023-07-30:
Resolved as of https://github.com/openwrt/packages/commit/779af4d40ccdc0f2a798ee6b6849abb37d202f1b

closed 2023-07-30 by diizzyy