openwrt/packages #21294: xtables-addons fail to build

very20101, 2023-06-07:
linux-armvirt_64
ubuntu 22.04

make[4]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24'
make  all-recursive
make[5]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24'
Making all in extensions
make[6]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions'
Xtables-addons 3.24 - Linux 6.1.31
if [ -n "/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-6.1.31" ]; then make -C /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-6.1.31 M=/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions modules; fi;
make[7]: Entering directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-6.1.31'
  CC [M]  /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions/LUA/xt_LUA_target.o
In file included from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions/LUA/controller.h:24,
                 from /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions/LUA/xt_LUA_target.c:27:
/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions/LUA/lua/lua.h:12:10: fatal error: stddef.h: No such file or directory
   12 | #include <stddef.h>
      |          ^~~~~~~~~~
compilation terminated.
make[9]: *** [scripts/Makefile.build:250: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions/LUA/xt_LUA_target.o] Error 1
make[7]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/linux-6.1.31'
make[8]: *** [scripts/Makefile.build:500: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions/LUA] Error 2
make[6]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions'
make[7]: *** [Makefile:2012: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/extensions] Error 2
make[6]: *** [Makefile:468: modules] Error 2
make[5]: *** [Makefile:625: all-recursive] Error 1
make[4]: *** [Makefile:397: all] Error 2
make[3]: *** [Makefile:177: /workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24/.built] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24'
make[4]: Leaving directory '/workdir/openwrt/build_dir/target-aarch64_generic_musl/linux-armvirt_64/xtables-addons-3.24'
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/net/xtables-addons'
time: package/feeds/packages/xtables-addons/compile#1.02#0.23#1.32
    ERROR: package/feeds/packages/xtables-addons failed to build.
make[2]: *** [package/Makefile:120: package/feeds/packages/xtables-addons/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:114: /workdir/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:232: world] Error 2
Error: Process completed with exit code 2.

ptpt52, 2023-06-11:
same issue with my build.
since kernel update to 6.1, this build failure occr.

ptpt52, 2023-06-11:
```
make[2]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/package/network/utils/iptables'
echo "libxtables" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "libip4tc" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "libip6tc" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "libiptext" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "libiptext6" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "libiptext-nft" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "xtables-nft" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "iptables-nft" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "iptables-mod-conntrack-extra" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "iptables-mod-ipopt" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
echo "ip6tables-nft" >> /mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/iptables.default.install
make[2]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt/package/network/utils/iptables'
time: package/network/utils/iptables/compile#0.16#0.08#0.23
make[2]: Entering directory '/mnt/Data/Sources/openwrt/openwrt-feeds/packages/net/xtables-addons'
rm -f /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/.built
touch /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/.built_check
make  -C /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24 KCFLAGS="-fmacro-prefix-map=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl=target-aarch64_cortex-a72_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/include -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/hostpkg/include -I/mnt/Data/Sources/openwrt/x-wrt/staging_dir/target-aarch64_cortex-a72_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="aarch64-openwrt-linux-musl-" ARCH="arm64" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Jun 10 22:02:23 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/mnt/Data/Sources/openwrt/x-wrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/natflow.symvers /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/natcap.symvers /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/rproxy.symvers /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/nat46.symvers /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/mac80211.symvers /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/symvers/jool.symvers" KERNELRELEASE=6.1.32 DESTDIR="/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/ipkg-install" DEPMOD="/bin/true" all
make[3]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24'
make  all-recursive
make[4]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24'
Making all in extensions
make[5]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions'
Xtables-addons 3.24 - Linux 6.1.32
if [ -n "/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.1.32" ]; then make -C /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.1.32 M=/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions modules; fi;
make[6]: Entering directory '/mnt/Data/Sources/openwrt/x-wrt/tmpfs/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.1.32'
  CC [M]  /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions/LUA/xt_LUA_target.o
In file included from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions/LUA/controller.h:24,
                 from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions/LUA/xt_LUA_target.c:27:
[... cut ...]
time: package/feeds/packages/xtables-addons/compile#0.85#0.32#1.15
    ERROR: package/feeds/packages/xtables-addons failed to build.
package/Makefile:118: recipe for target 'package/feeds/packages/xtables-addons/compile' failed
make[1]: *** [package/feeds/packages/xtables-addons/compile] Error 1
make[1]: Leaving directory '/mnt/Data/Sources/openwrt/x-wrt'
/mnt/Data/Sources/openwrt/x-wrt/include/toplevel.mk:230: recipe for target 'package/feeds/packages/xtables-addons/compile' failed
make: *** [package/feeds/packages/xtables-addons/compile] Error 2
```

Ansuel, 2023-06-11:
Fix incoming :D just a change in 6.1

commit e182b15e (2023-06-11): xtables-addons: fix compilation error on kernel 6.1

Fix compilation error on kernel 6.1.
Fix compilation error:
In file included from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions/LUA/controller.h:24,
                 from /mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions/LUA/xt_LUA_target.c:27:
/mnt/Data/Sources/openwrt/x-wrt/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/xtables-addons-3.24/extensions/LUA/lua/lua.h:12:10: [... cut]
closed 2023-06-11 by Ansuel