openwrt/packages #21300: nginx-full (nginx-all-module): installation issue

iganeshk, 2023-06-07:
Maintainer: @hgl @Ansuel 
Environment:
  ```bash
  root@hostname:~# cat /etc/openwrt_release
  DISTRIB_ID='OpenWrt'
  DISTRIB_RELEASE='SNAPSHOT'
  DISTRIB_REVISION='r23293-98e6ea32a4'
  DISTRIB_TARGET='mvebu/cortexa9'
  DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
  DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r23293-98e6ea32a4'
  DISTRIB_TAINTS=''
  ```
## Description: Unable to install nginx-full (nginx-all-module) on a fresh snapshot sysupgrade

Plausible issue:
```makefile 
define Package/nginx-ssl
  $(Package/nginx/default)
  TITLE += with SSL support
  VARIANT:=ssl
  DEPENDS+= +NGINX_PCRE:libpcre \
	+NGINX_PCRE:nginx-ssl-util +!NGINX_PCRE:nginx-ssl-util-nopcre \
	+NGINX_HTTP_GZIP:zlib +NGINX_DAV:libxml2
  EXTRA_DEPENDS:=nginx-ssl-util$(if $(CONFIG_NGINX_PCRE),,-nopcre) (>=1.5-1) (<2)
  CONFLICTS:=nginx-full
endef

define Package/nginx-full
  $(Package/nginx/default)
  TITLE += with ALL module selected
  DEPENDS+=+libpcre +nginx-ssl-util +zlib +libxml2 \
    $(foreach m,$(PKG_MOD_ALL),+nginx-mod-$(m))
  EXTRA_DEPENDS:=nginx-ssl-util (>=1.5-1) (<2)
  VARIANT:=full
  PROVIDES += nginx-ssl
endef
```

PS: Shouldn't there be a placeholder package `nginx-all-module` which points to `nginx-full` when a package is deprecated?

Log:
  ```bash
  root@hostname:~# opkg list-installed
  addrwatch - 1.0.2-3
  addrwatch-mysql - 1.0.2-3
  avahi-dbus-daemon - 0.8-7
  avahi-utils - 0.8-7
  base-files - 1532-r23293-98e6ea32a4
  busybox - 1.36.1-1
  ca-bundle - 20230311-1
  cgi-io - 2022-08-10-901b0f04-21
  coreutils - 9.3-1
  coreutils-cp - 9.3-1
  coreutils-dd - 9.3-1
  curl - 8.1.1-1
  dbus - 1.13.18-12
  ddns-scripts - 2.8.2-37
  ddns-scripts-cloudflare - 2.8.2-37
  ddns-scripts-services - 2.8.2-37
  dnsmasq - 2.89-4
  dropbear - 2022.82-5
  ethtool - 6.3-1
  firewall4 - 2023-03-23-04a06bd7-1
  fstools - 2023-02-28-bfe882d5-1
  fwtool - 2019-11-12-8f7fe925-1
  getrandom - 2022-08-13-4c7b720b-2
  hostapd-common - 2023-03-29-bb945b98-1.2
  htop - 3.2.2-1
  iperf - 2.1.9-1
  iperf3 - 3.13-1
  irqbalance - 1.9.2-2
  iw - 5.19-1
  iwinfo - 2023-05-17-c9f5c3f7-1
  jansson4 - 2.14-3
  jshn - 2023-05-23-75a3b870-1
  jsonfilter - 2018-02-04-c7e938d6-1
  kernel - 5.15.114-1-e540a09a2e0d39e0201bb36712ecc7f6
  kmod-bluetooth - 5.15.114-1
  kmod-btmrvl - 5.15.114-1
  kmod-cfg80211 - 5.15.114+6.1.24-4
  kmod-crypto-acompress - 5.15.114-1
  kmod-crypto-aead - 5.15.114-1
  kmod-crypto-ccm - 5.15.114-1
  kmod-crypto-cmac - 5.15.114-1
  kmod-crypto-crc32c - 5.15.114-1
  kmod-crypto-ctr - 5.15.114-1
  kmod-crypto-ecb - 5.15.114-1
  kmod-crypto-ecdh - 5.15.114-1
  kmod-crypto-gcm - 5.15.114-1
  kmod-crypto-gf128 - 5.15.114-1
  kmod-crypto-ghash - 5.15.114-1
  kmod-crypto-hash - 5.15.114-1
  kmod-crypto-hmac - 5.15.114-1
  kmod-crypto-kpp - 5.15.114-1
  kmod-crypto-lib-chacha20 - 5.15.114-1
  kmod-crypto-lib-chacha20poly1305 - 5.15.114-1
  kmod-crypto-lib-curve25519 - 5.15.114-1
  kmod-crypto-lib-poly1305 - 5.15.114-1
  kmod-crypto-manager - 5.15.114-1
  kmod-crypto-null - 5.15.114-1
  kmod-crypto-rng - 5.15.114-1
  kmod-crypto-seqiv - 5.15.114-1
  kmod-crypto-sha512 - 5.15.114-1
  kmod-gpio-button-hotplug - 5.15.114-3
  kmod-hid - 5.15.114-1
  kmod-input-core - 5.15.114-1
  kmod-input-evdev - 5.15.114-1
  kmod-lib-crc-ccitt - 5.15.114-1
  kmod-lib-crc16 - 5.15.114-1
  kmod-lib-crc32c - 5.15.114-1
  kmod-lib-lzo - 5.15.114-1
  kmod-mac80211 - 5.15.114+6.1.24-4
  kmod-mmc - 5.15.114-1
  kmod-mwifiex-sdio - 5.15.114+6.1.24-4
  kmod-mwlwifi - 5.15.114+2023-04-29-6a436714-1
  kmod-nf-conntrack - 5.15.114-1
  kmod-nf-conntrack6 - 5.15.114-1
  kmod-nf-flow - 5.15.114-1
  kmod-nf-log - 5.15.114-1
  kmod-nf-log6 - 5.15.114-1
  kmod-nf-nat - 5.15.114-1
  kmod-nf-reject - 5.15.114-1
  kmod-nf-reject6 - 5.15.114-1
  kmod-nfnetlink - 5.15.114-1
  kmod-nft-core - 5.15.114-1
  kmod-nft-fib - 5.15.114-1
  kmod-nft-nat - 5.15.114-1
  kmod-nft-offload - 5.15.114-1
  kmod-nls-base - 5.15.114-1
  kmod-ppp - 5.15.114-1
  kmod-pppoe - 5.15.114-1
  kmod-pppox - 5.15.114-1
  kmod-regmap-core - 5.15.114-1
  kmod-scsi-core - 5.15.114-1
  kmod-slhc - 5.15.114-1
  kmod-udptunnel4 - 5.15.114-1
  kmod-udptunnel6 - 5.15.114-1
  kmod-usb-core - 5.15.114-1
  kmod-wireguard - 5.15.114-1
  libacl - 2.3.1-2
  libatomic1 - 12.3.0-4
  libattr - 2.5.1-1
  libavahi-client - 0.8-7
  libavahi-dbus-support - 0.8-7
  libblobmsg-json20220927 - 2022-09-27-ea560134-1
  libblobmsg-json20230523 - 2023-05-23-75a3b870-1
  libc - 1.2.4-4
  libcap - 2.69-1
  libcurl4 - 8.1.1-1
  libdaemon - 0.14-5
  libdbus - 1.13.18-12
  libevdev - 1.13.0-1
  libevent2-7 - 2.1.12-1
  libexpat - 2.5.0-1
  libgcc1 - 12.3.0-4
  libgd - 2.3.3-1
  libgdbm - 1.21-2
  libiperf3 - 3.13-1
  libiwinfo-data - 2023-05-17-c9f5c3f7-1
  libiwinfo20230121 - 2023-05-17-c9f5c3f7-1
  libjpeg-turbo - 2.1.4-2
  libjson-c5 - 0.16-3
  libjson-script20220927 - 2022-09-27-ea560134-1
  libjson-script20230523 - 2023-05-23-75a3b870-1
  liblua5.1.5 - 5.1.5-10
  liblucihttp-lua - 2023-03-15-9b5b683f-1
  liblucihttp-ucode - 2023-03-15-9b5b683f-1
  liblucihttp0 - 2023-03-15-9b5b683f-1
  libmariadb3 - 3.1.18-1
  libmbedtls12 - 2.28.3-1
  libmnl0 - 1.0.5-1
  libncurses6 - 6.4-2
  libnftnl11 - 1.2.5-2
  libnghttp2-14 - 1.51.0-1
  libnl-tiny2023-04-02 - 2023-04-02-11b7c5f0-1
  libopenssl-conf - 3.0.8-10
  libopenssl3 - 3.0.8-10
  libpcap1 - 1.10.4-1
  libpcre - 8.45-5
  libpng - 1.6.39-1
  libpthread - 1.2.4-4
  libqrencode - 4.1.1-1
  librt - 1.2.4-4
  libstdcpp6 - 12.3.0-4
  libubox20220927 - 2022-09-27-ea560134-1
  libubox20230523 - 2023-05-23-75a3b870-1
  libubus-lua - 2022-06-15-9913aa61-1
  libubus20220615 - 2022-06-15-9913aa61-1
  libuci-lua - 2023-03-05-04d0c46c-1
  libuci20130104 - 2023-03-05-04d0c46c-1
  libuclient20201210 - 2023-04-13-007d9454-1
  libucode20220812 - 2023-04-03-51638672-1
  libusb-1.0-0 - 1.0.26-3
  libustream-mbedtls20201210 - 2023-02-25-498f6e26-1
  libuuid1 - 2.39-2
  libuv1 - 1.44.1-1
  libwebp - 1.3.0-1
  libxml2 - 2.11.4-1
  logd - 2022-08-13-4c7b720b-2
  lua - 5.1.5-10
  luci-app-advanced-reboot - 1.0.1-8
  luci-app-argon-config - 0.8-beta
  luci-app-ddns - git-23.118.79453-d9dcf98
  luci-app-firewall - git-23.095.34229-148759a5
  luci-app-opkg - git-23.009.82915-ec3aac4
  luci-base - git-23.118.79121-6fb185f
  luci-compat - git-22.297.83017-673f382
  luci-lib-base - git-22.308.54612-9118452
  luci-lib-ip - git-20.250.76529-62505bd
  luci-lib-ipkg - git-18.318.71164-4bbe325
  luci-lib-json - git-18.184.37844-cde13dc
  luci-lib-jsonc - git-22.079.54693-45f411b
  luci-lib-nixio - git-22.222.71555-88b9088
  luci-lua-runtime - git-23.119.80867-2a9c3a3
  luci-mod-admin-full - git-19.253.48496-3f93650
  luci-mod-network - git-23.150.26360-9b12a48
  luci-mod-rpc - git-21.020.56896-af422b1
  luci-mod-status - git-23.063.40670-8c7c56f
  luci-mod-system - git-23.118.78765-58f7b27
  luci-proto-ipv6 - git-21.148.48881-79947af
  luci-proto-ppp - git-21.158.38888-88b9d84
  luci-proto-wireguard - git-23.073.81399-59a6f06
  luci-theme-argon-master - 2.2.9.4
  luci-theme-bootstrap - git-23.085.34270-d94a728
  luci-theme-material - git-23.088.30860-f464199
  mtd - 26
  mwifiex-sdio-firmware - 20230515-1
  mwlwifi-firmware-88w8964 - 2023-04-29-6a436714-1
  nano - 7.2-2
  ncdu - 1.18-1
  netifd - 2023-06-04-ec9dba72-1
  nftables-json - 1.0.7-1
  nginx-ssl-util - 1.6-17
  nginx-util - 1.6-17
  odhcp6c - 2023-05-12-bcd28363-20
  odhcpd-ipv6only - 2023-04-05-40ab806b-1
  openssh-client - 9.3p1-3
  openssh-keygen - 9.3p1-3
  openssh-sftp-server - 9.3p1-3
  openssl-util - 3.0.8-10
  openwrt-keyring - 2022-03-25-62471e69-2
  opkg - 2022-02-24-d038e5b6-2
  ppp - 2.4.9.git-2021-01-04-4
  ppp-mod-pppoe - 2.4.9.git-2021-01-04-4
  
  root@hostname:~# opkg list-installed | grep nginx
  
  root@hostname:~# opkg install nginx-full
  Installing nginx-full (1.24.0-2) to root...
  Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/nginx-full_1.24.0-2_arm_cortex-a9_vfpv3-d16.ipk
  Installing nginx-util (1.6-17) to root...
  Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/nginx-util_1.6-17_all.ipk
  Installing nginx-ssl-util (1.6-17) to root...
  Downloading https://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3-d16/packages/nginx-ssl-util_1.6-17_arm_cortex-a9_vfpv3-d16.ipk
  Installing nginx-ssl (1.24.0-2) to root...
  Configuring nginx-util.
  Configuring nginx-ssl-util.
  Collected errors:
   * resolve_conffiles: Existing conffile /etc/nginx/uci.conf.template is different from the conffile in the new package. The new conffile will be placed at /etc/nginx/uci.conf.template-opkg.
   * check_conflicts_for: The following packages conflict with nginx-ssl:
   * check_conflicts_for:         nginx-full *
   * opkg_install_cmd: Cannot install package nginx-full.
  ```

Ansuel, 2023-07-05:
@brackenhill-mob the conflict issue in theory it's not caused by the package rename...

to fix this just remove nginx-ssl and install nginx-full. It's not really a bug. (or I could be wrong and I'm missing something)

iganeshk, 2023-07-05:
> @brackenhill-mob the conflict issue in theory it's not caused by the package rename...
> 
> to fix this just remove nginx-ssl and install nginx-full. It's not really a bug. (or I could be wrong and I'm missing something)

fix doesn't work, as you can see from the above log, there's no nginx package, it still fails to install

Ansuel, 2023-07-05:
I bisected the bug but I need more testing with this. Anyway I confirmed the bug and I know the problem so this will be fixed and backported :D

closed 2023-07-07 by Ansuel

peterwillcn, 2024-04-22:
@Ansuel  Keep the principle simple, merge nginx ssl and nginx full, and remove the dependencies of nginx ssl util and nginx util

commit 3d0183e9 (2023-07-07): nginx: don't install all module for FULL variant

We currently have a more or less circular dependency with nginx ssl and
full variant.

FULL variant depends on every nginx module. Every nginx module depends
on nginx-ssl.

Since nginx-full depends on an nginx module, nginx-ssl is installed as
module depends on it and then the installation fails as nginx-full
conflicts with nginx-ssl.

nginx-full in it's meaning is nginx built with every config selected and
it should not have module as dependency. In fact an user should always
install them separetly as while other things, local modification to  [... cut]
commit 42cf7d0e (2023-07-07): nginx: don't install all module for FULL variant

We currently have a more or less circular dependency with nginx ssl and
full variant.

FULL variant depends on every nginx module. Every nginx module depends
on nginx-ssl.

Since nginx-full depends on an nginx module, nginx-ssl is installed as
module depends on it and then the installation fails as nginx-full
conflicts with nginx-ssl.

nginx-full in it's meaning is nginx built with every config selected and
it should not have module as dependency. In fact an user should always
install them separetly as while other things, local modification to  [... cut]