openwrt/packages #21307: [adblock] Do not store pcap files on reports directory, use /tmp

schuettecarsten, 2023-06-08:
Adblock allows to specify a "Report Directory".
This directory should be used for calculated report data, not for the the tcpdump pcap files.

dibdot, 2023-06-17:
Sounds weird - the default is on /tmp and you do not provide any reasoning. Closing this.

closed 2023-06-17 by dibdot

schuettecarsten, 2023-06-18:
Sorry for long delay @debdot.

Problem is that you can move the directory to your attached USB device or somewhere else. That might be useful to persist the report results. But in this case, the pcap files are also moved to the USB device. And that's not a good idea. I suggest to separate the reports result folder and the temp data folder into two different settings.

dibdot, 2023-06-18:
Moving this (incl. the pcap files) is useful to save RAM on the limited router.