openwrt/packages #21318: https-dns-proxy bootstrap DNS not working with interfaces other than wan and wan6

professor-jonny, 2023-06-09:
Maintainer: Maintainer: @stangri
Environment: (ipq40xx, Wallystech DR4029, 10/6/23 SNAPSHOT, r23115-2d90531153)

Description:
Bootstrap DNS fails to work on interfaces other than wan and wan6 interfaces as it is hard coded in the initscript I believe.
because of this bootstrap won't work with wwan devices or nonstandard interface names.

Becomes an issue if bootstrap does not work to obtain addresses for NTP server at boot breaking ability for DNS run over Wireguard vpn as it requires the clock time set correctly.

brada4, 2023-06-09:
https://github.com/openwrt/openwrt/pull/11947 solution s pending

stangri, 2023-06-11:
> Bootstrap DNS fails to work on interfaces other than wan and wan6 interfaces as it is hard coded in the initscript I believe.

There's nothing hardcoded to access bootstrap servers thru wan/wan6 only.

> Becomes an issue if bootstrap does not work to obtain addresses for NTP server at boot breaking ability for DNS run over Wireguard vpn as it requires the clock time set correctly.

As @jow- posted in the linked PR, add the dnsmasq config options to ensure the NTP servers are accessed without encryption.

brada4, 2023-06-11:
Linked PR is to streamline the chicken and egg dilemma configuration, while indeed manual workaround will mostly work.

professor-jonny, 2023-06-12:
I think the linked PR or suggested edit by jow will fix my bootstrap DNS problem ill wait for an upstream fix, for the time  I will start https_dns_proxy  after the NTP sets the time.

I noticed wan and wan6 referenced in the init scrip I'm guessing this is to restart https-dns-proxy on a wan change?

I am using mwan3 to control failover on a wwan connection will there be any compatibility issues with inter operability with a nonstandard named wan interface such as wwan_4 wwan_6 as there is/has been with nonstandard lan interfaces names?

brada4, 2023-06-12:
Hardcoded wan name https://github.com/openwrt/packages/blob/1ea13ed8a168459568e1ce831db3c1ddf63b8905/net/https-dns-proxy/files/https-dns-proxy.hotplug.iface#L3 does not do anything for initial start, problem is that device needs DNS service to set correct time (get IP-s of NTP servers) , before https-dns-proxy will be able to validate upstream SSL certificates.
Restart on hotplug might not be needed at all.

The PR contains workaround on how to fix that up using well known DNS service to resolve NTP server, while the PR proposes using default DNS to do that seamlessly.

stangri, 2024-01-29:
@professor-jonny are you still using `https-dns-proxy`? I rewrote the boot() a while ago and if there's dns resolution failure on boot, the https-dns-proxy should stop and come back up on WAN update. Did that fix the boot-up status for you?

professor-jonny, 2024-01-29:
I'm currently using httpdnsproxy and everything seems to work, I'm really happy with it thanks :-)

I am using a script to restart wan interface if my link goes down as I'm using an Ethernet connected 4g mobile router.
Httpsdns proxy comes back up instantly unlike previous when i had to stop and restart your service.
I don't have to edit the script to alter my nonstandard interface names after every update, but I do believe it only picks up the first two interface's from memory.

The only thing I'd really want is wildcard mac support to block oui manufactures and random mac addresses but that is related to dnsmasq no being inline with the features upstream.

professor-jonny, 2024-01-29:
I do admit I have a 4g minipci card in my router and I gave up using it with openwrt and use an external oem modem.
I had issues with DNS resolution every time the QMI wireless service went down.
In all honesty I had issues with the internet not reconnecting automatically and got fed up, it could been that being the issue, but it did look like "WAN" was hard code in your script as opposed to detecting the wan interface.
qmi dynamically creates an interface on connection.

stangri, 2024-01-29:
Seems I can close it then!

closed 2024-01-29 by stangri