openwrt/packages #21364: mwan3: route deletion occurs if only one of multiple routes with on source address selector is deleted

l-jonas, 2023-06-14:
Maintainer: @feckert @aaronjg 
Environment: OpenWrt 22.03

Description: Already reported at <https://forum.openwrt.org/t/mwan3-feedback-on-ipv6-and-missing-features/29483/41>, but without any response yet:

I'm having issues with default routes that get lost.

mwan3 is configured to remove the source address selector when copying the routes from the main table to the interface specific ones. This works as it intended.

When routes from the main table are removed, the same happens: The source address selectors is removed and this adjusted route is removed from the interface specific tables. This works, too.

The problem occurs when the main routing table contains multiple different routes that are equal after removing the source address selector. Then, removing one of them from the main table deletes the derived routes from all of them from the interface specific tables.

If this happens with the default route, then the interface specific tables contain no default route and all internet traffic that should use this interface is not forwarded anymore.

And yes, this happens with the default route for me. When my IPv6 address changes, new routes are added and some old default routes are removed some time later.

I'm using the following workaround now: modify /usr/sbin/mwan3rtmon; between route_line=... and handle_route, add the following:

```
if [ "$action" = "del" ]; then
	if mwan3_get_routes | grep -qxF "$route_line"; then
		LOG debug "deleted but route still exists - $route_line"
		return
	fi
fi
```

This does not correctly update the list of connected networks, but it's only a workaround.

However, I think that this is very inefficient with calling mwan3_get_routes. I think that it would be simpler to use ip monitor just as a trigger to regenerate the interface specific tables - or use it as a trigger for a complete regeneration in case of a removed route that has a source address selector.

feckert, 2023-06-28:
> The problem occurs when the main routing table contains multiple different routes that are equal after removing the source address selector. Then, removing one of them from the main table deletes the derived routes from all of them from the interface specific tables.

What do you mean with `multiple different`. Can you give me an `ip route show` dump ?

l-jonas, 2023-06-28:
The relevant beginning of the output from ``ip -6 route show``. Note that the first two groups of the source addresses and the whole gateway addresses (after via) were replaced for privacy reasons, but there relations were kept.

```
default from 2000:1:a5ab:6400::/64 via fe80::1 dev eth0.4 proto static metric 512 pref medium
default from 2000:1:a5ab:64f0::/60 via fe80::1 dev eth0.4 proto static metric 512 pref medium
default from 2000:1:b1ce:9100::/64 via fe80::1 dev eth0.4 proto static metric 512 pref medium
default from 2001:2:8d20:a900::/64 via fe80::2 dev eth0.5 proto static metric 512 pref medium
default from 2001:2:8d20:a9f0::/60 via fe80::2 dev eth0.5 proto static metric 512 pref medium
```

Without the source address selectors (the transformation that mwan3 does internally), it looks like this:

```
default via fe80::1 dev eth0.4 proto static metric 512 pref medium
default via fe80::1 dev eth0.4 proto static metric 512 pref medium
default via fe80::1 dev eth0.4 proto static metric 512 pref medium
default via fe80::2 dev eth0.5 proto static metric 512 pref medium
default via fe80::2 dev eth0.5 proto static metric 512 pref medium
```

``ip a`` shows that the address within ``2000:1:b1ce:9100::/64`` is deprecated. The system will remove
the address and the related route at some time in the future - but it did not remove it before the new
address and route were created.

l-jonas, 2023-08-10:
@feckert Can you provide any feedback on this?

feckert, 2023-08-16:
I don't use the mwan3 with IPv6.

The question is, is the mwan3 doing something wrong? If the source address is part of the route decision, then it should also be added to the interface routing tables. I would have to take a closer look in the code. I would be willing to make the change if it makes sense and doesn't break other setups.

 Please open a pull request with your suggested changes to include the source address in the routing table with rtmon.

l-jonas, 2023-08-16:
> If the source address is part of the route decision, then it should also be added to the interface routing tables.

The source address is used to decide which interface should be used (like in the routing table that I posted). That is the correct behavior when not using mwan3. However, with mwan3, the goal is to eventually use another interface so that should be ignored (this is what mwan3 does currently).

To make the packet forwardable, its src IP address must be rewritten to match the interface (snat/masquerade). That can be done in the postrouting chain of the firewall that is executed after the routing decision was made. Due to that, the routing must use the interface mwan3 wants to use although the routing with source address selectors would not do it. So it's the easiest solution to strip the source address selectors like mwan3 does it already.

The problem is that rtmon deletes routes when it should not do it because it generalizes the removed routes to ones without src address filter - as I already described in the first post. To quote my possible solution from the first post:

> However, I think that this is very inefficient with calling mwan3_get_routes. I think that it would be simpler to use ip monitor just as a trigger to regenerate the interface specific tables - or [...]

feckert, 2023-08-17:
Please prepare a pullrequest for this issue. So I can see what you want to do. And the I could verify that this change does not break any setups.

feckert, 2024-02-20:
Fixed with commit https://github.com/openwrt/packages/commit/e119a57e97e894f0636df999d82807ae5c494030.

closed 2024-02-20 by feckert