openwrt/packages #21377: adblock feature to allow dhcp hostname lookup

professor-jonny, 2023-06-15:
Maintainer: @dibdot
Environment: 22.03.3 snapshot  ipq4029 (wallystech dr4029)

Description: 
Is it possible that we could add the ability to view hostnames in the adblock interface as opposed to IP addresses maybe a toggle or both some how?

dibdot, 2023-06-17:
You mean DNS reporting? That's already working, e.g.
[image]
[image]

It depends heavily on your DNS backend configuration and the `adb_represolve` option which is disabled by default due to performance reasons (see readme).

professor-jonny, 2023-06-20:
ok i will try that my clients only display an ip address despite being named hosts in the dhcp config.

dibdot, 2023-06-20:
Unbound? If so, add domain entries as well...

professor-jonny, 2023-06-20:
I have dual dnsmasq instances and that setting is enabled in my adblock config.
[image]
[image]
[image]

closed 2025-10-19 by dibdot