openwrt/packages #21392: mwan3 双wan口负载均衡，从外网访问高跃点的wan口udp数据没有原路返回，udp数据流向了低跃点wan口导致丢包。

dongc123, 2023-06-17:
[image]
固件版本OpenWrt 23.05.0-rc1 平台x86/64 内核版本5.15.114 mwan3把wan1和wan2做了负载均衡，wan1 ip:175.0.82.xxx 跃点30     wan2 ip:175.0.80.xxx 跃点20   当upd数据包从wan1进入后，从wan2流出没有原路返回，上图是wan1和wan2抓包。

feckert, 2023-09-20:
The discussions will take place in English.

250king, 2023-11-25:
I can make a translate. I think maybe I have the same problem because many classmates are always attached with network bad when they play CSGO (need UDP). Maybe it is based on UDP problem.

Firmware Version: OpenWrt 23.05.0-rc1
Target Platform: x86/64
Kernel Version: 5.15.114

His router has two interfaces are wan1 and wan2, and all of them are used for mwan3.
The gateway metric of wan1 is 30. The wan2 is 20.
When he tried to send a UDP packet throght the wan1. The packet is backed from the wan2 instead of wan1. The image is shown the situtation.