openwrt/packages #21393: mwan3 dual wan port load balancing, accessing the udp data of the high-hop wan port from the external network does not return the original route, and the udp data flows to the low-hop wan port, resulting in packet loss.

dongc123, 2023-06-17:
[image]
Firmware version OpenWrt 23.05.0-rc1 platform x86/64 kernel version 5.15.114 mwan3 load balances wan1 and wan2, wan1 ip: 175.0.82.xxx hop 30 wan2 ip: 175.0.80.xxx hop 20 when After the upd data packet enters from wan1, it flows out from wan2 without going back the same way. The picture above shows the packets captured by wan1 and wan2.

ptpt52, 2023-08-24:
ip route show table 1
ip route show table 2
ip rule
ip route
ifconfig wan1
ifconfig wan2

post all these output

feckert, 2023-09-20:
No response from the author to the query back, so I am closing the issue.

closed 2023-09-20 by feckert