openwrt/packages #21400: haproxy: fail to build for mipsel_24kc_24kf

MoetaYuko, 2023-06-17:
Maintainer: @gladiac
Environment: mipsel_24kc_24kf, 23.05 & snapshot

Description:

```
make TARGET=linux-musl -C /home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc+24kf_musl/haproxy-ssl/haproxy-2.6.13 DESTDIR="/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc+24kf_musl/haproxy-ssl/haproxy-2.6.13/ipkg-install" CC="mipsel-openwrt-linux-musl-gcc" PCREDIR="/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/staging_dir/target-mipsel_24kc+24kf_musl/usr/" USE_LUA=1 LUA_LIB_NAME="lua5.3" LUA_INC="/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/staging_dir/target-mipsel_24kc+24kf_musl/usr/include/lua5.3" LUA_LIB="/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/staging_dir/target-mipsel_24kc+24kf_musl/usr/lib" SMALL_OPTS="-DBUFSIZE=16384 -DMAXREWRITE=1030 -DSYSTEM_MAXCONN=165530" USE_ZLIB=1 USE_PCRE=1 USE_PCRE_JIT=1 USE_PTHREAD_PSHARED=1 USE_LIBATOMIC=1 USE_PROMEX=1 VERSION="2.6.13" SUBVERS="-1" VERDATE="2023/06/14" IGNOREGIT=1 USE_OPENSSL=1 ADDLIB="-lcrypto -lm" CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -ffile-prefix-map=/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc+24kf_musl/haproxy-ssl/haproxy-2.6.13=haproxy-2.6.13 -mips16 -minterlink-mips16 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fno-strict-aliasing -Wdeclaration-after-statement -Wno-unused-label -Wno-sign-compare -Wno-unused-parameter -Wno-clobbered -Wno-missing-field-initializers -Wno-cast-function-type -Wno-address-of-packed-member -Wtype-limits -Wshift-negative-value -Wshift-overflow=2 -Wduplicated-cond -Wnull-dereference -fwrapv -fasynchronous-unwind-tables -Wno-null-dereference" LD="mipsel-openwrt-linux-musl-gcc" LDFLAGS="-L/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/staging_dir/toolchain-mipsel_24kc+24kf_gcc-12.3.0_musl/usr/lib -L/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/staging_dir/toolchain-mipsel_24kc+24kf_gcc-12.3.0_musl/lib -znow -zrelro"
make[3]: Entering directory '/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/build_dir/target-mipsel_24kc+24kf_musl/haproxy-ssl/haproxy-2.6.13'
  CC      src/version.o
  LD      haproxy
/home/xxx/openwrt/openwrt-sdk-pistachio-generic_gcc-12.3.0_musl.Linux-x86_64/staging_dir/toolchain-mipsel_24kc+24kf_gcc-12.3.0_musl/bin/../lib/gcc/mipsel-openwrt-linux-musl/12.3.0/../../../../mipsel-openwrt-linux-musl/bin/ld: BFD (GNU Binutils) 2.40.0 assertion fail elfxx-mips.c:11278
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:1011: haproxy] Error 1
```

MoetaYuko, 2023-06-17:
setting `PKG_BUILD_FLAGS:=no-mips16` fixes the error, unsure if correct tho

MoetaYuko, 2023-06-20:
seems to be a binutils bug, builds fine with binutils 2.37 (default version of 22.03)

Ansuel, 2024-03-20:
@MoetaYuko can you test this?

https://gist.github.com/Ansuel/f689afa59b211ed546effd682859d9d0 (binutils patch)

MoetaYuko, 2024-03-21:
it does fix linking

MoetaYuko, 2024-04-04:
@Ansuel The issue is marked fixed for binutils 2.43, but the patches are huge and thus could be non-trivial to backport. What's your plan for OpenWrt stable?