openwrt/packages #21401: adblock adguard_tracking incorrect size marking

nshopik, 2023-06-17:
# Issue template (remove lines from top till here)

Maintainer: @\<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWrt version)

Description:

```
Format code blocks by wrapping them with pairs of ```
```
adguard_tracking source size market as S but size of that file 13,6M, which fit into XL category.

dibdot, 2023-06-18:
Please check debug logs to see the really used entries after adblock processing, e.g.:

```
Sun Jun 18 14:27:45 2023 user.debug adblock-4.1.5[16571]: f_list   ::: name: adguard_tracking, mode: download, cnt: 362803, in_rc: 0, out_rc: 0
Sun Jun 18 14:27:45 2023 user.debug adblock-4.1.5[16571]: f_list   ::: name: adguard_tracking, mode: backup, cnt: 362803, in_rc: 0, out_rc: 0
Sun Jun 18 14:27:47 2023 user.debug adblock-4.1.5[16571]: f_list   ::: name: -, mode: merge, cnt: 362803, in_rc: 4, out_rc: 0
Sun Jun 18 14:27:50 2023 user.debug adblock-4.1.5[16571]: f_tld    ::: source: /tmp/tmp.oFBfHO/adb_list.overall, cnt: 362803, cnt_tld: 51382
Sun Jun 18 14:27:51 2023 user.debug adblock-4.1.5[16571]: f_list   ::: name: -, mode: final, cnt: 51382, in_rc: 4, out_rc: 0
```
This means that adblock has compressed this list of 362803 entries to 51382 (top level domain compression). Depending on further processed lists, the number may be reduced further (omission of duplicates). I think "S" is still an appropriate value.

closed 2023-06-18 by dibdot

nshopik, 2023-06-19:
Yeah, though I started getting OOB for this list on 128MB device. And I assume because of its uncompressed size, even though I agree removing duplicates its not that big.