openwrt/packages #21448: dockerd: breaks nat loopback rooting

asdil12, 2023-06-27:
Maintainer: G-M0N3Y-2503
Environment: bananapi-r3, aarch64, r22400-1558bbd116

Description:
When dockerd is installed and running (even without any containers started),
there are issues reaching hosts via port forwardings.
It works fine to reach a forwarded host from outside of the network (WAN), but from inside of the network (LAN) it doesn't work.
I don't even get a SYNACK or connection refused.

```
root@saphira2:/etc/config# cat network

config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'sfp2'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ip6assign '60'
	option defaultroute '0'
	list ipaddr '10.131.0.8/16'

config device
	option name 'br-wan'
	option type 'bridge'
	list ports 'eth1'
	list ports 'wan'

config device
	option name 'eth1'
	option macaddr 'XX:XX:XX:XX:75:33'

config interface 'wan'
	option device 'br-wan'
	option proto 'pppoe'
	option username 'XXXXX'
	option password 'XXXXX'
	option ipv6 'auto'

config device
	option name 'wan'
	option macaddr 'XX:XX:XX:XX:75:33'

config device
	option type '8021q'
	option ifname 'br-wan'
	option vid '7'
	option name 'br-wan.7'

config interface 'debug'
	option proto 'static'
	option device 'br-wan'
	option gateway '10.131.0.8'
	list dns '1.1.1.1'
	list ipaddr '10.10.1.2/24'

config interface 'foo'
	option proto 'wireguard'
	option private_key 'XXXX='
	list addresses '10.127.2.131/24'

config wireguard_foo
	option description 'XXXX.de'
	option public_key 'XXXX='
	option preshared_key 'XXXX='
	list allowed_ips '10.127.0.0/16'
	list allowed_ips '10.128.0.0/9'
	list allowed_ips '172.22.0.0/24'
	option endpoint_host 'XXXX.de'
	option endpoint_port '51821'
	option persistent_keepalive '25'
	option route_allowed_ips '1'

config interface 'debug_fritz'
	option proto 'dhcp'
	option device 'br-wan'
	option defaultroute '0'
	option peerdns '0'
	option delegate '0'

config interface 'iost'
	option proto 'static'
	list ipaddr '172.19.0.8/24'
	option device 'br-lan.3'

config device
	option type '8021q'
	option ifname 'br-lan'
	option vid '3'
	option name 'br-lan.3'

config interface 'docker'
	option device 'docker0'
	option proto 'none'
	option auto '0'

config device
	option type 'bridge'
	option name 'docker0'
```

```
root@saphira2:/etc/config# cat firewall

config defaults
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option synflood_protect '1'

config zone
	option name 'lan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	list network 'lan'

config zone
	option name 'wan'
	option input 'REJECT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	list network 'wan'
	list network 'debug'
	list network 'debug_fritz'

config forwarding
	option src 'lan'
	option dest 'wan'

config rule
	option name 'Allow-DHCP-Renew'
	option src 'wan'
	option proto 'udp'
	option dest_port '68'
	option target 'ACCEPT'
	option family 'ipv4'

config rule
	option name 'Allow-Ping'
	option src 'wan'
	option proto 'icmp'
	option icmp_type 'echo-request'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-IGMP'
	option src 'wan'
	option proto 'igmp'
	option family 'ipv4'
	option target 'ACCEPT'

config rule
	option name 'Allow-DHCPv6'
	option src 'wan'
	option proto 'udp'
	option dest_port '546'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-MLD'
	option src 'wan'
	option proto 'icmp'
	option src_ip 'fe80::/10'
	list icmp_type '130/0'
	list icmp_type '131/0'
	list icmp_type '132/0'
	list icmp_type '143/0'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Input'
	option src 'wan'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	list icmp_type 'router-solicitation'
	list icmp_type 'neighbour-solicitation'
	list icmp_type 'router-advertisement'
	list icmp_type 'neighbour-advertisement'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-ICMPv6-Forward'
	option src 'wan'
	option dest '*'
	option proto 'icmp'
	list icmp_type 'echo-request'
	list icmp_type 'echo-reply'
	list icmp_type 'destination-unreachable'
	list icmp_type 'packet-too-big'
	list icmp_type 'time-exceeded'
	list icmp_type 'bad-header'
	list icmp_type 'unknown-header-type'
	option limit '1000/sec'
	option family 'ipv6'
	option target 'ACCEPT'

config rule
	option name 'Allow-IPSec-ESP'
	option src 'wan'
	option dest 'lan'
	option proto 'esp'
	option target 'ACCEPT'

config rule
	option name 'Allow-ISAKMP'
	option src 'wan'
	option dest 'lan'
	option dest_port '500'
	option proto 'udp'
	option target 'ACCEPT'

config zone
	option name 'foo'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	list network 'foo'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'ikarus ssh'
	list proto 'tcp'
	option src 'wan'
	option src_dport 'XXX22'
	option dest_ip '10.131.0.10'
	option dest_port '22'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'ikarus plex'
	list proto 'tcp'
	option src 'wan'
	option src_dport '32400'
	option dest_ip '10.131.0.10'
	option dest_port '32400'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'ikarus https'
	option src 'wan'
	option src_dport '443'
	option dest_ip '10.131.0.10'
	option dest_port '443'
	list proto 'tcp'
	list proto 'udp'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'ikarus influxdb'
	list proto 'tcp'
	option src 'wan'
	option src_dport '42868'
	option dest_ip '10.131.0.10'
	option dest_port '42868'

config redirect
	option dest 'lan'
	option target 'DNAT'
	option name 'ikarus syncthing'
	option src 'wan'
	option src_dport '22000'
	option dest_ip '10.131.0.10'
	option dest_port '22000'

config zone
	option name 'iost'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	list network 'iost'

config forwarding
	option src 'iost'
	option dest 'wan'

config forwarding
	option src 'lan'
	option dest 'iost'

config rule
	option name 'iot mqtt'
	list proto 'tcp'
	option src 'iost'
	option dest 'lan'
	list dest_ip '10.131.0.10'
	option dest_port '1883'
	option target 'ACCEPT'

config zone 'docker'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'ACCEPT'
	option name 'docker'
	list network 'docker'

config forwarding
	option src 'foo'
	option dest 'lan'

config forwarding
	option src 'lan'
	option dest 'foo'
```

blackbrayn, 2024-11-12:
Just experienced the same thing, after many trials, uninstalling dockerd package + reboot solved the issue.

mbooij, 2025-03-03:
same here (with 23.05 and 24.10)

Avrution, 2025-09-05:
Been fighting this same issue for days and none of the solutions have worked for me.  Even this one still blocks loopback/pinhole.  Did you have to change any other rules or just this and boom, worked?

> Update/workaround: manualy setting loopback source IP to external and refective zone (both in Advanced settings of the port forward rule in the firewall menu) to wan fixed this for me. Both where default and working before dockerd install.

atikhobaev, 2026-02-27:
Installing Docker on OpenWrt forces `bridge-nf-call-iptables` to `1`, which breaks internal Hairpin NAT (NAT Loopback) on `br-lan`.

These settings are automatically written to `/etc/sysctl.d/12-br-netfilter-ip.conf`. To fix this, simply comment out the lines in that file. In my case, they were:

```
# net.bridge.bridge-nf-call-iptables=1
# net.bridge.bridge-nf-call-ip6tables=1
```

(Or just change the values to 0). This prevents Docker from overriding your network configuration and restores NAT Loopback functionality.

Additionally, to ensure Docker doesn't drop forwarded packets while these netfilter checks are disabled, add `"ip-forward-no-drop": true` to your `/etc/docker/daemon.json`